tactcomplabs/rev #27: AUIPC simulation bug

Artlands, 2023-01-31:
The `auipc` instruction for RV64 has a bug when encountering negative immediates and it generates an incorrect address. The bug is that it did not sign-extend the value after bit shifting.

To reproduce the issue, compile the following C program with `riscv64-unknown-elf-gcc -march=rv64imafd`. Since we have some issues with decoding the compressed instruction, the test is compiled without the `compressed` option. 

Here is a snippet of the simulation output:
[image]

When investing in the disassembled object file, we can see that the correct jump address should be `0x10fe0' instead of `0x100010fe0`.

[image]

Sign-extending the bit shifted immediate will solve the issue:

[image]

jleidel, 2023-01-31:
Will test in our current dev branch and report back

jleidel, 2023-02-21:
fix merged to devel

closed 2023-02-21 by jleidel