tactcomplabs/rev #40: Missing support for ELF init/fini sections

ukasz, 2023-04-19:
ELF's init section is used for things like functions defined with constructor attribute and static class members initialization. Currently init and fini sections are not executed before/after main.
The following code snippet represents the problem:

```
#define assert(x) if (!(x)) { asm(".dword 0x00000000"); }
int x = 0;
__attribute__((constructor)) void foo() {
        x++;
}

int main() {
        assert(x == 1);
        return 0;
}
```

jleidel, 2023-04-19:
this is a known issue.  we're working it.  RISC-V doesn't fully support portions of the ELF spec

ukasz, 2023-08-05:
I am changing that to P0 because it is super annoying to deal with as we work more and more with c++ code. This issue has huge negative impact for c++ applications: a) static members of a function cannot be initialized properly b) constructors of globally defined classed are never called.

Examples below:
```
class X {
        public:
        X() {
                magic = 42;
        }
        int magic;
};

X x;

int main() {
        assert(x.magic == 42);

        return 0;
}
```
class X {
        public:
        static int magic;
};

int X::magic = 42;

int main() {
        assert(X::magic == 42);

        return 0;
}
```

jleidel, 2023-08-05:
feel free to start execution at `_start`

ukasz, 2023-08-07:
I wasn't aware of the lack of .init/.fini sections on RV, but the same mechanism seems to be implemented on risc using .text and .plt sections.

I tried starting directly from _start, our libc runs calls set_tid_address syscall, I know that you are working on TLS support, would it be a problem to add support for this syscall, or should we try to do something on our end with it?

I briefly added a dummy set_tid_address syscall implementation and with that in place all three examples posted here work fine for us.

```
diff --git a/src/RevProc.cc b/src/RevProc.cc
index 6175d23..b76acc0 100644
--- a/src/RevProc.cc
+++ b/src/RevProc.cc
@@ -2412,6 +2412,7 @@ void RevProc::InitEcallTable(){
     {93,  &RevProc::ECALL_exit},
     {94,  &RevProc::ECALL_exit_group},      // Not implemented
     {95,  &RevProc::ECALL_waitid},          // Not implemented
+    {96,  &RevProc::ECALL_set_tid_address}, // Not implemented
     {99,  &RevProc::ECALL_set_robust_list}, // Not implemented
     {100, &RevProc::ECALL_get_robust_list}, // Not implementedt
     {101, &RevProc::ECALL_nanosleep},       // Not implemented
@@ -2789,6 +2790,11 @@ void RevProc::ECALL_munmap(){
   return;
 }

+void RevProc::ECALL_set_tid_address(){
+  output->verbose(CALL_INFO, 2, 0, "ECALL: set_tid_address called\n");
+  return;
+}
+
```

closed 2023-08-25 by jleidel