tactcomplabs/rev #51: Dual memset fail

ukasz, 2023-05-11:
The code listed below does not work properly.
Tested on devel branch (84260da280304c05493da2d7ee18bdeb5a01f39a) and standard config.
First memset produces correct results, but we hit an assert after a second memset call, which is unexpected.
`
#include <string.h>
#define assert(x) if (!(x)) { asm(".byte 0x00"); asm(".byte 0x00"); asm(".byte 0x00"); asm(".byte 0x00"); }
#define N (1024)
char mem[N];

int main() {
         memset(mem, 42, N);
         for(int i = 0; i < N; i++)
             assert(mem[i] == 42);
         memset(mem, 0, N);
         for(int i = 0; i < N; i++)
                 assert(mem[i] == 0);
        return 0;
}
`

donofrio, 2023-05-17:
Issue determined to be a problem with the store instruction - fix pushed to latest devel branch

ukasz, 2023-05-19:
Fix confirmed, can be closed.

closed 2023-05-19 by jleidel