tactcomplabs/rev #54: strstr fail

ukasz, 2023-05-19:
We see the following strstr test failing on devel branch (tested sha: cd753f7c20)
```
#include <stdio.h>
#include <string.h>
#define assert(x)                                                              \
  if (!(x)) {                                                                  \
    asm(".byte 0x00");                                                         \
    asm(".byte 0x00");                                                         \
    asm(".byte 0x00");                                                         \
    asm(".byte 0x00");                                                         \
  }

const char * const text = "Hello I am a normal text with some pattern hidden inside.";

int main() {
        assert(strstr(text, "pattern") == text + 35);
        return 0;
}
```

ukasz, 2023-06-16:
Still present on (581618e82)

jleidel, 2023-06-29:
Tested to work as of 80d0edb

ukasz, 2023-06-29:
For me it still fails when compiled with our LLVM toolchain.

jleidel, 2023-06-30:
Can we get the exact compilation string using your toolchain?  We can compile to an executable, but the toolchain has unimplemented instructions: 

disassembly: 
```
000000000001120c <main>:
   1120c:       1101                    add     sp,sp,-32
   1120e:       ec06                    sd      ra,24(sp)
   11210:       e822                    sd      s0,16(sp)
   11212:       1000                    add     s0,sp,32
   11214:       4501                    li      a0,0
   11216:       fea42623                sw      a0,-20(s0)
   1121a:       00010537                lui     a0,0x10
   1121e:       15950513                add     a0,a0,345 # 10159 <text-0x4f>
   11222:       fea43023                sd      a0,-32(s0)
   11226:       000105b7                lui     a1,0x10
   1122a:       19358593                add     a1,a1,403 # 10193 <text-0x15>
   1122e:       513000ef                jal     11f40 <strstr>
   11232:       fe043583                ld      a1,-32(s0)
   11236:       02358593                add     a1,a1,35
   1123a:       00b50863                beq     a0,a1,1124a <main+0x3e>
   1123e:       0040006f                j       11242 <main+0x36>
   11242:       0000                    unimp
   11244:       0000                    unimp
   11246:       0040006f                j       1124a <main+0x3e>
   1124a:       4501                    li      a0,0
   1124c:       60e2                    ld      ra,24(sp)
   1124e:       6442                    ld      s0,16(sp)
   11250:       6105                    add     sp,sp,32
   11252:       8082                    ret  
```

closed 2023-08-05 by jleidel