tactcomplabs/rev #56: Atomic increment doesn't work

ukasz, 2023-05-19:
We observe cpp atomic operations to fail on data sizes wider than char (tested sha: cd753f7 compiled with LLVM)
Simplest example of such behavior is depicted below.
```
#include <atomic>
#define assert(x) if (!(x)) { asm(".byte 0x00"); asm(".byte 0x00"); asm(".byte 0x00"); asm(".byte 0x00"); }

#define N 1000
int main() {
        std::atomic<int> a;
        a = 0;
        a++;
        assert(a == 1);
}
```

ukasz, 2023-06-16:
Even if atomic operations cannot be "atomic" yet, we would like to see their behavior emulated, so they are functionally working on a single thread.

jleidel, 2023-08-05:
the single threaded versions are functionally emulated using only the embedded memory infrastructure; currently adding support for these on the `amo` branch

jleidel, 2023-08-25:
fixed

closed 2023-08-25 by jleidel