tactcomplabs/rev #58: REV segfault on workload execution

marcinkoss, 2023-05-24:
* Take SPMV code https://github.com/riscv-software-src/riscv-tests/tree/master/benchmarks/spmv
* Build with RISC-V gcc musl (gcc build instuctions can be found at: https://github.com/riscv-software-src/riscv-tests/tree/master and set correct gcc compiler )
* run with REV_EXE=spmv.riscv timeout 20s sst config.py

Workload executes for couple of seconds and then crashes. This happens with both gcc and llvm compiled binary.
```
RevCPU[cpu:DecodeInst:319977000]: Core 0 ; Thread 0; PC:InstPayload = 0x8000b10c:0xb50533
RevCPU[cpu:ClockTick:319977000]: Core 0 ; Thread 0; Executing PC= 0x8000b10c
RevCPU[cpu:ClockTick:319977000]: Core 0 ; ThreadID 0; Retiring PC= 0x8000b10c
RevCPU[cpu:DecodeInst:319978000]: Core 0 ; Thread 0; PC:InstPayload = 0x8000b110:0x53087
RevCPU[cpu:ClockTick:319978000]: Core 0 ; Thread 0; Executing PC= 0x8000b110
 *** Process received signal ***
 Signal: Segmentation fault (11)
 Signal code: Invalid permissions (2)
 Failing at address: 0x7fdd64028088
 [ 0] /lib/x86_64-linux-gnu/libc.so.6(+0x42520)[0x7fdd6c917520]
 [ 1] rev/src/librevcpu.so(_ZN3SST6RevCPU6RevMem7ReadMemEmmPv+0xb0)[0x7fdd68afdec0]
 [ 2] rev/src/librevcpu.so(_ZN3SST6RevCPU6RevMem10ReadDoubleEm+0x24)[0x7fdd68afe1f4]
[ 3] rev/src/librevcpu.so(_ZN3SST6RevCPU5RV32D3fldEPNS0_10RevFeatureEPNS0_10RevRegFileEPNS0_6RevMemENS0_7RevInstE+0x38)[0x7fdd68b12b18]
[ 4] rev/src/librevcpu.so(_ZN3SST6RevCPU7RevProc9ClockTickEm+0x388)[0x7fdd68b0b618]
[ 5] rev/src/librevcpu.so(_ZN3SST6RevCPU6RevCPU9clockTickEm+0xa6)[0x7fdd68ae0e66]
[ 6] sst(_ZN3SST5Clock7executeEv+0x10c)[0x55a840109cec]
[ 7] sst(_ZN3SST15Simulation_impl3runEv+0x3d3)[0x55a84019ef93]
[ 8] sst(+0xf6e54)[0x55a8400ebe54]
[ 9] sst(main+0x18a3)[0x55a8400d1753]
[10] /lib/x86_64-linux-gnu/libc.so.6(+0x29d90)[0x7fdd6c8fed90]
[11] /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0x80)[0x7fdd6c8fee40]
[12] sst(_start+0x25)[0x55a8400e93e5]
 *** End of error message ***
Segmentation fault
```

Config file:
```
import os
import sst
import sys

program = os.getenv("REV_EXE", "xxx")

# Define SST core options
sst.setProgramOption("timebase", "1ps")
sst.setProgramOption("stopAtCycle", "0s")

# Tell SST what statistics handling we want
sst.setStatisticLoadLevel(4)

max_addr_gb = 1

# Define the simulation components
comp_cpu = sst.Component("cpu", "revcpu.RevCPU")
comp_cpu.addParams({
        "verbose" : verbose_level,                    # Verbosity
        "numCores" : 1,                               # Number of cores
        "clock" : "1.0GHz",                           # Clock
        "memSize" : 1024*1024*10,                     # Memory size in bytes
        "machine" : "[0:RV64IMAFD]",                  # Core:Config; RV64I for core 0
        "memCost" : "[0:1:10]",                       # Memory loads required 1-10 cycles
        "program" : program,                          # Target executable
        "splash" : 1                                  # Display the splash message
})

sst.setStatisticOutput("sst.statOutputCSV")
sst.enableAllStatisticsForAllComponents()

# EOF
```

donofrio, 2023-05-24:
Try adding  "startAddr" : "[0:0x000]" to your list of comp_cpu params

Do you see the same failures with the unknown-elf compiler?

marcinkoss, 2023-05-25:
It does not really matter what compiler I used (and yes - I tested with unknown-elf compiler as well).
Start address was selected as main() function.
```
0000000080002b5c <main>:
    80002b5c:   fffff7b7                lui     a5,0xfffff
    80002b60:   00001737                lui     a4,0x1
    80002b64:   81010113                add     sp,sp,-2032
    80002b68:   06078793                add     a5,a5,96 # fffffffffffff060 <_tbss_end+0xffffffff7fff2a5c>
```

In attachment log with verbose=6 generated with riscv64-unknown-elf-gcc (raw REV log, converted to executed code)
[spmv.log](https://github.com/tactcomplabs/rev/files/11562392/spmv.log)
[parsed.log](https://github.com/tactcomplabs/rev/files/11562396/parsed.log)

marcinkoss, 2023-06-19:
It seems this is no longer issue in newest devel branch.