tactcomplabs/rev #59: Infinite loop on  element remove from list

marcinkoss, 2023-05-24:
* Take towers code https://github.com/riscv-software-src/riscv-tests/tree/master/benchmarks/towers
* Build with RISC-V gcc musl (gcc build instuctions can be found at: https://github.com/riscv-software-src/riscv-tests/tree/master and set correct gcc compiler )
* run with REV_EXE=towers.riscv timeout 20s sst config.py
Workload executes and enters infinite loop in list_pop() function.
This happens with both gcc and llvm compiled binary.

below you can find call from main() and first iterations in list_pop()
```
8000308c >> <main>:     00810513                add     a0,sp,8
80003090 >> <main>:     88cff0ef                jal     8000211c <towers_clear>
8000211c >> <towers_clear>:     00852703                lw      a4,8(a0)
80002120 >> <towers_clear>:     04e05063                blez    a4,80002160 <towers_clear+0x44>
80002124 >> <towers_clear>:     9a818813                add     a6,gp,-1624 # 80003270 <g_nodeFreeList>
80002128 >> <towers_clear>:     01053783                ld      a5,16(a0)
8000212c >> <towers_clear>:     00883683                ld      a3,8(a6)
80002130 >> <towers_clear>:     00c0006f                j       8000213c <towers_clear+0x20>
8000213c >> <towers_clear>:     0087b603                ld      a2,8(a5)
80002140 >> <towers_clear>:     fff7071b                addw    a4,a4,-1
80002144 >> <towers_clear>:     00d7b423                sd      a3,8(a5)
80002148 >> <towers_clear>:     00078593                mv      a1,a5
8000214c >> <towers_clear>:     fe0714e3                bnez    a4,80002134 <towers_clear+0x18>
80002134 >> <towers_clear>:     00060793                mv      a5,a2
80002138 >> <towers_clear>:     00058693                mv      a3,a1
8000213c >> <towers_clear>:     0087b603                ld      a2,8(a5)
80002140 >> <towers_clear>:     fff7071b                addw    a4,a4,-1
80002144 >> <towers_clear>:     00d7b423                sd      a3,8(a5)
80002148 >> <towers_clear>:     00078593                mv      a1,a5
8000214c >> <towers_clear>:     fe0714e3                bnez    a4,80002134 <towers_clear+0x18>
80002134 >> <towers_clear>:     00060793                mv      a5,a2
80002138 >> <towers_clear>:     00058693                mv      a3,a1
8000213c >> <towers_clear>:     0087b603                ld      a2,8(a5)
80002140 >> <towers_clear>:     fff7071b                addw    a4,a4,-1
80002144 >> <towers_clear>:     00d7b423                sd      a3,8(a5)
80002148 >> <towers_clear>:     00078593                mv      a1,a5
8000214c >> <towers_clear>:     fe0714e3                bnez    a4,80002134 <towers_clear+0x18>
80002134 >> <towers_clear>:     00060793                mv      a5,a2
80002138 >> <towers_clear>:     00058693                mv      a3,a1
8000213c >> <towers_clear>:     0087b603                ld      a2,8(a5)
80002140 >> <towers_clear>:     fff7071b                addw    a4,a4,-1
80002144 >> <towers_clear>:     00d7b423                sd      a3,8(a5)
80002148 >> <towers_clear>:     00078593                mv      a1,a5
8000214c >> <towers_clear>:     fe0714e3                bnez    a4,80002134 <towers_clear+0x18>
80002134 >> <towers_clear>:     00060793                mv      a5,a2
80002138 >> <towers_clear>:     00058693                mv      a3,a1
8000213c >> <towers_clear>:     0087b603                ld      a2,8(a5)
80002140 >> <towers_clear>:     fff7071b                addw    a4,a4,-1
80002144 >> <towers_clear>:     00d7b423                sd      a3,8(a5)
80002148 >> <towers_clear>:     00078593                mv      a1,a5
8000214c >> <towers_clear>:     fe0714e3                bnez    a4,80002134 <towers_clear+0x18>
80002134 >> <towers_clear>:     00060793                mv      a5,a2
80002138 >> <towers_clear>:     00058693                mv      a3,a1
8000213c >> <towers_clear>:     0087b603                ld      a2,8(a5)
80002140 >> <towers_clear>:     fff7071b                addw    a4,a4,-1
80002144 >> <towers_clear>:     00d7b423                sd      a3,8(a5)
80002148 >> <towers_clear>:     00078593                mv      a1,a5
8000214c >> <towers_clear>:     fe0714e3                bnez    a4,80002134 <towers_clear+0x18>
80002134 >> <towers_clear>:     00060793                mv      a5,a2
80002138 >> <towers_clear>:     00058693                mv      a3,a1
8000213c >> <towers_clear>:     0087b603                ld      a2,8(a5)
80002140 >> <towers_clear>:     fff7071b                addw    a4,a4,-1
80002144 >> <towers_clear>:     00d7b423                sd      a3,8(a5)
80002148 >> <towers_clear>:     00078593                mv      a1,a5
8000214c >> <towers_clear>:     fe0714e3                bnez    a4,80002134 <towers_clear+0x18>
```

This is infinitte loop
```
80002134 >> <towers_clear>:     00060793                mv      a5,a2
80002138 >> <towers_clear>:     00058693                mv      a3,a1
8000213c >> <towers_clear>:     0087b603                ld      a2,8(a5)
80002140 >> <towers_clear>:     fff7071b                addw    a4,a4,-1
80002144 >> <towers_clear>:     00d7b423                sd      a3,8(a5)
80002148 >> <towers_clear>:     00078593                mv      a1,a5
8000214c >> <towers_clear>:     fe0714e3                bnez    a4,80002134 <towers_clear+0x18>
```

donofrio, 2023-06-20:
This appears to only manifest with -O2, I will continue debug, but can you verify that this issue does not manifest with -O0?

donofrio, 2023-06-21:
Root caused and fixed in latest ([2c6320c]) devel. Towers benchmark now included in test suite and works with -O2

marcinkoss, 2023-06-21:
Thanks, I will check if after code sync

marcinkoss, 2023-06-23:
Confirmed - it is working fine now.