tactcomplabs/rev #65: broken bge instruction

ukasz, 2023-06-20:
The following c code fails due to reading data out of memory because conditional branch was not taken.
First two iterations are fine, but third does not stop causing an assert via comparing garbage from the stack.
```
#define assert(x)                                                              \
  if (!(x)) {                                                                  \
    asm(".byte 0x00");                                                         \
    asm(".byte 0x00");                                                         \
    asm(".byte 0x00");                                                         \
    asm(".byte 0x00");                                                         \
  }

struct FE{
        int ec;
};

int x[] = {0, 1};
int Forums_ = 2;

int main() {
        struct FE forums[2];
        forums[0].ec = 0;
        forums[1].ec = 1;

        for (int i = 0; i < Forums_; i++) {
                assert(x[i] == forums[i].ec);
        }
        return 0;
}
```
Compiled as:
```
$ clang -march=rv64imafd tmp.c -O2 -o tmp.exe

```
The execution log shows that branch at bge instruction  is not taken when a1 and a2 registers are equal.
```
00000000000111c0 <main>:
   111c0: 13 01 01 ff   addi    sp, sp, -16
   111c4: 37 35 01 00   lui     a0, 19
   111c8: 83 25 45 f2   lw      a1, -220(a0)
   111cc: 13 06 10 00   li      a2, 1
   111d0: 13 16 06 02   slli    a2, a2, 32
   111d4: 23 34 c1 00   sd      a2, 8(sp)
   111d8: 63 50 b0 04   blez    a1, 0x11218 <main+0x58>
   111dc: 13 06 00 00   li      a2, 0
   111e0: b7 36 01 00   lui     a3, 19
   111e4: 93 86 c6 f1   addi    a3, a3, -228
   111e8: 13 07 81 00   addi    a4, sp, 8
   111ec: 6f 00 40 01   j       0x11200 <main+0x40>
   111f0: 13 06 16 00   addi    a2, a2, 1
   111f4: 13 07 47 00   addi    a4, a4, 4
   111f8: 93 86 46 00   addi    a3, a3, 4
   **111fc: 63 5e b6 00   bge     a2, a1, 0x11218 <main+0x58>**
   11200: 83 a7 06 00   lw      a5, 0(a3)
   11204: 03 28 07 00   lw      a6, 0(a4)
   11208: e3 84 07 ff   beq     a5, a6, 0x111f0 <main+0x30>
   1120c: 00 00         <unknown>
   1120e: 00 00         <unknown>
   11210: 83 25 45 f2   lw      a1, -220(a0)
   11214: 6f f0 df fd   j       0x111f0 <main+0x30>
   11218: 13 05 00 00   li      a0, 0
   1121c: 13 01 01 01   addi    sp, sp, 16
   11220: 67 80 00 00   ret

```
And the offending part is:

```

RevCPU[cpu:DecodeInst:29000]: Core 0 ; Thread 0; PC:InstPayload = **0x111fc:**0xb65e63
RevCPU[cpu:ClockTick:29000]: Core 0 ; Thread 0; Executing PC= 0x111fc
RevCPU[cpu:ClockTick:29000]: Core 0 ; ThreadID 0; Retiring PC= 0x111fc
RevCPU[cpu:ClockTick:29000]: Core 0 REG[0] 0x0
RevCPU[cpu:ClockTick:29000]: Core 0 REG[1] 0x0
RevCPU[cpu:ClockTick:29000]: Core 0 REG[2] 0x7e7ffed0
RevCPU[cpu:ClockTick:29000]: Core 0 REG[3] 0x1371c
RevCPU[cpu:ClockTick:29000]: Core 0 REG[4] 0x0
RevCPU[cpu:ClockTick:29000]: Core 0 REG[5] 0x0
RevCPU[cpu:ClockTick:29000]: Core 0 REG[6] 0x0
RevCPU[cpu:ClockTick:29000]: Core 0 REG[7] 0x0
RevCPU[cpu:ClockTick:29000]: Core 0 REG[8] 0x1371c
RevCPU[cpu:ClockTick:29000]: Core 0 REG[9] 0x0
RevCPU[cpu:ClockTick:29000]: Core 0 REG[10] 0x13000
**RevCPU[cpu:ClockTick:29000]: Core 0 REG[11] 0x2
RevCPU[cpu:ClockTick:29000]: Core 0 REG[12] 0x2**
RevCPU[cpu:ClockTick:29000]: Core 0 REG[13] 0x12f24
RevCPU[cpu:ClockTick:29000]: Core 0 REG[14] 0x7e7ffee0
RevCPU[cpu:ClockTick:29000]: Core 0 REG[15] 0x1
RevCPU[cpu:ClockTick:29000]: Core 0 REG[16] 0x1
RevCPU[cpu:ClockTick:29000]: Core 0 REG[17] 0x0
RevCPU[cpu:ClockTick:29000]: Core 0 REG[18] 0x0
RevCPU[cpu:ClockTick:29000]: Core 0 REG[19] 0x0
RevCPU[cpu:ClockTick:29000]: Core 0 REG[20] 0x0
RevCPU[cpu:ClockTick:29000]: Core 0 REG[21] 0x0
RevCPU[cpu:ClockTick:29000]: Core 0 REG[22] 0x0
RevCPU[cpu:ClockTick:29000]: Core 0 REG[23] 0x0
RevCPU[cpu:ClockTick:29000]: Core 0 REG[24] 0x0
RevCPU[cpu:ClockTick:29000]: Core 0 REG[25] 0x0
RevCPU[cpu:ClockTick:29000]: Core 0 REG[26] 0x0
RevCPU[cpu:ClockTick:29000]: Core 0 REG[27] 0x0
RevCPU[cpu:ClockTick:29000]: Core 0 REG[28] 0x0
RevCPU[cpu:ClockTick:29000]: Core 0 REG[29] 0x0
RevCPU[cpu:ClockTick:29000]: Core 0 REG[30] 0x0
RevCPU[cpu:ClockTick:29000]: Core 0 REG[31] 0x0
RevCPU[cpu:clockTick:30000]: Cycle: 30
RevCPU[cpu:DecodeInst:30000]: Core 0 ; Thread 0; PC:InstPayload = **0x11200**:0x6a783
```

ukasz, 2023-06-20:
A much simpler example:
```
int main() {
        asm("li a2,2; li a3,2; bge a2,a3, 0x2; .byte 0x00; .byte 0x00; ");
        return 0;
}
```

jleidel, 2023-06-20:
Verifying the bug.  Test case created.  Disassembly of test case is: 
```
000000000001019c <main>:
   1019c:       ff010113                add     sp,sp,-16
   101a0:       00813423                sd      s0,8(sp)
   101a4:       01010413                add     s0,sp,16
   101a8:       00200613                li      a2,2
   101ac:       00200693                li      a3,2
   101b0:       00d64463                blt     a2,a3,101b8 <main+0x1c>
   101b4:       e55ef06f                j       8 <exit-0x100e0>
   101b8:       00000000                .word   0x00000000
   101bc:       00200613                li      a2,2
   101c0:       00200693                li      a3,2
   101c4:       00c6c463                blt     a3,a2,101cc <main+0x30>
   101c8:       e41ef06f                j       8 <exit-0x100e0>
   101cc:       00000000                .word   0x00000000
   101d0:       00000793                li      a5,0
   101d4:       00078513                mv      a0,a5
   101d8:       00813403                ld      s0,8(sp)
   101dc:       01010113                add     sp,sp,16
   101e0:       00008067                ret
```

jleidel, 2023-06-20:
bug verified with riscv-unknown-elf-gcc 12.2.0; SST 12.1.0 & 13.0.0; triaging now

jleidel, 2023-06-20:
Notice that the test case never actually generates `bge` instructions.  As a result, the branches are never taken.

ukasz, 2023-06-20:
Yeah, but it is a compiler thing. Use minimized asm code example I posted later to reproduce bug.

jleidel, 2023-06-20:
The disassembly posted early was produced using the minimized asm code.

donofrio, 2023-06-21:
Please re-test with [2c6320c] - I believe this issue is also addressed

ukasz, 2023-06-21:
Fix confirmed

closed 2023-06-21 by ukasz