tactcomplabs/rev #66: Infinite loop on divw instruction

marcinkoss, 2023-06-20:
Compile and run program:
```
int main() {
        for (int i=0;i<2;i++)
        {
                int zz = (i+1)/i;
        }
        return 0;
}
```
Execution enters infinite loop on divw instruction:

```
628 >> <main>:  fe010113                addi    sp,sp,-32
62c >> <main>:  00813c23                sd      s0,24(sp)
630 >> <main>:  02010413                addi    s0,sp,32
634 >> <main>:  fe042423                sw      zero,-24(s0)
638 >> <main>:  0280006f                j       660 <main+0x38>
660 >> <main>:  fe842783                lw      a5,-24(s0)
664 >> <main>:  0007871b                sext.w  a4,a5
668 >> <main>:  00100793                li      a5,1
66c >> <main>:  fce7d8e3                bge     a5,a4,63c <main+0x14>
63c >> <main>:  fe842783                lw      a5,-24(s0)
640 >> <main>:  0017879b                addiw   a5,a5,1
644 >> <main>:  0007879b                sext.w  a5,a5
648 >> <main>:  fe842703                lw      a4,-24(s0)
64c >> <main>:  02e7c7bb                divw    a5,a5,a4
64c >> <main>:  02e7c7bb                divw    a5,a5,a4
64c >> <main>:  02e7c7bb                divw    a5,a5,a4
64c >> <main>:  02e7c7bb                divw    a5,a5,a4
64c >> <main>:  02e7c7bb                divw    a5,a5,a4
64c >> <main>:  02e7c7bb                divw    a5,a5,a4
64c >> <main>:  02e7c7bb                divw    a5,a5,a4
```

jleidel, 2023-06-20:
confirmed infinite loop exists

marcinkoss, 2023-06-23:
I can confirm there is no infinite loop
however above code behaves incorectly as it does (i+1)/i for i=0 (divide by 0).

jleidel, 2023-06-30:
Fix and tests pushed in 3e692bd

closed 2023-07-18 by marcinkoss