tactcomplabs/rev #67: set/read floating-point exceptions does not work

marcinkoss, 2023-06-21:
Compile and run below code:

```
#define assert(x) if (!(x)) { asm(".dword 0x00000000"); }

#include <fenv.h>
#include <stdint.h>

void test_except(uint64_t e) {
	feclearexcept(FE_ALL_EXCEPT);
	int r = feraiseexcept(e);
	assert (r == 0);
	assert(fetestexcept(e) > 0);
}

int main() {
	test_except(FE_DIVBYZERO);
	test_except(FE_INEXACT);
	test_except(FE_INVALID);
        test_except(FE_OVERFLOW);
	test_except(FE_UNDERFLOW);
	return 0;
}
```
When calling fetestexcept() we should get value of previously set exception. In runtime it seems that no exception is raised.
See also: 
https://en.cppreference.com/w/c/numeric/fenv/FE_exceptions
https://en.cppreference.com/w/cpp/numeric/fenv/fetestexcept

leekillough, 2024-06-25:
Please verify it has been fixed

leekillough, 2024-07-16:
`feraiseexcept()` returns nonzero if its argument is nonzero, which seems to be a library bug (I compiled with ` -frounding-math -ftrapping-math -fsignaling-nans`):
```
00000000000101fa <feraiseexcept>:
   101fa:       897d                    and     a0,a0,31
   101fc:       00152073                csrs    fflags,a0
   10200:       00a03533                snez    a0,a0
   10204:       8082                    ret
```
If we ignore the return value of `feraiseexcept()`, the test passes.

leekillough, 2024-07-17:
The return value of `feraiseexcept()` should always be zero, which it is in [current glibc sources](https://github.com/bminor/glibc/blob/21c3f4b5368686ade28d90d8c7d79c4c95c72c1b/sysdeps/riscv/rvf/fraiseexcpt.c). It seems like [Newlib has this bug](https://github.com/bminor/newlib/blob/master/newlib/libm/machine/riscv/feraiseexcept.c).

Closing this issue as completed.