tactcomplabs/rev #72: Programmer can write value to x0 (zero) register

marcinkoss, 2023-06-28:
x0 (zero) register is hardwired 0. This means any write to this register should be discarded and all read of this register must return 0.

Run below code
int main() {
       asm (" li a0, 5; addi zero, a0, 0; addi a0, zero, 6;")   ;
       return 0;
}

I expect that a0=6, but it's 11 instead.

This is related to https://github.com/tactcomplabs/rev/issues/64.

See attached logs for more details with tracked register changes
[image]

marcinkoss, 2023-06-28:
I checked this on riscv/qemu and gdb and if user try to write value into x0 value gets discarded.

donofrio, 2023-06-30:
We have added a change to Rev that the 32 and 64 bit Register files will always ave register 0 set to zero,  This test case was also added (Test "x0").  Should be working in [62a406b]

marcinkoss, 2023-07-18:
I can confirm that x0 is not corrupted now

closed 2023-07-18 by marcinkoss