tactcomplabs/rev #73: mret instruction corrupts x0 register

marcinkoss, 2023-06-29:
Follow steps from https://github.com/tactcomplabs/rev/issues/64
See also: https://github.com/tactcomplabs/rev/issues/72

At PC=0x190 REV executes mret instruction which leads to x0 corruption.
Problematic instruction is:
 73 00 20 30        mret
it's get decoded as lui and 0x30200000 value is written into x0 (it seems 2 bytes from mret instruction was used to overwrite x0).

```
RevCPU[cpu:ClockTick:77000]: Core 0 ; ThreadID 0; Retiring PC= 0x8000018c
RevCPU[cpu:clockTick:78000]: Cycle: 78
RevCPU[cpu:DecodeInst:78000]: Core 0 ; Thread 0; PC:InstPayload = 0x80000190:0x30200073
RevCPU[cpu:ClockTick:78000]: Core 0 ; Thread 0; Executing PC= 0x80000190
--------------- lui 32
RevCPU[cpu:ClockTick:78000]: Core 0 ; ThreadID 0; Retiring PC= 0x80000190
RevCPU[cpu:ClockTick:78000]: Core 0 REG[0] 0x0 -> 0x30200000
RevCPU[cpu:clockTick:79000]: Cycle: 79
RevCPU[cpu:DecodeInst:79000]: Core 0 ; Thread 0; PC:InstPayload = 0x80000194:0x200193
RevCPU[cpu:ClockTick:79000]: Core 0 ; Thread 0; Executing PC= 0x80000194
```

Now, while I understand we might not support mret instuction in REV - it should not corrupt any other data.
I think in case  REV executes not supported instruction it should:
* write this into to log file - with some details
* not corrupt code/data

I debuged this on mret instruction, but I see sometimes issues with ret/jal jumping into wrong address and this might be something related.
[full_run.txt](https://github.com/tactcomplabs/rev/files/11903325/full_run.txt)

See full execution log in attachment

jleidel, 2023-06-29:
we have a temporary fix in place where we hammer x0 to zero after the instruction executes in adb62a9

marcinkoss, 2023-06-30:
One more note - mret instruction was decoded as lui - which should not happen.

jleidel, 2023-06-30:
`mret` is not currently supported in Rev