tactcomplabs/rev #75: SLTI instruction fails

ukasz, 2023-06-30:
After enabling SLTI ISA test we observe fails related to sign extension.
On the other hand SLTUI tests are passing, where all numbers as treated as unsigned.

```
TEST_IMM_OP( 15, slti, 1, 0xffffffffffffffff, 0x001 );

00000000000111e4 <test_15>:
   111e4: 93 01 f0 00   li      gp, 15
   111e8: 93 00 f0 ff   li      ra, -1
   111ec: 13 a7 10 00   slti    a4, ra, 1
   111f0: 93 03 10 00   li      t2, 1
   111f4: 63 16 77 00   bne     a4, t2, 0x11200 <fail>
   111f8: 63 12 30 00   bne     zero, gp, 0x111fc <pass>
```
   
-1 is smaller than, so a2 should be set to 1, while it is not.
   
Other failing test cases are:
```
TEST_IMM_OP( 7,  slti, 1, 0xffffffff80000000, 0x000 );
TEST_IMM_OP( 8,  slti, 1, 0xffffffff80000000, 0x800 );
TEST_IMM_OP( 12, slti, 1, 0xffffffff80000000, 0x7ff );
```

donofrio, 2023-06-30:
Sign extension fixed and slti test added.  Should be working properly in [62a406b]

ukasz, 2023-07-07:
Fix confirmed

closed 2023-07-14 by ukasz