tactcomplabs/rev #79: Stack corruption for parent process when fork() syscall is called

aperdeus, 2023-07-07:
The following test which checks if the stack has been properly copied for the child process created by fork() should not fail.

`#include <stdlib.h>
#include <unistd.h>
#include <sys/types.h>
#include <sys/wait.h>

#include "../../../common/syscalls/syscalls.h"

#define assert(x) if (!(x)) { asm(".byte 0x00"); asm(".byte 0x00"); asm(".byte 0x00"); asm(".byte 0x00"); }
int main() {
  int valOnStack = 1;
  rev_fork();
  assert(valOnStack==1);
  valOnStack = 0;

  return 0;
}`

rkabrick, 2023-07-10:
We are working on architecting a new SMT implementation which will fix this

jleidel, 2023-09-28:
is this still an issue?

jleidel, 2023-10-25:
closing due to no response

closed 2023-10-25 by jleidel