tactcomplabs/rev #80: Improper behaviour of auipc instruction

marcinkoss, 2023-07-11:
Download and run rv64ui-p-auipc test from [riscv](https://github.com/riscv-software-src/riscv-tests)
Expected: test passing without trouble
Actual result: test fails, full execution log:

```
800000a0 >> <reset_vector>:  13 0a 00 00        li      s4, 0
800000a4 >> <reset_vector>:  93 0a 00 00        li      s5, 0
800000a8 >> <reset_vector>:  13 0b 00 00        li      s6, 0
800000ac >> <reset_vector>:  93 0b 00 00        li      s7, 0
800000b0 >> <reset_vector>:  13 0c 00 00        li      s8, 0
800000b4 >> <reset_vector>:  93 0c 00 00        li      s9, 0
800000b8 >> <reset_vector>:  13 0d 00 00        li      s10, 0
800000bc >> <reset_vector>:  93 0d 00 00        li      s11, 0
800000c0 >> <reset_vector>:  13 0e 00 00        li      t3, 0
800000c4 >> <reset_vector>:  93 0e 00 00        li      t4, 0
800000c8 >> <reset_vector>:  13 0f 00 00        li      t5, 0
800000cc >> <reset_vector>:  93 0f 00 00        li      t6, 0
800000d0 >> <reset_vector>:  73 25 40 f1        csrr    a0, mhartid
800000d4 >> <reset_vector>:  63 10 05 00        bnez    a0, 0x800000d4 <reset_vector+0x80>
800000d8 >> <reset_vector>:  97 02 00 00        auipc   t0, 0
800000dc >> <reset_vector>:  93 82 02 01        addi    t0, t0, 16
800000e0 >> <reset_vector>:  73 90 52 30        csrw    mtvec, t0
800000e4 >> <reset_vector>:  73 50 44 74        csrwi   1860, 8
800000e8 >> <reset_vector>:  97 02 00 00        auipc   t0, 0
800000ec >> <reset_vector>:  93 82 02 01        addi    t0, t0, 16
800000f0 >> <reset_vector>:  73 90 52 30        csrw    mtvec, t0
800000f4 >> <reset_vector>:  73 50 00 18        csrwi   satp, 0
800000f8 >> <reset_vector>:  97 02 00 00        auipc   t0, 0
800000fc >> <reset_vector>:  93 82 02 02        addi    t0, t0, 32
80000100 >> <reset_vector>:  73 90 52 30        csrw    mtvec, t0
80000104 >> <reset_vector>:  93 02 f0 ff        li      t0, -1
80000108 >> <reset_vector>:  93 d2 b2 00        srli    t0, t0, 11
8000010c >> <reset_vector>:  73 90 02 3b        csrw    pmpaddr0, t0
80000110 >> <reset_vector>:  93 02 f0 01        li      t0, 31
80000114 >> <reset_vector>:  73 90 02 3a        csrw    pmpcfg0, t0
80000118 >> <reset_vector>:  73 50 40 30        csrwi   mie, 0
8000011c >> <reset_vector>:  97 02 00 00        auipc   t0, 0
80000120 >> <reset_vector>:  93 82 42 01        addi    t0, t0, 20
80000124 >> <reset_vector>:  73 90 52 30        csrw    mtvec, t0
80000128 >> <reset_vector>:  73 50 20 30        csrwi   medeleg, 0
8000012c >> <reset_vector>:  73 50 30 30        csrwi   mideleg, 0
80000130 >> <reset_vector>:  93 01 00 00        li      gp, 0
80000134 >> <reset_vector>:  97 02 00 00        auipc   t0, 0
80000138 >> <reset_vector>:  93 82 02 ed        addi    t0, t0, -304
8000013c >> <reset_vector>:  73 90 52 30        csrw    mtvec, t0
80000140 >> <reset_vector>:  13 05 10 00        li      a0, 1
80000144 >> <reset_vector>:  13 15 f5 01        slli    a0, a0, 31
80000148 >> <reset_vector>:  63 5c 05 00        bgez    a0, 0x80000160 <reset_vector+0x10c>
80000160 >> <reset_vector>:  97 02 00 80        auipc   t0, 524288
80000164 >> <reset_vector>:  93 82 02 ea        addi    t0, t0, -352
80000168 >> <reset_vector>:  63 8a 02 00        beqz    t0, 0x8000017c <reset_vector+0x128>
8000016c >> <reset_vector>:  73 90 52 10        csrw    stvec, t0
80000170 >> <reset_vector>:  b7 b2 00 00        lui     t0, 11
80000174 >> <reset_vector>:  9b 82 92 10        addiw   t0, t0, 265
80000178 >> <reset_vector>:  73 90 22 30        csrw    medeleg, t0
8000017c >> <reset_vector>:  73 50 00 30        csrwi   mstatus, 0
80000180 >> <reset_vector>:  97 02 00 00        auipc   t0, 0
80000184 >> <reset_vector>:  93 82 42 01        addi    t0, t0, 20
80000188 >> <reset_vector>:  73 90 12 34        csrw    mepc, t0
8000018c >> <reset_vector>:  73 25 40 f1        csrr    a0, mhartid
80000190 >> <reset_vector>:  73 00 20 30        mret
80000194 >> <test_2>:  93 01 20 00      li      gp, 2
80000198 >> <test_2>:  17 25 00 00      auipc   a0, 2
8000019c >> <test_2>:  13 05 c5 71      addi    a0, a0, 1820
800001a0 >> <test_2>:  ef 05 40 00      jal     a1, 0x800001a4 <test_2+0x10>
800001a4 >>   <test_2>:  33 05 b5 40    sub     a0, a0, a1
800001a8 >>   <test_2>:  b7 23 00 00    lui     t2, 2
800001ac >>   <test_2>:  9b 83 03 71    addiw   t2, t2, 1808
800001b0 >>   <test_2>:  63 14 75 02    bne     a0, t2, 0x800001d8 <fail>
800001b4 >>   <test_3>:  93 01 30 00    li      gp, 3
800001b8 >>   <test_3>:  17 e5 ff ff    auipc   a0, 1048574
800001bc >>   <test_3>:  13 05 c5 8f    addi    a0, a0, -1796
800001c0 >>   <test_3>:  ef 05 40 00    jal     a1, 0x800001c4 <test_3+0x10>
800001c4 >>     <test_3>:  33 05 b5 40          sub     a0, a0, a1
800001c8 >>     <test_3>:  b7 e3 ff ff          lui     t2, 1048574
800001cc >>     <test_3>:  9b 83 03 8f          addiw   t2, t2, -1808
800001d0 >>     <test_3>:  63 14 75 00          bne     a0, t2, 0x800001d8 <fail>
800001d8 >>     <fail>:  0f 00 f0 0f    fence
800001dc >>     <fail>:  63 80 01 00    beqz    gp, 0x800001dc <fail+0x4>
800001e0 >>     <fail>:  93 91 11 00    slli    gp, gp, 1
800001e4 >>     <fail>:  93 e1 11 00    ori     gp, gp, 1
800001e8 >>     <fail>:  93 08 d0 05    li      a7, 93
800001ec >>     <fail>:  13 85 01 00    mv      a0, gp
800001f0 >>     <fail>:  73 00 00 00    ecall
```

marcinkoss, 2023-07-11:
Proposed fix in REV code

diff --git a/include/insns/RV32I.h b/include/insns/RV32I.h
index 12fb205..2a01ac3 100644
--- a/include/insns/RV32I.h
+++ b/include/insns/RV32I.h
@@ -299,7 +299,7 @@ namespace SST{
         }else{
           SEXT(tmp, Inst.imm << 12, 32);
           R->RV64[Inst.rd] = 0x00;
-          R->RV64[Inst.rd] = (Inst.imm << 12) + dt_u64(R->RV64_PC,64);
+          R->RV64[Inst.rd] = tmp + dt_u64(R->RV64_PC,64);
           R->RV64_PC += Inst.instSize;
         }
         return true;

donofrio, 2023-07-11:
That is a good proposal - we should use that tmp value :) Fix applied in devel [https://github.com/tactcomplabs/rev/commit/24400478c1c4d12e3f3a117f4b2632925c39c1ae]

marcinkoss, 2023-07-20:
Can confirm it works at our end. Thanks

closed 2023-07-20 by marcinkoss