tactcomplabs/rev #81: two HARTs share same stack

aperdeus, 2023-07-12:
Two HARTs in REV share the same stack which is causing unexpected behaviors 

Reproducer code:
```
#include <stdlib.h>
#include <sys/mman.h>
#include "../../common/syscalls/syscalls.h"
#define assert(x) if (!(x)) { asm(".byte 0x00"); asm(".byte 0x00"); asm(".byte 0x00"); asm(".byte 0x00"); }

#define ARR_SIZE 100

int second_main () {
        int arr[ARR_SIZE];
        for(int i = ARR_SIZE-1; i >= 0; i--) {
                arr[i] = i+i;
        }

        for(int i = ARR_SIZE-1; i >= 0; i--) {
                assert(arr[i] == i+i);
        }
        return 0;
}

int main () {
        int arr[ARR_SIZE];
        for(int i = 0; i < ARR_SIZE; i++) {
                arr[i] = i*i;
        }

        for(int i = 0; i < ARR_SIZE; i++) {
                assert(arr[i] == i*i);
        }

        return 0;
}
```

REV config:
```
comp_cpu = sst.Component("cpu", "revcpu.RevCPU")
comp_cpu.addParams({
        "verbose" : verbose_level,                    # Verbosity
        "numCores" : 2,                               # Number of cores
        "clock" : "1.0GHz",                           # Clock
        "memSize" : 1024*1024*1024,                     # Memory size in bytes
        "machine" : "[CORES:RV64IMAFD]",                  # Core:Config; RV64I for core 0
        "startAddr" : ep,                             # Starting address for core 0
        "memCost" : "[0:1:10]",                       # Memory loads required 1-10 cycles
        "program" : program,                          # Target executable
        "splash" : 0,                                 # Display the splash message
        "args"   : rev_exe_args
})
```

aperdeus, 2023-08-02:
Found possible workaround by providing stack pointer address for second hard manually via assembly instruction.  

```
#include <stdlib.h>
#define assert(x) if (!(x)) { asm(".byte 0x00"); asm(".byte 0x00"); asm(".byte 0x00"); asm(".byte 0x00"); }

#define ARR_SIZE 1024

#define STACK_SIZE 1024*100
int stack[STACK_SIZE];

void __attribute__((noinline)) second() {
	int arr[ARR_SIZE];
	for(int i = ARR_SIZE-1; i >= 0; i--) {
		arr[i] = i+i;
	}

	for(int i = ARR_SIZE-1; i >= 0; i--) {
		assert(arr[i] == i+i);
	}
}

void fix_stack () {
	asm ("mv sp, %0" :: "r" (stack+STACK_SIZE-1));
	second();
	return;
}

int main () {
	int arr[ARR_SIZE];
	for(int i = 0; i < ARR_SIZE; i++) {
		arr[i] = i*i;
	}

	for(int i = 0; i < ARR_SIZE; i++) {
		assert(arr[i] == i*i);
	}
	return 0;
}
```

rkabrick, 2023-10-01:
this should be fixed now. Can you confirm?

closed 2023-10-25 by jleidel