tactcomplabs/rev #82: BEQ implementation is incorrect

marcinkoss, 2023-07-13:
Use LLVM toolchain, build and run below code snippet:
```
#define assert(x) if (!(x)) { asm(".dword 0x00000000"); }

#include <stdio.h>

int main() {
        char buf[] = { 'a', 'a', 0 };
        char dest[3];
        int read_count;
        int n = sscanf(buf, "%s%n", dest, &read_count);

        return 0;
}
```

Actual result is:
`FATAL: RevCPU[cpu:DecodeCompressed:1089000]: Error: failed to decode instruction at PC=0x10284; Compressed instructions not enabled!`

Last instructions executed:
```
11b64 >>   <__copy_tls>:  03 39 01 01   ld      s2, 16(sp)
11b68 >>   <__copy_tls>:  83 39 81 00   ld      s3, 8(sp)
11b6c >>   <__copy_tls>:  03 3a 01 00   ld      s4, 0(sp)
11b70 >>   <__copy_tls>:  13 01 01 03   addi    sp, sp, 48
11b74 >>   <__copy_tls>:  67 80 00 00   ret
```

Issue here is that ret jumps to PC which is incorrect, but reason for that is earlier in execution when BEQZ instruction jumps into wrong address (from 0x12118 to 0x11AF8):

```
12110 >>   <vfscanf>:  03 45 1a 00      lbu     a0, 1(s4)
12114 >>   <vfscanf>:  13 0a 1a 00      addi    s4, s4, 1
12118 >>   <vfscanf>:  e3 00 05 1e      beqz    a0, 0x12af8 <vfscanf+0xad4>
11af8 >>   <__copy_tls>:  33 04 e5 00   add     s0, a0, a4
11afc >>   <__copy_tls>:  63 08 09 04   beqz    s2, 0x11b4c <__copy_tls+0xa8>
11b00 >>   <__copy_tls>:  93 09 84 0c   addi    s3, s0, 200
```

marcinkoss, 2023-07-13:
This is most likely caused by wrong implementation of BEQ instruction as for BEQ IMM has 13 bits instead of 12, so instead of:
`R->RV64_PC = R->RV64_PC + (int64_t)(td_u64(Inst.imm,12));`
should be
`R->RV64_PC = R->RV64_PC + (int64_t)(td_u64(Inst.imm,13));`

Most likely other instructions have similar bugs. Please check & fix other instructions as well

donofrio, 2023-07-14:
Confirmed - that should be a 13bit immd for BEQ as well as other B-type instructions.  Thanks for the detailed bug report. All branch instructions corrected in devel 9a884a7.

marcinkoss, 2023-07-20:
Confirmed - works fine with that change. Thanks.

closed 2023-07-20 by marcinkoss