tactcomplabs/rev #83: [P3] Double type comparision does not work correctly

ukasz, 2023-07-17:
On double data type we see less and greater operators not working correctly, see example below:
```
#define assert(x) if (!(x)) { asm(".dword 0x00000000"); }

int main() {
        volatile double da = 0.199;
        volatile double db = 0.2;

        assert(da < db);  // fails
        assert(da <= db); // ok
        assert(db > da);  // fails
        assert(db >= da); // ok
        da = 0.2;
        assert(da == db); // ok
}
```

On long double data type only the equal operator works correctly.

leekillough, 2023-08-09:
So far:
1. The comparison instruction simulation was comparing `rs1` to itself, instead of to `rs2`.
       `-`          `if( R->DPF[Inst.rs1] < R->DPF[Inst.rs1] ){`
         `+`          `if( R->DPF[Inst.rs1] < R->DPF[Inst.rs2] ){`

3. The test case fails at `-O0`, but passes at `-O2`, once `#1` is fixed. At `-O0`, both `rs1` and `rs2` are `0.0`, which means the test passes if it allows for `0.0 == 0.0` (`<=` `>=` and `==`), but fails otherwise.

ukasz, 2023-08-09:
I changed rs1 to rs2 in three places as you proposed and this fixes double data type. Problem with long double still exists after that, but as far as I understand long doubles are emulated on riscv, so we must be running into a different bug there if that is not working correctly.

leekillough, 2023-08-10:
For some reason when I use your example test case and it's compiled at `-O0`, both `rs1` and `rs2` read as `0.0` when the instruction simulation function is called. At `-O2`, it goes away.

I am working on a refactoring fix which will make the code much easier to read and maintain.

I'll look at the `long double` issue later. (It would be legal if `long double` `==` `double` on RISC-V, since C only requires `long double >= double >= float`.

leekillough, 2023-09-19:
Fixed by #115

closed 2023-09-19 by leekillough