tactcomplabs/rev #86: [P1] Program execution exits prematurely.

marcinkoss, 2023-07-25:
Use LLVM toolchain, build & compile below code snippet.
```
CXX=$(TIGRE_TOOLCHAIN_INSTALL)/bin/clang++
ARCH=rv64imafd
$(CXX) -march=$(ARCH) -O0 -o program.exe
```

```
#include <iostream>

using namespace std;

int vd  =5;
int main() {

  shared_ptr<int> p= make_shared<int>(vd);
  cout << std::endl;

  return 0;
}
```

Result: execution of program reaches 'jalr' instruction in which it jumps to address 0x0 and stop running. This is not issue with jalr instruction itself but for some reason memory content is corrupted and data read from stack are not correct which results into jalr behaviour.

marcinkoss, 2023-07-25:
Attached logs (objdump and parsed log file)
[logs.zip](https://github.com/tactcomplabs/rev/files/12158925/logs.zip)

marcinkoss, 2023-07-25:
ee attached image. Value popped from stack is equal to addi/sb instructions.
[image]

ukasz, 2023-09-23:
Tested on devel, no longer an issue and can be closed.

closed 2023-09-23 by marcinkoss