tactcomplabs/rev #91: munmap test fails on clang

ukasz, 2023-08-04:
We get the following error while running munmap test compiled with clang, REVsha: @051e16413c

`FATAL: RevCPU[cpu:InitStaticMem:0]: Loader Error: Attempting to initialize static memory however __BSS_END__ = 0x0
SST Fatal Backtrace Information:
    0 : sst(_ZNK3SST6Output5fatalEjPKcS2_iS2_z+0x4ae) [0x55da8c65744e]
    1 : /home/dev/riscv/off/rev3/rev4/src/librevcpu.so(_ZN3SST6RevCPU9RevLoader13InitStaticMemEv+0xe2) [0x7fa58ec7ca32]
    2 : /home/dev/riscv/off/rev3/rev4/src/librevcpu.so(_ZN3SST6RevCPU9RevLoaderC2ENSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEES7_PNS0_6RevMemEPNS_6OutputE+0xfb) [0x7fa58ec7e04b]
    3 : /home/dev/riscv/off/rev3/rev4/src/librevcpu.so(_ZN3SST6RevCPU6RevCPUC1EmRNS_6ParamsE+0x12de) [0x7fa58ec63a5e]
    4 : /home/dev/riscv/off/rev3/rev4/src/librevcpu.so(_ZN3SST3ELI14DerivedBuilderINS_6RevCPU6RevCPUENS_9ComponentEJmRNS_6ParamsEEE6createEmS6_+0x2a) [0x7fa58ec7310a]
    5 : sst(_ZN3SST7Factory15CreateComponentEmRKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEERNS_6ParamsE+0x403) [0x55da8c642bb3]
    6 : sst(_ZN3SST15Simulation_impl13performWireUpERNS_11ConfigGraphERKNS_8RankInfoEm+0x8e) [0x55da8c682e0e]
    7 : sst(+0xea62e) [0x55da8c5d762e]
    8 : sst(main+0x18c3) [0x55da8c5bede3]
    9 : /lib/x86_64-linux-gnu/libc.so.6(+0x29d90) [0x7fa5954aed90]
   10 : /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0x80) [0x7fa5954aee40]
   11 : sst(_start+0x25) [0x55da8c5d4da5]
`

The thing we had to modify was changing NULL to 0 i rev_mmap call.
`
diff --git a/test/syscalls/munmap/munmap.c b/test/syscalls/munmap/munmap.c
index 4ca017f..f0f4fe9 100644
--- a/test/syscalls/munmap/munmap.c
+++ b/test/syscalls/munmap/munmap.c
@@ -5,7 +5,7 @@ int main() {
   uint32_t *addr;

   // Create an anonymous memory mapping
-  addr = rev_mmap(NULL,                 // Let rev choose the address
+  addr =(uint32_t*) rev_mmap(0,                 // Let rev choose the address
               237 * sizeof(uint32_t),
               PROT_READ | PROT_WRITE | PROT_EXEC, // RWX permissions
               MAP_PRIVATE | MAP_ANONYMOUS, // Not shared, anonymous
`

rkabrick, 2023-08-04:
To clarify: The test passes once you changed NULL to 0?

ukasz, 2023-08-04:
No I wouldn't bother you with such issue, changing NULL to 0 is needed to compile the test because clang is more picky about types.

ukasz, 2023-08-05:
Update, this is not munmap specific all tests seems to be failing due to the same issue - log with BSS_END

jleidel, 2023-08-05:
please post logs

ukasz, 2023-08-06:
```
root@lxp4d:/home/dev/riscv/rev/test/ex2# ./run_ex2.sh
clang -march=rv64imafd -o ex2.exe ex2.c -static
WARNING: Building component "cpu" with no links assigned.
RevCPU[cpu:RevCPU:0]: Building Rev with 1 cores
RevCPU[cpu:LoadElf:0]: Symbol Table Entry [:0x11f04]
RevCPU[cpu:LoadElf:0]: Symbol Table Entry [_DYNAMIC:0x0]
RevCPU[cpu:LoadElf:0]: Symbol Table Entry [_Exit:0x119cc]
RevCPU[cpu:LoadElf:0]: Symbol Table Entry [_Exit.c:0x0]
RevCPU[cpu:LoadElf:0]: Symbol Table Entry [___environ:0x12f10]
RevCPU[cpu:LoadElf:0]: Symbol Table Entry [__copy_tls:0x115b8]
RevCPU[cpu:LoadElf:0]: Symbol Table Entry [__default_guardsize:0x12f08]
RevCPU[cpu:LoadElf:0]: Symbol Table Entry [__default_stacksize:0x12f04]
RevCPU[cpu:LoadElf:0]: Symbol Table Entry [__environ:0x12f10]
RevCPU[cpu:LoadElf:0]: Symbol Table Entry [__environ.c:0x0]
RevCPU[cpu:LoadElf:0]: Symbol Table Entry [__fini_array_end:0x1117c]
RevCPU[cpu:LoadElf:0]: Symbol Table Entry [__fini_array_start:0x1117c]
RevCPU[cpu:LoadElf:0]: Symbol Table Entry [__funcs_on_exit:0x11958]
RevCPU[cpu:LoadElf:0]: Symbol Table Entry [__global_pointer$:0x13704]
RevCPU[cpu:LoadElf:0]: Symbol Table Entry [__hwcap:0x12f38]
RevCPU[cpu:LoadElf:0]: Symbol Table Entry [__init_array_end:0x1117c]
RevCPU[cpu:LoadElf:0]: Symbol Table Entry [__init_array_start:0x1117c]
RevCPU[cpu:LoadElf:0]: Symbol Table Entry [__init_libc:0x11214]
RevCPU[cpu:LoadElf:0]: Symbol Table Entry [__init_ssp:0x11210]
RevCPU[cpu:LoadElf:0]: Symbol Table Entry [__init_tls:0x1168c]
RevCPU[cpu:LoadElf:0]: Symbol Table Entry [__init_tls.c:0x0]
RevCPU[cpu:LoadElf:0]: Symbol Table Entry [__init_tp:0x11528]
RevCPU[cpu:LoadElf:0]: Symbol Table Entry [__libc:0x130c0]
RevCPU[cpu:LoadElf:0]: Symbol Table Entry [__libc_exit_fini:0x1195c]
RevCPU[cpu:LoadElf:0]: Symbol Table Entry [__libc_start_init:0x11434]
RevCPU[cpu:LoadElf:0]: Symbol Table Entry [__libc_start_main:0x11480]
RevCPU[cpu:LoadElf:0]: Symbol Table Entry [__libc_start_main.c:0x0]
RevCPU[cpu:LoadElf:0]: Symbol Table Entry [__progname:0x12f28]
RevCPU[cpu:LoadElf:0]: Symbol Table Entry [__progname_full:0x12f30]
RevCPU[cpu:LoadElf:0]: Symbol Table Entry [__set_thread_area:0x11ef8]
RevCPU[cpu:LoadElf:0]: Symbol Table Entry [__stdio_exit:0x11958]
RevCPU[cpu:LoadElf:0]: Symbol Table Entry [__sysinfo:0x12f20]
RevCPU[cpu:LoadElf:0]: Symbol Table Entry [__thread_list_lock:0x12f18]
RevCPU[cpu:LoadElf:0]: Symbol Table Entry [_environ:0x12f10]
RevCPU[cpu:LoadElf:0]: Symbol Table Entry [_fini:0x11958]
RevCPU[cpu:LoadElf:0]: Symbol Table Entry [_init:0x1120c]
RevCPU[cpu:LoadElf:0]: Symbol Table Entry [_start:0x1117c]
RevCPU[cpu:LoadElf:0]: Symbol Table Entry [_start_c:0x11198]
RevCPU[cpu:LoadElf:0]: Symbol Table Entry [builtin_tls:0x12f70]
RevCPU[cpu:LoadElf:0]: Symbol Table Entry [crt1.c:0x0]
RevCPU[cpu:LoadElf:0]: Symbol Table Entry [crti.c:0x0]
RevCPU[cpu:LoadElf:0]: Symbol Table Entry [crtn.c:0x0]
RevCPU[cpu:LoadElf:0]: Symbol Table Entry [default_attr.c:0x0]
RevCPU[cpu:LoadElf:0]: Symbol Table Entry [defsysinfo.c:0x0]
RevCPU[cpu:LoadElf:0]: Symbol Table Entry [dummy:0x11958]
RevCPU[cpu:LoadElf:0]: Symbol Table Entry [dummy1:0x11210]
RevCPU[cpu:LoadElf:0]: Symbol Table Entry [environ:0x12f10]
RevCPU[cpu:LoadElf:0]: Symbol Table Entry [ex2.c:0x0]
RevCPU[cpu:LoadElf:0]: Symbol Table Entry [exit:0x119a8]
RevCPU[cpu:LoadElf:0]: Symbol Table Entry [exit.c:0x0]
RevCPU[cpu:LoadElf:0]: Symbol Table Entry [libc.c:0x0]
RevCPU[cpu:LoadElf:0]: Symbol Table Entry [libc_exit_fini:0x1195c]
RevCPU[cpu:LoadElf:0]: Symbol Table Entry [libc_start_init:0x11434]
RevCPU[cpu:LoadElf:0]: Symbol Table Entry [libc_start_main_stage2:0x114e0]
RevCPU[cpu:LoadElf:0]: Symbol Table Entry [main:0x111c0]
RevCPU[cpu:LoadElf:0]: Symbol Table Entry [main_tls:0x12f40]
RevCPU[cpu:LoadElf:0]: Symbol Table Entry [memcpy:0x119e8]
RevCPU[cpu:LoadElf:0]: Symbol Table Entry [memcpy.c:0x0]
RevCPU[cpu:LoadElf:0]: Symbol Table Entry [memset:0x11dfc]
RevCPU[cpu:LoadElf:0]: Symbol Table Entry [memset.c:0x0]
RevCPU[cpu:LoadElf:0]: Symbol Table Entry [program_invocation_name:0x12f30]
RevCPU[cpu:LoadElf:0]: Symbol Table Entry [program_invocation_short_name:0x12f28]
RevCPU[cpu:LoadElf:0]: Symbol Table Entry [static_init_tls:0x1168c]
RevCPU[cpu:LoadProgramArgs:0]: Loading program argv[0] = ex2.exe
FATAL: RevCPU[cpu:InitStaticMem:0]: Loader Error: Attempting to initialize static memory however __BSS_END__ = 0x0
SST Fatal Backtrace Information:
    0 : sst(_ZNK3SST6Output5fatalEjPKcS2_iS2_z+0x4ae) [0x55619c3a144e]
    1 : ../../src/librevcpu.so(_ZN3SST6RevCPU9RevLoader13InitStaticMemEv+0xe2) [0x7f6944c51a32]
    2 : ../../src/librevcpu.so(_ZN3SST6RevCPU9RevLoaderC2ENSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEES7_PNS0_6RevMemEPNS_6OutputE+0xfb) [0x7f6944c5304b]
    3 : ../../src/librevcpu.so(_ZN3SST6RevCPU6RevCPUC1EmRNS_6ParamsE+0x12de) [0x7f6944c38a5e]
    4 : ../../src/librevcpu.so(_ZN3SST3ELI14DerivedBuilderINS_6RevCPU6RevCPUENS_9ComponentEJmRNS_6ParamsEEE6createEmS6_+0x2a) [0x7f6944c4810a]
    5 : sst(_ZN3SST7Factory15CreateComponentEmRKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEERNS_6ParamsE+0x403) [0x55619c38cbb3]
    6 : sst(_ZN3SST15Simulation_impl13performWireUpERNS_11ConfigGraphERKNS_8RankInfoEm+0x8e) [0x55619c3cce0e]
    7 : sst(+0xea62e) [0x55619c32162e]
    8 : sst(main+0x18c3) [0x55619c308de3]
    9 : /lib/x86_64-linux-gnu/libc.so.6(+0x29d90) [0x7f694746cd90]
   10 : /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0x80) [0x7f694746ce40]
   11 : sst(_start+0x25) [0x55619c31eda5]
--------------------------------------------------------------------------
MPI_ABORT was invoked on rank 0 in communicator MPI_COMM_WORLD
with errorcode 99.

NOTE: invoking MPI_ABORT causes Open MPI to kill all MPI processes.
You may or may not see output from other processes, depending on
exactly when Open MPI kills them.
```

jleidel, 2023-08-07:
what branch of Rev, what commit tag, what commit tag of Clang?

jleidel, 2023-08-07:
please post your linker script.  the symbol output looks like a mixture of static and dynamic linking

ukasz, 2023-08-07:
It was devel REV branch a711f2a3d0  and 8354fb7b3 commit of our clang, but it is also reproducible on newest commit: d2ebe164a3. Have you tried using our toolchain to reproduce this issue (cd scripts ./build_toolchain.sh -s and then add toolchain/llvm-cross-compiler-sst/bin to PATH, rebuild ex2 test and run) ?

jleidel, 2023-08-08:
Yes, all tests are executed against GCC 12.2.0 and your Clang compiler.

jleidel, 2023-08-08:
rebuilt your compiler from d2ebe164a3d4236491a5945723b7288b51138508 (HEAD on 2023/08/08).  Retested against `rev:devel` @ a711f2a3d07782f2656e5d99f11c7299d8657507.  

Test results:
```
100% tests passed, 0 tests failed out of 85

Label Time Summary:
all              = 170.21 sec*proc (25 tests)
blas-required    =   0.00 sec*proc (2 tests)
rv32             =  20.72 sec*proc (5 tests)
rv64             = 149.48 sec*proc (20 tests)

Total Test time (real) = 418.49 sec

The following tests did not run:
	 10 - TEST_DOT_SINGLE (Disabled)
	 11 - TEST_DOT_DOUBLE (Disabled)
```

jleidel, 2023-08-25:
fixed on devel

closed 2023-08-25 by jleidel