tactcomplabs/rev #92: [P1] munmap test fails

ukasz, 2023-08-04:
Current mmap/munmap test doesn't check for validity of data, once we add that it fails - see loop with an assert.
**Compiled with: riscv64-linux-gnu-gcc**
```
#include "../../../common/syscalls/syscalls.h"
#include "unistd.h"

#define assert(x) if (!(x)) { asm(".byte 0x00"); asm(".byte 0x00"); asm(".byte 0x00"); asm(".byte 0x00"); }

int main() {
  uint32_t *addr;
#define N 237

  // Create an anonymous memory mapping
  addr =(uint32_t*) rev_mmap(0,                 // Let rev choose the address
              N * sizeof(uint32_t),
              PROT_READ | PROT_WRITE | PROT_EXEC, // RWX permissions
              MAP_PRIVATE | MAP_ANONYMOUS, // Not shared, anonymous
              -1,                   // No file descriptor because it's an anonymous mapping
              0);                   // No offset, irrelevant for anonymous mappings

  for( uint32_t i=0; i< N; i++ ){
    addr[i] = i;
  }
  for (uint32_t i = 0; i < N ; i++)
        assert(addr[i] == i);
```

rkabrick, 2023-08-04:
While this is **not a fix** could you try replacing all `uint32_t` in the file with `uint64_t` and see if that works? I was able to recreate the issue when running the code you provided, however, when I replace everything and make it 64-bit the test ran without issue. 

I will continue to investigate what is happening with the 32-bit version.

leekillough, 2023-09-19:
I cannot reproduce it after #115, which had a lot of templating to make code more generic w.r.t 32/64-bit.

ukasz, 2023-09-25:
Fix confirmed at @1b330d6a44

closed 2023-09-25 by ukasz