tactcomplabs/rev #95: [P1] fp clasification errors for double

ukasz, 2023-08-10:
Classification as infinity does not work correctly on double data type, tested with float works fine.
Tested with the patch that I think solves #83
Below is the C code snipped:
```
#include <cmath>
#include <cfloat>

#define assert(x) if (!(x)) { asm(".byte 0x0; .byte 0x0; .byte 0x0; .byte 0x00"); }
using namespace std;

int main () {
        volatile double x;
        x = INFINITY;
        assert(isinf(x) == true); // fail
        x = -INFINITY;
        assert(isinf(x) == true); // fail
        return 0;
}
```
LLVM-generate assembly (unknown instruction is just an assert)
```
00000000000111c0 <main>:
   111c0: 13 01 01 fe   addi    sp, sp, -32
   111c4: 23 3c 11 00   sd      ra, 24(sp)
   111c8: 23 38 81 00   sd      s0, 16(sp)
   111cc: 13 04 01 02   addi    s0, sp, 32
   111d0: 13 05 00 00   li      a0, 0
   111d4: 23 26 a4 fe   sw      a0, -20(s0)
   111d8: 13 05 f0 7f   li      a0, 2047
   111dc: 13 15 45 03   slli    a0, a0, 52
   111e0: 23 30 a4 fe   sd      a0, -32(s0)
   111e4: 07 35 04 fe   fld     fa0, -32(s0)
   111e8: ef 00 c0 04   jal     0x11234 <_Z5isinfB7v160000Ua9enable_ifILb1EEd>
   111ec: 63 18 05 00   bnez    a0, 0x111fc <main+0x3c>
   111f0: 6f 00 40 00   j       0x111f4 <main+0x34>
   111f4: 00 00         <unknown>
   111f6: 00 00         <unknown>
   111f8: 6f 00 40 00   j       0x111fc <main+0x3c>
   111fc: 13 05 f0 ff   li      a0, -1
   11200: 13 15 45 03   slli    a0, a0, 52
   11204: 23 30 a4 fe   sd      a0, -32(s0)
   11208: 07 35 04 fe   fld     fa0, -32(s0)
   1120c: ef 00 80 02   jal     0x11234 <_Z5isinfB7v160000Ua9enable_ifILb1EEd>
   11210: 63 18 05 00   bnez    a0, 0x11220 <main+0x60>
   11214: 6f 00 40 00   j       0x11218 <main+0x58>
   11218: 00 00         <unknown>
   1121a: 00 00         <unknown>
   1121c: 6f 00 40 00   j       0x11220 <main+0x60>
   11220: 13 05 00 00   li      a0, 0
   11224: 83 30 81 01   ld      ra, 24(sp)
   11228: 03 34 01 01   ld      s0, 16(sp)
   1122c: 13 01 01 02   addi    sp, sp, 32
   11230: 67 80 00 00   ret

0000000000011234 <_Z5isinfB7v160000Ua9enable_ifILb1EEd>:
   11234: 13 01 01 fe   addi    sp, sp, -32
   11238: 23 3c 11 00   sd      ra, 24(sp)
   1123c: 23 38 81 00   sd      s0, 16(sp)
   11240: 13 04 01 02   addi    s0, sp, 32
   11244: 27 34 a4 fe   fsd     fa0, -24(s0)
   11248: 07 30 84 fe   fld     ft0, -24(s0)
   1124c: 53 20 00 22   fabs.d  ft0, ft0
   11250: 37 35 01 00   lui     a0, 19
   11254: 87 30 85 f6   fld     ft1, -152(a0)
   11258: 53 25 10 a2   feq.d   a0, ft0, ft1
   1125c: 83 30 81 01   ld      ra, 24(sp)
   11260: 03 34 01 01   ld      s0, 16(sp)
   11264: 13 01 01 02   addi    sp, sp, 32
   11268: 67 80 00 00   ret
```

Looking briefly at the dump suspected instructions are fabs.d or feq.d.
Please let us know if you are able to reproduce it.

leekillough, 2023-08-10:
I'm working on a general refactoring of floating-point routines which should fix a lot of errors and simplify the code. For example, `fsgnjxd` _copies_ the sign of the second operand to the first, instead of XORing it.

ukasz, 2023-08-10:
Thanks for the notification, I am looking forward to test it.

leekillough, 2023-08-10:
[Preview](https://github.com/tactcomplabs/rev/compare/devel...leekillough:rev:dcmp?expand=1) (work is still in progress).

leekillough, 2023-09-19:
Fixed by https://github.com/tactcomplabs/rev/pull/115

closed 2023-09-19 by leekillough