tactcomplabs/rev #97: [P1] write syscall + memh = segfault

ukasz, 2023-08-11:
Running rev_write test example with memory hierarchy config from memh_2 test leads to a segfault. Other memory accessing syscalls are likely affected as well.
```
RevCPU[cpu:ECALL_write:516000]: ECALL_write called
[user-x] *** Process received signal ***
[user-x] Signal: Segmentation fault (11)
[user-x] Signal code: Address not mapped (1)
[user-x] Failing at address: 0x3fec0
[user-x] [ 0] /lib/x86_64-linux-gnu/libc.so.6(+0x42520)[0x7fe28b5df520]
[user-x] [ 1] /home/dev/tigre/frameworks.simulators.agile.rev/src/librevcpu.so(_ZN3SST6RevCPU6RevMem7ReadMemEmmPv+0xc0)[0x7fe2817b3fc0]
[user-x] [ 2] /home/dev/tigre/frameworks.simulators.agile.rev/src/librevcpu.so(_ZN3SST6RevCPU7RevProc11ECALL_writeEv+0xd8)[0x7fe2817d39d8]
[user-x] [ 3] /home/dev/tigre/frameworks.simulators.agile.rev/src/librevcpu.so(_ZN3SST6RevCPU7RevProc9ExecEcallEv+0xb9)[0x7fe2817d6459]
[user-x] [ 4] /home/dev/tigre/frameworks.simulators.agile.rev/src/librevcpu.so(_ZN3SST6RevCPU7RevProc9ClockTickEm+0xc9d)[0x7fe2817dc09d]
[user-x] [ 5] /home/dev/tigre/frameworks.simulators.agile.rev/src/librevcpu.so(_ZN3SST6RevCPU6RevCPU9clockTickEm+0xa6)[0x7fe2817924c6]
[user-x] [ 6] sst(_ZN3SST5Clock7executeEv+0x10c)[0x55c7eb5c59cc]
[user-x] [ 7] sst(_ZN3SST15Simulation_impl3runEv+0x3d3)[0x55c7eb65e4d3]
[user-x] [ 8] sst(+0xf808e)[0x55c7eb5a808e]
[user-x] [ 9] sst(main+0x186b)[0x55c7eb58de0b]
[user-x] [10] /lib/x86_64-linux-gnu/libc.so.6(+0x29d90)[0x7fe28b5c6d90]
[user-x] [11] /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0x80)[0x7fe28b5c6e40]
[user-x] [12] sst(_start+0x25)[0x55c7eb5a54b5]
[user-x] *** End of error message ***
Segmentation fault (core dumped)
```

jleidel, 2023-08-30:
syscalls do not currently support memHierarchy

jleidel, 2023-09-06:
Fixes for RV64 are present in 0506826 and https://github.com/tactcomplabs/rev/pull/113

ukasz, 2023-09-28:
I can confirm that I no longer see the segfault, but there is a different issue. The output between configuration with memh and without differs. I used configs from tests/write and tested rev devel commit d3b9a48a1e1f.
When basic memory is used output looks fine, but with memh it is chopped of and mixed up a bit.
To clean up logs I set up verbose to 1 in both configs.
Below is the log difference.
```
# REV_EXE=write.exe sst rev-test.py
WARNING: Building component "cpu" with no links assigned.
RevCPU[cpu:RevCPU:0]: Building Rev with 1 cores
RevCPU[cpu:RevCPU:0]:
*******
/**////**
/**   /**   *****  **    **
/*******   **///**/**   /**
/**///**  /*******//** /**
/**  //** /**////  //****
/**   //**//******  //**
//     //  //////    //

RevCPU[cpu:RevCPU:0]: Initialization of RevCPUs complete.
Greetings
Greetings - this is a much longer text string. Just larger than 64
 cpu.TotalCycles.core_0 : Accumulator : Sum.u64 = 116; SumSQ.u64 = 13456; Count.u64 = 1; Min.u64 = 116; Max.u64 = 116;
 cpu.CyclesWithIssue.core_0 : Accumulator : Sum.u64 = 113; SumSQ.u64 = 12769; Count.u64 = 1; Min.u64 = 113; Max.u64 = 113;
 cpu.FloatsRead.core_0 : Accumulator : Sum.u64 = 0; SumSQ.u64 = 0; Count.u64 = 1; Min.u64 = 0; Max.u64 = 0;
 cpu.FloatsWritten.core_0 : Accumulator : Sum.u64 = 0; SumSQ.u64 = 0; Count.u64 = 1; Min.u64 = 0; Max.u64 = 0;
 cpu.DoublesRead.core_0 : Accumulator : Sum.u64 = 0; SumSQ.u64 = 0; Count.u64 = 1; Min.u64 = 0; Max.u64 = 0;
 cpu.DoublesWritten.core_0 : Accumulator : Sum.u64 = 0; SumSQ.u64 = 0; Count.u64 = 1; Min.u64 = 0; Max.u64 = 0;
 cpu.BytesRead.core_0 : Accumulator : Sum.u64 = 762; SumSQ.u64 = 580644; Count.u64 = 1; Min.u64 = 762; Max.u64 = 762;
 cpu.BytesWritten.core_0 : Accumulator : Sum.u64 = 22751; SumSQ.u64 = 517608001; Count.u64 = 1; Min.u64 = 22751; Max.u64 = 22751;
 cpu.FloatsExec.core_0 : Accumulator : Sum.u64 = 0; SumSQ.u64 = 0; Count.u64 = 1; Min.u64 = 0; Max.u64 = 0;
 cpu.TLBHitsPerCore.core_0 : Accumulator : Sum.u64 = 59; SumSQ.u64 = 3481; Count.u64 = 1; Min.u64 = 59; Max.u64 = 59;
 cpu.TLBMissesPerCore.core_0 : Accumulator : Sum.u64 = 492; SumSQ.u64 = 242064; Count.u64 = 1; Min.u64 = 492; Max.u64 = 492;
Simulation is complete, simulated time: 116 ns

# REV_EXE=write.exe sst rev-test-memh.py
RevCPU[cpu:RevCPU:0]: Building Rev with 1 cores
RevCPU[cpu:RevCPU:0]:
*******
/**////**
/**   /**   *****  **    **
/*******   **///**/**   /**
/**///**  /*******//** /**
/**  //** /**////  //****
/**   //**//******  //**
//     //  //////    //

RevCPU[cpu:RevCPU:0]: Initialization of RevCPUs complete.
Grs
eetingGreetings - this is a much longer text string. Just larger6Simulation is complete, simulated time: 3.748 us
```

rkabrick, 2023-10-26:
@ukasz Can you confirm that this was fixed post `syscall-fixes` being merged?

ukasz, 2023-10-27:
Sorry I am not able to confirm that today. I see a segfault on mmap with rev-test.py and write on rev-devel branch and I am not sure if this is something on my end, because I assume you dont see it in CI.
From a brief investigation application keeps executing the last instuction of main function which happens to be ret.
```
<snip>
   174dc: 6f 00 40 00   j       0x174e0 <main+0x96>
   174e0: 01 45         li      a0, 0
   174e2: aa 60         ld      ra, 136(sp)
   174e4: 0a 64         ld      s0, 128(sp)
   174e6: 49 61         addi    sp, sp, 144
   174e8: 82 80         ret
   174ea: 00 00         unimp
```

```
<snip>
RevCPU[cpu:DecodeInst:6089500]: Core 0; Hart 0; Thread 1; PC:InstPayload = 0x174e8:0x8082
RevCPU[cpu:DecodeInst:6090000]: Core 0; Hart 0; Thread 1; PC:InstPayload = 0x174e8:0x8082
RevCPU[cpu:DecodeInst:6090500]: Core 0; Hart 0; Thread 1; PC:InstPayload = 0x174e8:0x8082
RevCPU[cpu:DecodeInst:6091000]: Core 0; Hart 0; Thread 1; PC:InstPayload = 0x174e8:0x8082
RevCPU[cpu:DecodeInst:6091500]: Core 0; Hart 0; Thread 1; PC:InstPayload = 0x174e8:0x8082
RevCPU[cpu:DecodeInst:6092000]: Core 0; Hart 0; Thread 1; PC:InstPayload = 0x174e8:0x8082
RevCPU[cpu:DecodeInst:6092500]: Core 0; Hart 0; Thread 1; PC:InstPayload = 0x174e8:0x8082
RevCPU[cpu:DecodeInst:6093000]: Core 0; Hart 0; Thread 1; PC:InstPayload = 0x174e8:0x8082
RevCPU[cpu:DecodeInst:6093500]: Core 0; Hart 0; Thread 1; PC:InstPayload = 0x174e8:0x8082
RevCPU[cpu:DecodeInst:6094000]: Core 0; Hart 0; Thread 1; PC:InstPayload = 0x174e8:0x8082
RevCPU[cpu:DecodeInst:6094500]: Core 0; Hart 0; Thread 1; PC:InstPayload = 0x174e8:0x8082
 | BaseAddr:  0x3ffffbff | TopAddr: 0x3fffffff | Size: 1024 Bytes
 | BaseAddr:  0x10040 | TopAddr: 0x11238 | Size: 4600 Bytes
 | BaseAddr:  0x10000 | TopAddr: 0x14000 | Size: 16384 Bytes
 | BaseAddr:  0x14054 | TopAddr: 0x19054 | Size: 20480 Bytes
 | BaseAddr:  0x19278 | TopAddr: 0x1a278 | Size: 4096 Bytes
 | BaseAddr:  0x0 | TopAddr: 0x1000 | Size: 4096 Bytes
 | BaseAddr:  0x3feffbff | TopAddr: 0x3ffffbff | Size: 1048576 Bytes
FATAL: RevCPU[cpu:CalcPhysAddr:6095000]: Segmentation Fault: Virtual address 0x74207265676e6f6c (PhysAddr = 0xffffffffffffffff) was not found in any mem segments
SST Fatal Backtrace Information:
    0 : sst(_ZNK3SST6Output5fatalEjPKcS2_iS2_z+0x4ae) [0x5562655e1c6e]
    1 : /usr/local/lib/sst-elements-library/librevcpu.so(_ZN3SST6RevCPU6RevMem12CalcPhysAddrEmm+0x465) [0x7ff96046c0a5]
    2 : /usr/local/lib/sst-elements-library/librevcpu.so(_ZN3SST6RevCPU6RevMem7ReadMemEjmmPvRKNS0_6MemReqEj+0x3e) [0x7ff96046ce6e]
    3 : /usr/local/lib/sst-elements-library/librevcpu.so(_ZN3SST6RevCPU13RevPrefetcher4FillEm+0x4db) [0x7ff9604d5d9b]
    4 : /usr/local/lib/sst-elements-library/librevcpu.so(_ZN3SST6RevCPU13RevPrefetcher7IsAvailEm+0x67) [0x7ff9604d6157]
    5 : /usr/local/lib/sst-elements-library/librevcpu.so(_ZN3SST6RevCPU7RevProc9ClockTickEm+0x725) [0x7ff9604a5755]
    6 : /usr/local/lib/sst-elements-library/librevcpu.so(_ZN3SST6RevCPU6RevCPU9clockTickEm+0x183) [0x7ff960434e83]
    7 : sst(_ZN3SST5Clock7executeEv+0x10c) [0x5562655729cc]
    8 : sst(_ZN3SST15Simulation_impl3runEv+0x3d3) [0x55626560b4d3]
    9 : sst(+0xf808e) [0x55626555508e]
   10 : sst(main+0x186b) [0x55626553ae0b]
   11 : /lib/x86_64-linux-gnu/libc.so.6(+0x29d90) [0x7ff96469cd90]
   12 : /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0x80) [0x7ff96469ce40]
   13 : sst(_start+0x25) [0x5562655524b5]
```

If you can confirm that you don't see any fails on write and munmap tests at e1da56a2 I'll try to debug it further.
My cmdline is just:
```
REV_EXE=write.exe sst rev-test.py
```

rkabrick, 2023-10-28:
I see the failure when using your compiler. I'll debug Monday.

leekillough, 2024-02-27:
Fixed by #244

closed 2024-02-27 by leekillough