tactcomplabs/rev #101: [P0] Various tests fails with 'segmentation fault'

marcinkoss, 2023-08-17:
compile and run below program with LLVM toolchain:
```
#include <cmath>
#define assert(x) if (!(x)) { asm(".byte 0x00"); asm(".byte 0x00"); asm(".byte 0x00"); asm(".byte 0x00"); }

int main() {
        pow(2,8);
        return 0;
}
```

Result:
```
FATAL: RevCPU[cpu:CalcPhysAddr:137000]: Segmentation Fault: Virtual address 0x3fe0000000000048 (PhysAddr = 0xffffffffffffffff) was not found in any mem segments
SST Fatal Backtrace Information:
    0 : sst(_ZNK3SST6Output5fatalEjPKcS2_iS2_z+0x4ae) [0x55670e25dc3e]
    1 : /home/dev/agile/17_libc_tests/wf_ci/frameworks.simulators.agile.rev/src/librevcpu.so(_ZN3SST6RevCPU6RevMem12CalcPhysAddrEmm+0x1ea) [0x7faaeceecf5a]
    2 : /home/dev/agile/17_libc_tests/wf_ci/frameworks.simulators.agile.rev/src/librevcpu.so(_ZN3SST6RevCPU6RevMem7ReadMemEmmPvj+0x37) [0x7faaeceedd67]
    3 : /home/dev/agile/17_libc_tests/wf_ci/frameworks.simulators.agile.rev/src/librevcpu.so(_ZN3SST6RevCPU5RV64I2ldEPNS0_10RevFeatureEPNS0_10RevRegFileEPNS0_6RevMemENS0_7RevInstE+0x79) [0x7faaecf1b909]
    4 : /home/dev/agile/17_libc_tests/wf_ci/frameworks.simulators.agile.rev/src/librevcpu.so(_ZN3SST6RevCPU7RevProc9ClockTickEm+0xb01) [0x7faaecf12b91]
    5 : /home/dev/agile/17_libc_tests/wf_ci/frameworks.simulators.agile.rev/src/librevcpu.so(_ZN3SST6RevCPU6RevCPU9clockTickEm+0xa6) [0x7faaececa796]
    6 : sst(_ZN3SST5Clock7executeEv+0x10c) [0x55670e1ee9cc]
    7 : sst(_ZN3SST15Simulation_impl3runEv+0x3d3) [0x55670e2874d3]
    8 : sst(+0xf808e) [0x55670e1d108e]
    9 : sst(main+0x186b) [0x55670e1b6e0b]
   10 : /lib/x86_64-linux-gnu/libc.so.6(+0x29d90) [0x7faaf4e28d90]
   11 : /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0x80) [0x7faaf4e28e40]
   12 : sst(_start+0x25) [0x55670e1ce4b5]
--------------------------------------------------------------------------
MPI_ABORT was invoked on rank 0 in communicator MPI_COMM_WORLD
with errorcode 11.

NOTE: invoking MPI_ABORT causes Open MPI to kill all MPI processes.
You may or may not see output from other processes, depending on
exactly when Open MPI kills them.
```

jleidel, 2023-08-21:
Python file?

marcinkoss, 2023-08-21:
Always the same, nothing special

```
import os
import sst
import sys

program = os.getenv("REV_EXE", "xxx")

# Define SST core options
sst.setProgramOption("timebase", "1ps")

# Tell SST what statistics handling we want
sst.setStatisticLoadLevel(4)

max_addr_gb = 1

# Define the simulation components
comp_cpu = sst.Component("cpu", "revcpu.RevCPU")
comp_cpu.addParams({
    "verbose": verbose_level,                    # Verbosity
    "numCores": 1,                               # Number of cores
    "clock": "1.0GHz",                           # Clock
    "memSize": 1024*1024*1024,                   # Memory size in bytes
    "machine": "[0:RV64IMAFD]",                  # Core:Config
    "memCost": "[0:1:10]",                       # Memory loads required 1-10 cycles
    "program": program,                          # Target executable
    "splash": 0,                                 # Display the splash message
    "args": ""
})

sst.setStatisticOutput("sst.statOutputCSV")
sst.enableAllStatisticsForAllComponents()
```

closed 2023-08-21 by jleidel

marcinkoss, 2023-08-21:
Could you post link to fix in devel ?