tactcomplabs/rev #103: [P1] RV64UD ISA tests fails

marcinkoss, 2023-08-18:
Most of ISA tests from https://github.com/riscv-software-src/riscv-tests/tree/master/isa/rv64ud fails.
This is umbrella task for ISA tests, to not create separate report for each instruction:
fadd.S
fclass.S
fcmp.S
fcvt.S
fcvt_w.S
fdiv.S
fmadd.S
fmin.S
ldst.S
move.S
recoding.S
structural.S

As an example, please compile and run:
```
#include <string.h>

#define assert(x) if (!(x)) { asm(".dword 0x00000000"); }

double sub_double(double x, double y) {
        return (x - y);
}

int main() {
        double pi = 3.14159265;
        double res = sub_double(pi, 0.00000001);
        long hexrep = 0;
        memcpy((void*)&hexrep, (void*)&res, 8);

        // Expected values is 3.1415926400000001 = 0x400921fb52713c03 in hex representation
        assert (hexrep == 0x400921fb52713c03)
        return 0;
}
```

Test should pass, but it does not.

leekillough, 2023-08-19:
Still [in progress](https://github.com/tactcomplabs/rev/pull/104), but hardly ready yet. Please be patient.

marcinkoss, 2023-08-21:
The same is with RV64UF extenstions - most tests fails. (https://github.com/riscv-software-src/riscv-tests/tree/master/isa/rv64uf)

leekillough, 2023-08-25:
Fixed by https://github.com/tactcomplabs/rev/pull/104
```
RevCPU[cpu:ClockTick:41000]: Core 0 ; ThreadID 0; Retiring PC= 0x10280
RevCPU[cpu:ClockTick:41000]: Thread 1024 completed execution.
RevCPU[cpu:ClockTick:41000]: Program execution complete
RevCPU[cpu:ClockTick:41000]: Program Stats: Total Cycles: 41 Busy Cycles: 39 Idle Cycles: 2 Eff: 0.951219
RevCPU[cpu:ClockTick:41000]:     Bytes Read: 280 Bytes Written: 3902 Floats Read: 0 Doubles Read 0 Floats Exec: 13 TLB Hits: 18 TLB Misses: 113 Inst Retired: 39
RevCPU[cpu:clockTick:41000]: Closing Processor 0 at Cycle: 41
RevCPU[cpu:clockTick:41000]: OK to end sim at cycle: 41
Simulation is complete, simulated time: 41 ns
```

closed 2023-09-19 by leekillough