tactcomplabs/rev #105: memCost paramer no longer affects simulation

ukasz, 2023-08-21:
Some time ago by changing memCost REV parameter we were able to get different number of cycles during simulation thereby different efficiency. Now changing this parameter makes no effect to simulation time. Tested with memset/memcpy functions on devel branch.

ukasz, 2023-08-21:
Test:
```
#include <string.h>
#define assert(x) if (!(x)) { asm(".byte 0x00"); asm(".byte 0x00"); asm(".byte 0x00"); asm(".byte 0x00"); }
#define N (1024*10)
#define MAX_OFFSET 8
char mem[N + MAX_OFFSET];
char dest[N + MAX_OFFSET];

int main() {
        for(int offset = 0; offset < MAX_OFFSET; offset++) {
                for(int doffset = 0; doffset < MAX_OFFSET; doffset++) {
                        for(int i = 0; i < N + MAX_OFFSET; i++) {
                                mem[i] = (char)i;
                                dest[i] = 0;
                        }
                        char *smem = &mem[offset];
                        char *dmem = &dest[doffset];
                        memcpy(dmem, smem, N);
                        for(int i = 0; i < N; i++)
                                assert(smem[i] == dmem[i]);
                }
        }
        return 0;
}
```
Config
```
#
# Copyright (C) 2023 Intel Corporation
#
# See LICENSE in the top level directory for licensing details
#

import os
import sst
import sys

util_dir = os.path.dirname(__file__) + "/../utils/"
sys.path.insert(0, util_dir)
from elfhelper import get_sym

program = os.getenv("REV_EXE", "xxx")

rev_exe_args = ""
if os.getenv("REV_EXE_ARGS") != None:
    rev_exe_args = os.getenv("REV_EXE_ARGS")

verbose_level = 2
if os.getenv("REV_VERBOSE_LEVEL") != None:
    verbose_level = os.getenv("REV_VERBOSE_LEVEL")

ep = get_sym(program, "main")
# User can overwrite start address with env variable
start_addr = os.getenv("REV_START_ADDRESS")
if start_addr != None:
    ep = start_addr
ep = "[0:0x" + ep + "]"

# Define SST core options
sst.setProgramOption("timebase", "1ps")

# Tell SST what statistics handling we want
sst.setStatisticLoadLevel(4)

max_addr_gb = 1

# Define the simulation components
comp_cpu = sst.Component("cpu", "revcpu.RevCPU")
comp_cpu.addParams({
    "verbose": verbose_level,                    # Verbosity
    "numCores": 1,                               # Number of cores
    "clock": "1.0GHz",                           # Clock
    "memSize": 1024*1024*1024,                   # Memory size in bytes
    "machine": "[0:RV64IMAFD]",                  # Core:Config
    "startAddr": ep,                             # Starting address for core 0
    "memCost": "[0:10:100]",                       # Memory loads required 1-10 cycles
    "program": program,                          # Target executable
    "splash": 0,                                 # Display the splash message
    "args": rev_exe_args
})

sst.setStatisticOutput("sst.statOutputCSV")
sst.enableAllStatisticsForAllComponents()

# EOF
```

jleidel, 2023-09-28:
this function is being deprecated

leekillough, 2024-07-17:
We should close this issue if it no longer applies