tactcomplabs/rev #110: [P0] Atomic increment does not work.

marcinkoss, 2023-08-29:
**Describe the bug**
Follow up to https://github.com/tactcomplabs/rev/issues/56.
Atomic add (and other atomics as well) still does not work.

```

**To Reproduce**
Build & compile code in rev/test/amo/amoadd_c:
```
uint64_t atom64 = 0;
int main() {
    __atomic_fetch_add(&atom64, 2, __ATOMIC_RELAXED);
    __atomic_fetch_add(&atom64, 2, __ATOMIC_CONSUME);
    __atomic_fetch_add(&atom64, 2, __ATOMIC_ACQUIRE);
    __atomic_fetch_add(&atom64, 2, __ATOMIC_RELEASE);
    __atomic_fetch_add(&atom64, 2, __ATOMIC_ACQ_REL);
    __atomic_fetch_add(&atom64, 2, __ATOMIC_SEQ_CST);

    return 0;
}

Value of atom64 is not 12 as expected but 0. This is not accepted behaviour.

All issues must be accompanied by: 
```
REV/devel branch:
commit 41f018759bc4123cdc6c260cee6f8c7d8ba72c3e (HEAD -> rev-devel, origin/rev-devel)
Merge: 7bcbc91 01c8834
Author: jleidel <jleidel@users.noreply.github.com>
Date:   Mon Aug 28 15:54:16 2023 -0500

    Merge pull request #109 from leekillough/uninitialized

    Add initialization of uninitialized variables
```

- The compiler utilized to build the test
clang

- The python or JSON simulation input
script in rev/test/amo/*.py

**Expected behavior**
Atomic add (and other atomics works as expected)

jleidel, 2023-08-29:
need more info

closed 2023-08-29 by jleidel

marcinkoss, 2023-08-30:
could you write what info you need?