tactcomplabs/rev #114: ECALLs do not support RV32 [P2]

donofrio, 2023-09-05:
**Describe the bug**
Currently all ECALL implementations take program arguments and return data exclusively to the RV64 register file. This will cause issues when running in an RV32 config

**To Reproduce**
Run any of the tests in the syscalls folder using an RV32 configuration (similar to the one in the EX1 test directory) 

**Expected behavior**
Expect that the tests pass in 32 or 64 bit mode

**Additional context**
Mostly we need to approach the ECALL implementations the same as the other RV instructions and branch based on RV32 vs RV64 to use the correct register file

leekillough, 2023-09-19:
Some of the changes in #115 addressed this problem, by using the new `R->GetX<T>(F, rs)` and `R->SetX(F, rd, value)` APIs, and not to directly access `R->RV32[rs]`, `R->RV64[rs]`, etc. Previously the code was always accessing `R->RV64[rs]`.

rkabrick, 2023-10-26:
I think we can close this? @donofrio

closed 2023-10-30 by rkabrick