tactcomplabs/rev #120: amoswap.d.aqrl Instruction Not Behaving as Expected

aperdeus, 2023-09-14:
**Issue Description:**
The amoswap.d.aqrl instruction is not behaving as expected and appears to be malfunctioning. The issue is evident when using this instruction via a compiler builtin. In the code segment provided below, the test should pass without any assertion failures, but it fails when executed on the REV platform. It appears that the amoswap.d.aqrl instruction does not perform the expected swap, and the variables involved are actually being zeroed out during the execution of amoswap.d.aqrl.

**To Reproduce:**
Environment:
sst --version
SST-Core Version (-dev)
sst-config --CXX
g++
sst-config --ELEMENT_CXXFLAGS
-std=c++17 -fPIC -DHAVE_CONFIG_H -I/home/aperdeus/git/sst-core/build/../sst-core-install/include
REV: devel (https://github.com/tactcomplabs/rev/commit/4a724e2edaaaab80396c05f97551cbebe2e436a3)
REV compiled with: g++
compiler: clang (intel)

To reproduce this issue, please follow the steps below:

1. Compile the code snippet provided.
2. Launch REV with 1 HART with following ISA: RV64IMAFD
3. Set the entry point for the following function: main
4. The test should exit without assertion, current behavior is assertion

```
#include <limits.h>
#include <stdint.h>
#include <stdio.h>

#define assert(x)                                                              \
  if (!(x)) {                                                                  \
    asm(".byte 0x00");                                                         \
    asm(".byte 0x00");                                                         \
    asm(".byte 0x00");                                                         \
    asm(".byte 0x00");                                                         \
  }

uint64_t swap_dest = 0xfee1dead, swap_src = (((uint64_t)0xdeadbeef) << 32), swap_prev = 0;

void test_single_thread() {
  __atomic_exchange(&swap_dest, &swap_src, &swap_prev, __ATOMIC_SEQ_CST);
  assert(swap_dest == (((uint64_t)0xdeadbeef) << 32));
  assert(swap_prev == 0xfee1dead);
}

int main() {
  // single thread test
  test_single_thread();
  return 0;
```

jleidel, 2023-09-28:
possibly fixed is PR #135

jleidel, 2023-10-06:
confirmed bug

jleidel, 2023-10-06:
confirming that the incoming value to the AMO for the swap value appears to be '0'

jleidel, 2023-10-10:
Fix coming in 99af375

jleidel, 2023-10-10:
This should be fixed now; added tests

aperdeus, 2023-10-17:
Confirmed to work on 4d22ad799fe907618efd65cd2a61139ad55b0d89

closed 2023-10-25 by jleidel