tactcomplabs/rev #130: fmadd.d returns invalid value

ukasz, 2023-09-22:
**Describe the bug**
Values returned by fma.double are either not valid or cannot be compared with expected value.
We observe similar problems with other math functions, so a bug comparing double data type is plausible.
Consider the following c code:
```
#define assert(x) if (!(x)) { asm(".byte 0x00"); asm(".byte 0x00"); asm(".byte 0x00"); asm(".byte 0x00"); }
#define T double
int main() {
        volatile T ad = 2.0;
        volatile T bd = 3.0;
        volatile T cd = 5.0;

        T rd = ad * bd + cd;
        assert(rd >= 10.99);
        assert(rd <= 11.01);
		return 0;
}
```
The above program hits an assert on double data type (on float works fine).

The problem is in fma instruction itself or with comparison on double type, which can be
reproduced by the code below:
```
        volatile T ad = 2.0;
        volatile T bd = 3.0;
        volatile T cd = 5.0;

        T rd = (ad * bd) + cd;
        assert(rd > 1000); //fails
        assert(rd < 1000); //fails
        assert(rd != 1000);//ok
``

Disassembly of the test:
```
0000000000011158 <main>:
   11158: 13 01 01 fc   addi    sp, sp, -64
   1115c: 23 3c 11 02   sd      ra, 56(sp)
   11160: 23 38 81 02   sd      s0, 48(sp)
   11164: 13 04 01 04   addi    s0, sp, 64
   11168: 13 05 00 00   li      a0, 0
   1116c: 23 26 a4 fe   sw      a0, -20(s0)
   11170: 13 05 10 00   li      a0, 1
   11174: 13 15 e5 03   slli    a0, a0, 62
   11178: 23 30 a4 fe   sd      a0, -32(s0)
   1117c: 37 15 80 00   lui     a0, 2049
   11180: 13 15 75 02   slli    a0, a0, 39
   11184: 23 3c a4 fc   sd      a0, -40(s0)
   11188: 37 55 00 01   lui     a0, 4101
   1118c: 13 15 65 02   slli    a0, a0, 38
   11190: 23 38 a4 fc   sd      a0, -48(s0)
   11194: 07 30 04 fe   fld     ft0, -32(s0)
   11198: 87 30 84 fd   fld     ft1, -40(s0)
   1119c: 07 31 04 fd   fld     ft2, -48(s0)
   111a0: 43 70 10 12   fmadd.d ft0, ft0, ft1, ft2
   111a4: 27 34 04 fc   fsd     ft0, -56(s0)
   111a8: 87 30 84 fc   fld     ft1, -56(s0)
   111ac: 37 25 01 00   lui     a0, 18
   111b0: 07 30 05 20   fld     ft0, 512(a0)
   111b4: 53 05 10 a2   fle.d   a0, ft0, ft1
   111b8: 63 18 05 00   bnez    a0, 0x111c8 <main+0x70>
   111bc: 6f 00 40 00   j       0x111c0 <main+0x68>
   111c0: 00 00         <unknown>
   111c2: 00 00         <unknown>
   111c4: 6f 00 40 00   j       0x111c8 <main+0x70>
   111c8: 07 30 84 fc   fld     ft0, -56(s0)
   111cc: 37 25 01 00   lui     a0, 18
   111d0: 87 30 85 20   fld     ft1, 520(a0)
   111d4: 53 05 10 a2   fle.d   a0, ft0, ft1
   111d8: 63 18 05 00   bnez    a0, 0x111e8 <main+0x90>
[... 291 lines cut ...]
RevCPU[cpu:ClockTick:24000]: Core 0 ; ThreadID 0; Retiring PC= 0x111b0
RevCPU[cpu:DecodeInst:25000]: Core 0 ; Thread 0; PC:InstPayload = 0x111b4:0xa2100553
RevCPU[cpu:ClockTick:25000]: Core 0 ; Thread 0; Executing PC= 0x111b4
RevCPU[cpu:ClockTick:25000]: Core 0 ; ThreadID 0; Retiring PC= 0x111b4
RevCPU[cpu:DecodeInst:26000]: Core 0 ; Thread 0; PC:InstPayload = 0x111b8:0x51863
RevCPU[cpu:ClockTick:26000]: Core 0 ; Thread 0; Executing PC= 0x111b8
RevCPU[cpu:ClockTick:26000]: Core 0 ; ThreadID 0; Retiring PC= 0x111b8
RevCPU[cpu:DecodeInst:27000]: Core 0 ; Thread 0; PC:InstPayload = 0x111bc:0x40006f
RevCPU[cpu:ClockTick:27000]: Core 0 ; Thread 0; Executing PC= 0x111bc
RevCPU[cpu:ClockTick:27000]: Core 0 ; ThreadID 0; Retiring PC= 0x111bc
FATAL: RevCPU[cpu:DecodeInst:28000]: Error: Core 0 failed to decode instruction at PC=0x111c0; Inst=0
SST Fatal Backtrace Information:
    0 : sst(_ZNK3SST6Output5fatalEjPKcS2_iS2_z+0x4ae) [0x55dc0524dcae]
    1 : /home/dev/riscv/off/rev3/devel/src/librevcpu.so(_ZN3SST6RevCPU7RevProc10DecodeInstEv+0x1a7) [0x7fbe88e43a37]
    2 : /home/dev/riscv/off/rev3/devel/src/librevcpu.so(_ZN3SST6RevCPU7RevProc9ClockTickEm+0x105a) [0x7fbe88e4549a]
    3 : /home/dev/riscv/off/rev3/devel/src/librevcpu.so(_ZN3SST6RevCPU6RevCPU9clockTickEm+0xc4) [0x7fbe88df7634]
    4 : sst(_ZN3SST5Clock7executeEv+0x10c) [0x55dc051d9e4c]
    5 : sst(_ZN3SST15Simulation_impl3runEv+0x3d3) [0x55dc05277513]
    6 : sst(+0xeebfe) [0x55dc051bfbfe]
    7 : sst(main+0x1853) [0x55dc051a5ce3]
    8 : /lib/x86_64-linux-gnu/libc.so.6(+0x29d90) [0x7fbe8daabd90]
    9 : /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0x80) [0x7fbe8daabe40]
   10 : sst(_start+0x25) [0x55dc051bd535]
Abort(-1) on node 0 (rank 0 in comm 0): application called MPI_Abort(MPI_COMM_WORLD, -1) - process 0
```

leekillough, 2023-09-22:
It appears to be a different problem related to memory. I can reduce the testcase to this (no asserts, FMAs, or FP compares), and it still occurs:
```
int main() {
  volatile double bd = 3.0;
  return 0;
}
```
This, with a global variable instead of a `volatile` local variable, also fails at both `-O0` and `-O2`:
```
#include<cstdio>
#define T double
T bd = 3.0;
int main() {
  printf("result %f\n", bd);
}
```
So does this trivial function, at both `-O0` and `-O2`:
```
int main() {
}
```
Please review the code for FP [FMA](https://github.com/tactcomplabs/rev/blob/1b330d6a44aa133110b6f77fda7a911d5ded1ea1/include/insns/RV32D.h#L59) and [FP comparisons](https://github.com/tactcomplabs/rev/blob/1b330d6a44aa133110b6f77fda7a911d5ded1ea1/include/RevInstTable.h#L618) and [here](https://github.com/tactcomplabs/rev/blob/1b330d6a44aa133110b6f77fda7a911d5ded1ea1/include/insns/RV32D.h#L90).

ukasz, 2023-09-23:
How exactly the code below fails?
```
int main() {
  volatile double bd = 3.0;
  return 0;
}
```

leekillough, 2023-09-23:
```
WARNING: Building component "cpu" with no links assigned.
RevCPU[cpu:RevCPU:0]: Building Rev with 1 cores
RevCPU[cpu:LoadElf:0]: Symbol Table Entry [:0x0]
RevCPU[cpu:LoadElf:0]: Symbol Table Entry [$xrv64i2p1_m2p0_a2p1_f2p2_d2p2_v1p0_zicsr2p0_zmmul1p0_zve32f1p0_zve32x1p0_zve64d1p0_zve64f1p0_zve64x1p0_zvl128b1p0_zvl32b1p0_zvl64b1p0:0x105f4]
RevCPU[cpu:LoadElf:0]: Symbol Table Entry [__BSS_END__:0x11df8]
RevCPU[cpu:LoadElf:0]: Symbol Table Entry [__DATA_BEGIN__:0x11660]
RevCPU[cpu:LoadElf:0]: Symbol Table Entry [__EH_FRAME_BEGIN__:0x11608]
RevCPU[cpu:LoadElf:0]: Symbol Table Entry [__FRAME_END__:0x11640]
RevCPU[cpu:LoadElf:0]: Symbol Table Entry [__SDATA_BEGIN__:0x11da8]
RevCPU[cpu:LoadElf:0]: Symbol Table Entry [__atexit.c:0x0]
RevCPU[cpu:LoadElf:0]: Symbol Table Entry [__bss_start:0x11dc0]
RevCPU[cpu:LoadElf:0]: Symbol Table Entry [__call_atexit.c:0x0]
RevCPU[cpu:LoadElf:0]: Symbol Table Entry [__call_exitprocs:0x10384]
RevCPU[cpu:LoadElf:0]: Symbol Table Entry [__do_global_dtors_aux:0x1017c]
RevCPU[cpu:LoadElf:0]: Symbol Table Entry [__do_global_dtors_aux_fini_array_entry:0x11658]
RevCPU[cpu:LoadElf:0]: Symbol Table Entry [__dso_handle:0x11db0]
RevCPU[cpu:LoadElf:0]: Symbol Table Entry [__errno:0x105f4]
RevCPU[cpu:LoadElf:0]: Symbol Table Entry [__fini_array_end:0x11660]
RevCPU[cpu:LoadElf:0]: Symbol Table Entry [__fini_array_start:0x11658]
RevCPU[cpu:LoadElf:0]: Symbol Table Entry [__frame_dummy_init_array_entry:0x11650]
RevCPU[cpu:LoadElf:0]: Symbol Table Entry [__global_pointer$:0x11e60]
RevCPU[cpu:LoadElf:0]: Symbol Table Entry [__init_array_end:0x11658]
RevCPU[cpu:LoadElf:0]: Symbol Table Entry [__init_array_start:0x11648]
RevCPU[cpu:LoadElf:0]: Symbol Table Entry [__libc_fini_array:0x104ac]
RevCPU[cpu:LoadElf:0]: Symbol Table Entry [__libc_init_array:0x1020c]
RevCPU[cpu:LoadElf:0]: Symbol Table Entry [__preinit_array_end:0x11644]
RevCPU[cpu:LoadElf:0]: Symbol Table Entry [__preinit_array_start:0x11644]
RevCPU[cpu:LoadElf:0]: Symbol Table Entry [__register_exitproc:0x1051c]
RevCPU[cpu:LoadElf:0]: Symbol Table Entry [_edata:0x11dc0]
RevCPU[cpu:LoadElf:0]: Symbol Table Entry [_end:0x11df8]
RevCPU[cpu:LoadElf:0]: Symbol Table Entry [_exit:0x105c4]
RevCPU[cpu:LoadElf:0]: Symbol Table Entry [_global_impure_ptr:0x11da8]
RevCPU[cpu:LoadElf:0]: Symbol Table Entry [_impure_ptr:0x11db8]
RevCPU[cpu:LoadElf:0]: Symbol Table Entry [_start:0x10130]
RevCPU[cpu:LoadElf:0]: Symbol Table Entry [atexit:0x10508]
RevCPU[cpu:LoadElf:0]: Symbol Table Entry [atexit.c:0x0]
RevCPU[cpu:LoadElf:0]: Symbol Table Entry [completed.1:0x11dc0]
RevCPU[cpu:LoadElf:0]: Symbol Table Entry [crt0.o:0x0]
RevCPU[cpu:LoadElf:0]: Symbol Table Entry [crtstuff.c:0x0]
RevCPU[cpu:LoadElf:0]: Symbol Table Entry [errno.c:0x0]
RevCPU[cpu:LoadElf:0]: Symbol Table Entry [exit:0x100e8]
RevCPU[cpu:LoadElf:0]: Symbol Table Entry [exit.c:0x0]
RevCPU[cpu:LoadElf:0]: Symbol Table Entry [fini.c:0x0]
RevCPU[cpu:LoadElf:0]: Symbol Table Entry [foo.cpp:0x0]
RevCPU[cpu:LoadElf:0]: Symbol Table Entry [frame_dummy:0x101c0]
RevCPU[cpu:LoadElf:0]: Symbol Table Entry [impure.c:0x0]
RevCPU[cpu:LoadElf:0]: Symbol Table Entry [impure_data:0x11660]
RevCPU[cpu:LoadElf:0]: Symbol Table Entry [init.c:0x0]
RevCPU[cpu:LoadElf:0]: Symbol Table Entry [lib_a-memset.o:0x0]
RevCPU[cpu:LoadElf:0]: Symbol Table Entry [main:0x101e0]
RevCPU[cpu:LoadElf:0]: Symbol Table Entry [memset:0x102a8]
RevCPU[cpu:LoadElf:0]: Symbol Table Entry [object.0:0x11dc8]
RevCPU[cpu:LoadElf:0]: Symbol Table Entry [register_fini:0x10118]
RevCPU[cpu:LoadElf:0]: Symbol Table Entry [sys_exit.c:0x0]
RevCPU[cpu:LoadProgramArgs:0]: Loading program argv[0] = a.out
RevCPU[cpu:RevFeature:0]: Core 0 ; Initializing feature set from machine string=RV32I
RevCPU[cpu:ParseMachineModel:0]: Core 0 ; Setting XLEN to 32
RevCPU[cpu:ParseMachineModel:0]: Core 0 ; Architecture string=I
RevCPU[cpu:SeedInstTable:0]: Core 0 ; Seeding instruction table for machine model=RV32I
RevCPU[cpu:EnableExt:0]: Core 0 ; Enabling extension=RV32I
RevCPU[cpu:InitTableMapping:0]: Core 0 ; Initializing table mapping for machine model=RV32I
RevCPU[cpu:InitTableMapping:0]: Core 0 ; Table Entry 55 = lui
RevCPU[cpu:InitTableMapping:0]: Core 0 ; Table Entry 23 = auipc
RevCPU[cpu:InitTableMapping:0]: Core 0 ; Table Entry 111 = jal
RevCPU[cpu:InitTableMapping:0]: Core 0 ; Table Entry 103 = jalr
RevCPU[cpu:InitTableMapping:0]: Core 0 ; Table Entry 99 = beq
RevCPU[cpu:InitTableMapping:0]: Core 0 ; Table Entry 355 = bne
RevCPU[cpu:InitTableMapping:0]: Core 0 ; Table Entry 1123 = blt
RevCPU[cpu:InitTableMapping:0]: Core 0 ; Table Entry 1379 = bge
[... 53 lines cut ...]
RevCPU[cpu:ClockTick:2000]: Core 0 ; ThreadID 0; Retiring PC= 0x101e0
RevCPU[cpu:DecodeInst:3000]: Core 0 ; Thread 0; PC:InstPayload = 0x101e4:0x813c23
FATAL: RevCPU[cpu:DecodeInst:3000]: Error: failed to decode instruction at PC=0x101e4; Enc=803
SST Fatal Backtrace Information:
    0 : sst(_ZNK3SST6Output5fatalEjPKcS2_iS2_z+0x495) [0x55ab8caa75b5]
    1 : build/src/librevcpu.so(_ZN3SST6RevCPU7RevProc10DecodeInstEv+0x6ff) [0x7fde7dbc6b03]
    2 : build/src/librevcpu.so(_ZN3SST6RevCPU7RevProc9ClockTickEm+0x35e) [0x7fde7dbc7c82]
    3 : build/src/librevcpu.so(_ZN3SST6RevCPU6RevCPU9clockTickEm+0xb4) [0x7fde7db5a50c]
    4 : build/src/librevcpu.so(_ZN3SST10SSTHandlerIbmNS_6RevCPU6RevCPUEvE13operator_implEm+0x75) [0x7fde7db824ed]
    5 : sst(_ZN3SST5Clock7executeEv+0xac) [0x55ab8ca383cc]
    6 : sst(_ZN3SST15Simulation_impl3runEv+0x3f3) [0x55ab8cad0333]
[... cut ...]
   10 : /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0x89) [0x7fde80823b49]
   11 : sst(_start+0x25) [0x55ab8ca1ca05]
--------------------------------------------------------------------------
MPI_ABORT was invoked on rank 0 in communicator MPI_COMM_WORLD
with errorcode -1.

NOTE: invoking MPI_ABORT causes Open MPI to kill all MPI processes.
You may or may not see output from other processes, depending on
exactly when Open MPI kills them.
--------------------------------------------------------------------------
```

ukasz, 2023-09-23:
Do you see similar problem with RV64G? RV32I doesn't have support for floating point instructions.

leekillough, 2023-09-23:
> Do you see similar problem with RV64G? RV32I doesn't have support for floating point instructions.

Sorry, I thought I was using an RV64G arch string. It was also late at night :)  I'll try it again later.

leekillough, 2023-09-23:
First test passes with RV64G; second test gets this error, which appears to be an assertion failure:
```
FATAL: RevCPU[cpu:DecodeInst:29000]: Error: Core 0 failed to decode instruction at PC=0x1024c; Inst=0
SST Fatal Backtrace Information:
    0 : sst(_ZNK3SST6Output5fatalEjPKcS2_iS2_z+0x495) [0x55c3dca775b5]
    1 : build/src/librevcpu.so(_ZN3SST6RevCPU7RevProc10DecodeInstEv+0x1aa) [0x7f68145c65ae]
    2 : build/src/librevcpu.so(_ZN3SST6RevCPU7RevProc9ClockTickEm+0x35e) [0x7f68145c7c82]
    3 : build/src/librevcpu.so(_ZN3SST6RevCPU6RevCPU9clockTickEm+0xb4) [0x7f681455a50c]
    4 : build/src/librevcpu.so(_ZN3SST10SSTHandlerIbmNS_6RevCPU6RevCPUEvE13operator_implEm+0x75) [0x7f68145824ed]
    5 : sst(_ZN3SST5Clock7executeEv+0xac) [0x55c3dca083cc]
    6 : sst(_ZN3SST15Simulation_impl3runEv+0x3f3) [0x55c3dcaa0333]
    7 : sst(+0xef175) [0x55c3dc9ef175]
    8 : sst(main+0x2cf1) [0x55c3dc9d6871]
    9 : /lib/x86_64-linux-gnu/libc.so.6(+0x23a90) [0x7f6817023a90]
   10 : /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0x89) [0x7f6817023b49]
   11 : sst(_start+0x25) [0x55c3dc9eca05]
--------------------------------------------------------------------------
MPI_ABORT was invoked on rank 0 in communicator MPI_COMM_WORLD
with errorcode -1.

NOTE: invoking MPI_ABORT causes Open MPI to kill all MPI processes.
You may or may not see output from other processes, depending on
exactly when Open MPI kills them.
--------------------------------------------------------------------------
```

This is the source I used for the second test:

```
#define assert(x) if (!(x)) { asm(".byte 0x00"); asm(".byte 0x00"); asm(".byte 0x00"); asm(".byte 0x00"); }
#define T double
int main() {
        volatile T ad = 2.0;
        volatile T bd = 3.0;
        volatile T cd = 5.0;

        T rd = (ad * bd) + cd;
        assert(rd > 1000); //fails
        assert(rd < 1000); //fails
        assert(rd != 1000);//ok
                return 0;
}
```

Since `rd == 11.0`, the `assert(rd > 1000);` _should_ fail. If I comment out that assertion, the second test passes.

I have not tried `RV32D`, because I would need to build a RV32 toolchain.

ukasz, 2023-09-25:
Second test is just to outline the problem in the first test. I just tried first test with latest devel and intel tigre llvm compiler and it still fails for me. Which compiler did you use to get pass on first test? I can provide a "portable" c file with hardcoded blob containing instructions if that helps.

leekillough, 2023-09-26:
```
$ echo $RVCC
riscv64-unknown-elf-gcc
```

ukasz, 2023-09-26:
For me on gcc with -O0 and -O1 compiler does not generate fmadd instruction. It needs -O2 or -O3 to do that and then the code above should reproduce the problem.
```
# riscv64-linux-gnu-gcc -march=rv64imafd ex2.c -O1 -o ex2.exe
# riscv64-linux-gnu-objdump -d ex2.exe | grep fmadd
# riscv64-linux-gnu-gcc -march=rv64imafd ex2.c -O2 -o ex2.exe
# riscv64-linux-gnu-objdump -d ex2.exe | grep fmadd
 5ac:   6ac7f7c3                fmadd.d fa5,fa5,fa2,fa3
```

Please let me know if you are able to reproduce it with -O2/O3

leekillough, 2023-11-02:
I decided to get away from compiler optimization-dependent issues, so I reproduced it with this case:
```c++
#define assert(x) if (!(x)) { asm(".word 0x00"); }
#define T double
int main() {
        volatile T ad = 2.0;
        volatile T bd = 3.0;
        volatile T cd = 5.0;
        volatile T rd;
        asm volatile(" fmadd.d %0, %1, %2, %3" : "=f"(rd) : "f"(ad), "f"(bd), "f"(cd) );
        assert(rd >= 10.99);
        assert(rd <= 11.01);
}
```
I verified that this case passes:
```c++
#define assert(x) if (!(x)) { asm(".word 0x00"); }
#define T float
int main() {
        volatile T ad = 2.0;
        volatile T bd = 3.0;
        volatile T cd = 5.0;
        volatile T rd;
        asm volatile(" fmadd.s %0, %1, %2, %3" : "=f"(rd) : "f"(ad), "f"(bd), "f"(cd) );
        assert(rd >= 10.99);
        assert(rd <= 11.01);
}
```
I am examining the decoding of:
```
   10140:       6ae7f7c3                fmadd.d fa5,fa5,fa4,fa3
```
vs:
```
   10140:       68e7f7c3                fmadd.s fa5,fa5,fa4,fa3
```

leekillough, 2023-11-07:
Please confirm this is fixed, and close if it is.

ukasz, 2023-11-07:
It works like a charm, thank you!

closed 2023-11-07 by ukasz