tactcomplabs/rev #134: Decoder does not recognize fcvt instructions correctly

ukasz, 2023-09-27:
Instruction fcvt is treated by instruction decoder as invalid instruction.
Some encodings of this instructions are working fine, other don't.
This is a regression, it used to work fine earlier.
Two examples of encodings not working correctly are:
```
fcvt.l.d a0, ft1, rup
fcvt.w.s a0, ft1, rup
```

**To Reproduce**
Compile the following program:
```
// riscv64-linux-gnu-gcc ex2.c -march=rv64g -O2 -o ex2.exe
int main() {
        asm(".word 0x00000513");//li a0, 0
        asm(".word 0xc220b553");//fcvt.l.d a0, ft1, rup
        asm("li a7, 93; li a0, 0; ecall"); //exit(0)
        return 0;
}
```
Run REV with config from ex2 test:
```
REV_VERBOSE_LEVEL=9 REV_EXE=/home/dev/riscv/off/rev3/test/tests_lit/../cpp/polymorphism.exe sst ../devel/test/ex2/rev-test-ex2.py
```

All issues must be accompanied by: 
- The version of SST utilized:
```
# sst --version
SST-Core Version (-dev)
# sst-config --CXX
g++
# sst-config --ELEMENT_CXXFLAGS
 -std=c++17 -fPIC -DHAVE_CONFIG_H -I/home/dev/riscv/sst-core-src/build/../sst-core-install/include
```
- The source code branch: devel
- The last commit hash: 577ac9c2c9aa13a51f6d2cc0b332e76ffc246107
- The compiler utilized to build Rev: gcc version 11.3.0 (Ubuntu 11.3.0-1ubuntu1~22.04.1)
- The compiler utilized to build the test: riscv64tigre-unknown-unknown-elf 6c8d6606259aca093839a1d165ead48680c59066
- The python config file: test/ex2/rev-test-ex2.py

**Expected behavior**
Application ends without throwing invalid instruction error.
**Trace**
```
# REV_EXE=ex2.exe sst ../../devel/test/ex2/rev-test-ex2.py
WARNING: Building component "cpu" with no links assigned.
RevCPU[cpu:RevCPU:0]: Building Rev with 1 cores
RevCPU[cpu:LoadElf:0]: Symbol Table Entry [:0x0]
RevCPU[cpu:LoadElf:0]: Symbol Table Entry [$d:0x570]
RevCPU[cpu:LoadElf:0]: Symbol Table Entry [$x:0x5ba]
RevCPU[cpu:LoadElf:0]: Symbol Table Entry [Scrt1.o:0x0]
RevCPU[cpu:LoadElf:0]: Symbol Table Entry [_DYNAMIC:0x1e10]
RevCPU[cpu:LoadElf:0]: Symbol Table Entry [_GLOBAL_OFFSET_TABLE_:0x2020]
RevCPU[cpu:LoadElf:0]: Symbol Table Entry [_IO_stdin_used:0x644]
RevCPU[cpu:LoadElf:0]: Symbol Table Entry [_ITM_deregisterTMCloneTable:0x0]
RevCPU[cpu:LoadElf:0]: Symbol Table Entry [_ITM_registerTMCloneTable:0x0]
RevCPU[cpu:LoadElf:0]: Symbol Table Entry [_PROCEDURE_LINKAGE_TABLE_:0x540]
RevCPU[cpu:LoadElf:0]: Symbol Table Entry [__BSS_END__:0x2050]
RevCPU[cpu:LoadElf:0]: Symbol Table Entry [__DATA_BEGIN__:0x2000]
RevCPU[cpu:LoadElf:0]: Symbol Table Entry [__FRAME_END__:0x688]
RevCPU[cpu:LoadElf:0]: Symbol Table Entry [__GNU_EH_FRAME_HDR:0x648]
RevCPU[cpu:LoadElf:0]: Symbol Table Entry [__SDATA_BEGIN__:0x2048]
RevCPU[cpu:LoadElf:0]: Symbol Table Entry [__TMC_END__:0x2008]
RevCPU[cpu:LoadElf:0]: Symbol Table Entry [__abi_tag:0x2b8]
RevCPU[cpu:LoadElf:0]: Symbol Table Entry [__bss_start:0x2048]
RevCPU[cpu:LoadElf:0]: Symbol Table Entry [__cxa_finalize@GLIBC_2.27:0x0]
RevCPU[cpu:LoadElf:0]: Symbol Table Entry [__data_start:0x2000]
RevCPU[cpu:LoadElf:0]: Symbol Table Entry [__do_global_dtors_aux:0x608]
RevCPU[cpu:LoadElf:0]: Symbol Table Entry [__do_global_dtors_aux_fini_array_entry:0x1e08]
[... 194 lines cut ...]
/**///**  /*******//** /**
/**  //** /**////  //****
/**   //**//******  //**
//     //  //////    //

RevCPU[cpu:RevCPU:0]: Initialization of RevCPUs complete.
RevCPU[cpu:DecodeInst:2000]: Core 0 ; Thread 0; PC:InstPayload = 0x570:0x513
RevCPU[cpu:ClockTick:2000]: Core 0 ; Thread 0; Executing PC= 0x570
RevCPU[cpu:ClockTick:2000]: Core 0 ; ThreadID 0; Retiring PC= 0x570
RevCPU[cpu:DecodeInst:3000]: Core 0 ; Thread 0; PC:InstPayload = 0x574:0xc220b553
FATAL: RevCPU[cpu:DecodeInst:3000]: Error: failed to decode instruction at PC=0x574; Enc=199507
SST Fatal Backtrace Information:
    0 : sst(_ZNK3SST6Output5fatalEjPKcS2_iS2_z+0x4ae) [0x55f1770b6cae]
    1 : /home/dev/riscv/off/rev3/devel/src/librevcpu.so(_ZN3SST6RevCPU7RevProc10DecodeInstEv+0x716) [0x7f5ad50e3226]
    2 : /home/dev/riscv/off/rev3/devel/src/librevcpu.so(_ZN3SST6RevCPU7RevProc9ClockTickEm+0xcec) [0x7f5ad50e43bc]
    3 : /home/dev/riscv/off/rev3/devel/src/librevcpu.so(_ZN3SST6RevCPU6RevCPU9clockTickEm+0xc4) [0x7f5ad50968c4]
    4 : sst(_ZN3SST5Clock7executeEv+0x10c) [0x55f177042e4c]
    5 : sst(_ZN3SST15Simulation_impl3runEv+0x3d3) [0x55f1770e0513]
    6 : sst(+0xeebfe) [0x55f177028bfe]
    7 : sst(main+0x1853) [0x55f17700ece3]
    8 : /lib/x86_64-linux-gnu/libc.so.6(+0x29d90) [0x7f5ad9c72d90]
    9 : /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0x80) [0x7f5ad9c72e40]
   10 : sst(_start+0x25) [0x55f177026535]
Abort(-1) on node 0 (rank 0 in comm 0): application called MPI_Abort(MPI_COMM_WORLD, -1) - process 0
```

ukasz, 2023-10-05:
Bug is still present at 5c89acd45e3235551d853e839ab8f068f639e05e

leekillough, 2023-10-10:
There have been long-standing problems with `fcvt`, and this is probably a "false regression" (i.e., it never should have passed before). The `fcvt` instructions are being systematically examined, and new tests are being added.

leekillough, 2023-10-11:
You can try [fcvt_decode](https://github.com/tactcomplabs/rev/tree/fcvt_decode) branch. Functionality is tentatively completed, but testing is still being developed.

ukasz, 2023-10-25:
This particular bug is fixed, but we see more problems around fcvt. Will fill another bug.

closed 2023-10-25 by ukasz

leekillough, 2023-10-25:
> This particular bug is fixed, but we see more problems around fcvt. Will fill another bug.

Rounding and FP exceptions are still being implemented and tested, not just for `fcvt`.  [Latest work in progress](https://github.com/tactcomplabs/rev/tree/fcvt_decode).

There may be some corner cases where if `rtz` rounding is not used, that it gets the wrong result.