tactcomplabs/rev #144: [P1] JAL instruction does behave strage

marcinkoss, 2023-10-04:
**Describe the bug**
Discovered when run llvm/libcxx/tests -> llvm-project/libcxx/test/std/utilities/format/format.range/format.range.fmtset/format.functions.format.pass.cpp
Test, when build with LLVM toolchain generates:
```
  9b42c: 83 35 04 fb   ld      a1, -80(s0)
   9b430: 13 05 04 fc   addi    a0, s0, -64
   9b434: 93 06 84 fa   addi    a3, s0, -88
   9b438: ef 50 8b 35   jal     0x150790 <_ZNSt3__13setIcNS_4lessIcEENS_9allocatorIcEEEC2B7v160000ESt16initializer_listIcERKS2_>
```
and:
```
0000000000150790 <_ZNSt3__13setIcNS_4lessIcEENS_9allocatorIcEEEC2B7v160000ESt16initializer_listIcERKS2_>:
  150790: 13 01 01 fa   addi    sp, sp, -96
  150794: 23 3c 11 04   sd      ra, 88(sp)
  150798: 23 38 81 04   sd      s0, 80(sp)
```

Now, when binary is run on REV model, when PC reaches instruction at address 0x9b438 then JAL instruction does jump to 0x50790 instead of 0x150790, and finish with error "cannot decode instruction at address 0x50790)

I added few logs in REV code, and IMM = b5358 and PC is 9B438 (which gives 0x150790):
RevCPU[cpu:ClockTick:69000]: Core 0; Hart 0; Thread 1; Executing PC= 0x9b438
jal imm = b5358 imm sign ext = ffffb5358

I think it might some arithmetic error on U2 numbers inside REV and "1" is lost somewhere.

**To Reproduce**
All issues must be accompanied by: 
- The version of SST utilized (`sst --version`; `sst-config --CXX`; `sst-config --ELEMENT_CXXFLAGS`)
#sst --version
SST-Core Version (-dev)
# sst-config --CXX
g++
# sst-config --ELEMENT_CXXFLAGS
 -std=c++17 -fPIC -DHAVE_CONFIG_H -I<sst-core-path>/build/../sst-core-install/include

- The source code branch being used
devel 
- The last commit hash
sha e1c1c71b28cd5dee0439c1ef5d20a0e64846fb57

- The compiler utilized to build Rev
gcc/g++, 11.3.0

- The compiler utilized to build the test
LLVM clang 

- The python or JSON simulation input
```
import os
import sst
import sys

program = os.getenv("REV_EXE", "xxx")

# Define SST core options
sst.setProgramOption("timebase", "1ps")

# Tell SST what statistics handling we want
sst.setStatisticLoadLevel(4)

max_addr_gb = 1

# Define the simulation components
comp_cpu = sst.Component("cpu", "revcpu.RevCPU")
comp_cpu.addParams({
    "verbose": verbose_level,                    # Verbosity
    "numCores": 1,                               # Number of cores
    "clock": "1.0GHz",                           # Clock
    "memSize": 4096*1024*1024,                   # Memory size in bytes
    "machine": "[0:RV64IMAFD]",                  # Core:Config
    "memCost": "[0:1:10]",                       # Memory loads required 1-10 cycles
    "program": program,                          # Target executable
    "splash": 0,                                 # Display the splash message
})

sst.setStatisticOutput("sst.statOutputCSV")
sst.enableAllStatisticsForAllComponents()

```
- C/CXX test source (or mention which test in the current source tree failed)
LLVM  toolchain repository
 llvm-project/libcxx/test/std/utilities/format/format.range/format.range.fmtset/format.functions.format.pass.cpp

**Expected behavior**
Code should jump into desired PC and continue to run.

**Trace**
```

RevCPU[cpu:ClockTick:67000]: Core 0 ; ThreadID 0; Retiring PC= 0x9b430
RevCPU[cpu:DecodeInst:68000]: Core 0 ; Thread 0; PC:InstPayload = 0x9b434:0xfa840693
RevCPU[cpu:ClockTick:68000]: Core 0; Hart 0; Thread 1; Executing PC= 0x9b434
RevCPU[cpu:ClockTick:68000]: Core 0 ; ThreadID 0; Retiring PC= 0x9b434
RevCPU[cpu:DecodeInst:69000]: Core 0 ; Thread 0; PC:InstPayload = 0x9b438:0x358b50ef
RevCPU[cpu:ClockTick:69000]: Core 0; Hart 0; Thread 1; Executing PC= 0x9b438
jal imm = b5358 immex = b5358 
custom= b5358 Sum = 50790  150790
carry=0
RevCPU[cpu:ClockTick:69000]: Core 0 ; ThreadID 0; Retiring PC= 0x9b438
root@marcinko-76d77847f-8nwmt:/home/dev/agile/33_libcxxtest/tigre/firmware.platform.agile.toolchain# tail log^C
root@marcinko-76d77847f-8nwmt:/home/dev/agile/33_libcxxtest/tigre/firmware.platform.agile.toolchain# REV_EXE=toolchain/build/llvm-cross-compiler-sst/llvm-project/llvm/runtimes/runtimes-bins/test/std/utilities/format/format.range/format.range.fmtset/Output/format.functions.format.pass.cpp.dir/t.tmp.exe timeout 30s sst rev/test/configs/basic-1c.py
FATAL: RevCPU[cpu:DecodeInst:71000]: Error: Core 0 failed to decode instruction at PC=0x50790; Inst=0
SST Fatal Backtrace Information:
    0 : sst(_ZNK3SST6Output5fatalEjPKcS2_iS2_z+0x4ae) [0x561bbab85c3e]
    1 : rev/src/librevcpu.so(_ZN3SST6RevCPU7RevProc10DecodeInstEv+0x1ab) [0x7f2336157d4b]
    2 : rev/src/librevcpu.so(_ZN3SST6RevCPU7RevProc9ClockTickEm+0xa74) [0x7f2336159014]
    3 : rev/src/librevcpu.so(_ZN3SST6RevCPU6RevCPU9clockTickEm+0x143) [0x7f2336103a23]
    4 : sst(_ZN3SST5Clock7executeEv+0x10c) [0x561bbab169cc]
    5 : sst(_ZN3SST15Simulation_impl3runEv+0x3d3) [0x561bbabaf4d3]
    6 : sst(+0xf808e) [0x561bbaaf908e]
    7 : sst(main+0x186b) [0x561bbaadee0b]
    8 : /lib/x86_64-linux-gnu/libc.so.6(+0x29d90) [0x7f233ff66d90]
    9 : /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0x80) [0x7f233ff66e40]
   10 : sst(_start+0x25) [0x561bbaaf64b5]
--------------------------------------------------------------------------
MPI_ABORT was invoked on rank 0 in communicator MPI_COMM_WORLD
with errorcode -1.

NOTE: invoking MPI_ABORT causes Open MPI to kill all MPI processes.
You may or may not see output from other processes, depending on
exactly when Open MPI kills them.

```
**Additional context**
Add any other context about the problem here.

leekillough, 2023-10-04:
Should be fixed in [JAL](https://github.com/tactcomplabs/rev/tree/JAL) .

Adding tests for the 2 most significant bits `00` `01` `10` `11`.

leekillough, 2023-10-04:
Please confirm fix and close.

marcinkoss, 2023-10-05:
I checked on original test and it works. I will do a full testing sweep and if fixed I'll close this report