tactcomplabs/rev #146: [FEATURE REQUEST] : Add ability to explicitly stall the core pipeline from a coprocessor subcomponent

jleidel, 2023-10-04:
**Rev Feature Request**
Add ability to explicitly stall the core pipeline from a coprocessor subcomponent

**Version of SST**
SST 13.0.0

**Additional context**
Multiple individuals have requested this feature, so we need a standard interface to pend the pipeline until the coprocessor clears it.

jleidel, 2023-10-04:
we need to provide the pipeline hazard callbacks to the coproc subcomponents *AND* provide a "big hammer" mechanism to effectively pend the pipeline w/o the need for hazards on the register file

donofrio, 2023-10-11:
Feature implemented in latest devel [e2ca951](https://github.com/tactcomplabs/rev/commit/e2ca95162da0c3e1ebd7e78c2060bd64d531cbf6) as part of PR 157

closed 2023-10-11 by donofrio