tactcomplabs/rev #149: amoadd_cxx test is failing with 64bit data type

ukasz, 2023-10-05:
**Describe the bug**
Amoadd_cxx test is failing when we use 64bit data type instead of 32bit int.

**To Reproduce**
Modify test/amo/amoadd_cxx test to use long instead of int data type:
```
// riscv64-linux-gnu-g++ amoadd_cxx.cc -march=rv64imafd  -o amoadd_cxx.exe
#include <atomic>
#define assert(x) if (!(x)) { asm(".byte 0x00"); asm(".byte 0x00"); asm(".byte 0x00"); asm(".byte 0x00"); }

int main() {
        std::atomic<long> a;
        a = 1;  //amoswap
        assert(a == 1);
        a++;    //amoadd
        assert(a == 2);
}

```
Run REV with config from memh_2 test:
```
 REV_EXE=amoadd_cxx.exe sst rev-test-amoadd_cxx.py
```

REV configuration:
```
# sst --version
SST-Core Version (-dev)
# sst-config --CXX
g++
# sst-config --ELEMENT_CXXFLAGS
 -std=c++17 -fPIC -DHAVE_CONFIG_H -I/home/dev/riscv/sst-core-src/build/../sst-core-install/include
```
- The source code branch: devel
- The last commit hash: 5c89acd45e3235551d853e839ab8f068f639e05e
- The compiler utilized to build Rev: gcc version 11.3.0 (Ubuntu 11.3.0-1ubuntu1~22.04.1)
- The compiler utilized to build the test: riscv64tigre-unknown-unknown-elf 6c8d6606259aca093839a1d165ead48680c59066
- The python config file: test/amoadd_cxx/rev-test-amoadd_cxx.py

**Expected behavior**
Application ends without throwing invalid instruction error.
**Trace**
```
REV_EXE=amoadd_cxx.exe sst rev-test-amoadd_cxx.py
WARNING: Building component "cpu" with no links assigned.
RevCPU[cpu:RevCPU:0]: Building Rev with 1 cores
RevCPU[cpu:LoadElf:0]: Symbol Table Entry [:0x0]
RevCPU[cpu:LoadElf:0]: Symbol Table Entry [$d:0x780]
RevCPU[cpu:LoadElf:0]: Symbol Table Entry [$x:0x8e0]
RevCPU[cpu:LoadElf:0]: Symbol Table Entry [Scrt1.o:0x0]
RevCPU[cpu:LoadElf:0]: Symbol Table Entry [_DYNAMIC:0x1e00]
RevCPU[cpu:LoadElf:0]: Symbol Table Entry [_GLOBAL_OFFSET_TABLE_:0x2028]
RevCPU[cpu:LoadElf:0]: Symbol Table Entry [_IO_stdin_used:0x934]
RevCPU[cpu:LoadElf:0]: Symbol Table Entry [_ITM_deregisterTMCloneTable:0x0]
RevCPU[cpu:LoadElf:0]: Symbol Table Entry [_ITM_registerTMCloneTable:0x0]
RevCPU[cpu:LoadElf:0]: Symbol Table Entry [_PROCEDURE_LINKAGE_TABLE_:0x600]
RevCPU[cpu:LoadElf:0]: Symbol Table Entry [_ZNKSt13__atomic_baseIlEcvlEv:0x87c]
RevCPU[cpu:LoadElf:0]: Symbol Table Entry [_ZNSt13__atomic_baseIlEaSEl:0x804]
RevCPU[cpu:LoadElf:0]: Symbol Table Entry [_ZNSt13__atomic_baseIlEppEi:0x8e0]
RevCPU[cpu:LoadElf:0]: Symbol Table Entry [_ZStanSt12memory_orderSt23__memory_order_modifier:0x7c0]
RevCPU[cpu:LoadElf:0]: Symbol Table Entry [__BSS_END__:0x2060]
RevCPU[cpu:LoadElf:0]: Symbol Table Entry [__DATA_BEGIN__:0x2000]
RevCPU[cpu:LoadElf:0]: Symbol Table Entry [__FRAME_END__:0xa60]
RevCPU[cpu:LoadElf:0]: Symbol Table Entry [__GNU_EH_FRAME_HDR:0x938]
RevCPU[cpu:LoadElf:0]: Symbol Table Entry [__SDATA_BEGIN__:0x2058]
RevCPU[cpu:LoadElf:0]: Symbol Table Entry [__TMC_END__:0x2008]
RevCPU[cpu:LoadElf:0]: Symbol Table Entry [__abi_tag:0x2b8]
RevCPU[cpu:LoadElf:0]: Symbol Table Entry [__bss_start:0x2058]
RevCPU[cpu:LoadElf:0]: Symbol Table Entry [__cxa_finalize@GLIBC_2.27:0x0]
RevCPU[cpu:LoadElf:0]: Symbol Table Entry [__data_start:0x2000]
[... 527 lines cut ...]
RevCPU[cpu:DecodeInst:112000]: Core 0 ; Thread 0; PC:InstPayload = 0x740:0xf037b3
RevCPU[cpu:ClockTick:112000]: Core 0; Hart 0; Thread 1; Executing PC= 0x740
RevCPU[cpu:ClockTick:112000]: Core 0 ; ThreadID 0; Retiring PC= 0x740
RevCPU[cpu:DecodeInst:113000]: Core 0 ; Thread 0; PC:InstPayload = 0x744:0xff7f793
RevCPU[cpu:ClockTick:113000]: Core 0; Hart 0; Thread 1; Executing PC= 0x744
RevCPU[cpu:ClockTick:113000]: Core 0 ; ThreadID 0; Retiring PC= 0x744
RevCPU[cpu:DecodeInst:114000]: Core 0 ; Thread 0; PC:InstPayload = 0x748:0x78463
RevCPU[cpu:ClockTick:114000]: Core 0; Hart 0; Thread 1; Executing PC= 0x748
RevCPU[cpu:ClockTick:114000]: Core 0 ; ThreadID 0; Retiring PC= 0x748
FATAL: RevCPU[cpu:DecodeInst:115000]: Error: Core 0 failed to decode instruction at PC=0x74c; Inst=0
SST Fatal Backtrace Information:
    0 : sst(_ZNK3SST6Output5fatalEjPKcS2_iS2_z+0x4ae) [0x555ae57accae]
    1 : /home/dev/riscv/off/rev3/devel/src/librevcpu.so(_ZN3SST6RevCPU7RevProc10DecodeInstEv+0x1ab) [0x7ff76d4a0dbb]
    2 : /home/dev/riscv/off/rev3/devel/src/librevcpu.so(_ZN3SST6RevCPU7RevProc9ClockTickEm+0xa74) [0x7ff76d4a2084]
    3 : /home/dev/riscv/off/rev3/devel/src/librevcpu.so(_ZN3SST6RevCPU6RevCPU9clockTickEm+0x143) [0x7ff76d44ca73]
    4 : sst(_ZN3SST5Clock7executeEv+0x10c) [0x555ae5738e4c]
    5 : sst(_ZN3SST15Simulation_impl3runEv+0x3d3) [0x555ae57d6513]
    6 : sst(+0xeebfe) [0x555ae571ebfe]
    7 : sst(main+0x1853) [0x555ae5704ce3]
    8 : /lib/x86_64-linux-gnu/libc.so.6(+0x29d90) [0x7ff77203ad90]
    9 : /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0x80) [0x7ff77203ae40]
   10 : sst(_start+0x25) [0x555ae571c535]
Abort(-1) on node 0 (rank 0 in comm 0): application called MPI_Abort(MPI_COMM_WORLD, -1) - process 0

```

jleidel, 2023-10-10:
fix coming in 99af375

jleidel, 2023-10-10:
this should be fixed now; added 64bit types to the tests

ukasz, 2023-10-25:
Fix confirmed, closing

closed 2023-10-25 by ukasz