tactcomplabs/rev #151: memh_1 test goes into an infinite loop

ajmandadi, 2023-10-05:
**Describe the bug**
On the latest devel branch, possibly after the hazard implementation changes were made, there are certain scenarios under which previously working tests, start to stall and keep looping on a set of instructions - trying to complete them. This is very more consistently seen when memH is enabled on rev.

**To Reproduce**
SST version: 13.0.0
SST --CXX: g++
SST --ELEMENT_CXXFLAGS: -std=c++17 -fPIC -DHAVE_CONFIG_H -I/opt/SST/13.0.0/include
Branch: devel
Last commit: 5c89acd
Rev compiler: gcc (GCC) 8.5.0 20210514 (Red Hat 8.5.0-20)
Riscv compiler: riscv64-unknown-elf-gcc (g2ee5e430018) 12.2.0
Python test file: rev/test/memh_1/rev-test-memh1.py

**Expected behavior**
Simulation ends giving all stats enabled.

**Trace**
RevCPU[cpu:DecodeInst:14202500]: Core 0 ; Thread 0; PC:InstPayload = 0x1020c:0xfae7dce3
RevCPU[cpu:DecodeInst:14203000]: Core 0 ; Thread 0; PC:InstPayload = 0x1020c:0xfae7dce3
RevCPU[cpu:DecodeInst:14203500]: Core 0 ; Thread 0; PC:InstPayload = 0x1020c:0xfae7dce3
RevCPU[cpu:DecodeInst:14204000]: Core 0 ; Thread 0; PC:InstPayload = 0x1020c:0xfae7dce3
RevCPU[cpu:DecodeInst:14204500]: Core 0 ; Thread 0; PC:InstPayload = 0x1020c:0xfae7dce3
RevCPU[cpu:DecodeInst:14205000]: Core 0 ; Thread 0; PC:InstPayload = 0x1020c:0xfae7dce3
RevCPU[cpu:DecodeInst:14205500]: Core 0 ; Thread 0; PC:InstPayload = 0x1020c:0xfae7dce3
RevCPU[cpu:DecodeInst:14206000]: Core 0 ; Thread 0; PC:InstPayload = 0x1020c:0xfae7dce3
RevCPU[cpu:DecodeInst:14206500]: Core 0 ; Thread 0; PC:InstPayload = 0x1020c:0xfae7dce3
RevCPU[cpu:DecodeInst:14207000]: Core 0 ; Thread 0; PC:InstPayload = 0x1020c:0xfae7dce3
RevCPU[cpu:ClockTick:14207000]: Core 0; Hart 0; Thread 1; Executing PC= 0x1020c
RevCPU[cpu:ClockTick:14207000]: Core 0 ; ThreadID 0; Retiring PC= 0x1020c
RevCPU[cpu:DecodeInst:14207500]: Core 0 ; Thread 0; PC:InstPayload = 0x101c4:0xfe842783
RevCPU[cpu:ClockTick:14207500]: Core 0; Hart 0; Thread 1; Executing PC= 0x101c4
RevCPU[cpu:ClockTick:14207500]: Core 0 ; ThreadID 0; Retiring PC= 0x101c4
RevCPU[cpu:DecodeInst:14208000]: Core 0 ; Thread 0; PC:InstPayload = 0x101c8:0x278713
RevCPU[cpu:DecodeInst:14208500]: Core 0 ; Thread 0; PC:InstPayload = 0x101c8:0x278713
RevCPU[cpu:DecodeInst:14209000]: Core 0 ; Thread 0; PC:InstPayload = 0x101c8:0x278713
RevCPU[cpu:DecodeInst:14209500]: Core 0 ; Thread 0; PC:InstPayload = 0x101c8:0x278713
RevCPU[cpu:DecodeInst:14210000]: Core 0 ; Thread 0; PC:InstPayload = 0x101c8:0x278713
RevCPU[cpu:DecodeInst:14210500]: Core 0 ; Thread 0; PC:InstPayload = 0x101c8:0x278713
RevCPU[cpu:DecodeInst:14211000]: Core 0 ; Thread 0; PC:InstPayload = 0x101c8:0x278713
RevCPU[cpu:DecodeInst:14211500]: Core 0 ; Thread 0; PC:InstPayload = 0x101c8:0x278713
RevCPU[cpu:DecodeInst:14212000]: Core 0 ; Thread 0; PC:InstPayload = 0x101c8:0x278713
.
.
.
.
.
.
.
[Keeps looping]

**Additional context**
Add any other context about the problem here.

jleidel, 2023-10-06:
can you disassemble the binary and provide a dump of what is at 0x101c8 (and surrounding instructions)?

ajmandadi, 2023-10-06:
Not able to Reproduce.
Will keep you posted incase i run into it again while testing. Thanks.

closed 2023-10-06 by ajmandadi