tactcomplabs/rev #158: [P1] REV model ignores "startAddr" value in config file.

marcinkoss, 2023-10-11:
Latest devel branch (sha fca545cb1dadf1ea523a456082f78cf73d3ac708)

Specify "startAddr" in config file other than 0x0 and main() symbol.
Expected result - code exeuction will start from specified address
Actual result - simulation starts from main or crashes when there is no main symbol.

from REV code, it seems that   Opts->GetStartAddr(id, StartAddr); is missing after 'uint64_t StartAddr = 0x00ull;"
```
  // set the pc
  uint64_t StartAddr = 0x00ull;

  std::string StartSymbol = "main";
  if( StartAddr == 0x00ull ){
     if( !Opts->GetStartSymbol( id, StartSymbol ) ){
       output.fatal(CALL_INFO, -1,
                     "Error: failed to init the start symbol address for main thread=\n");
     }
    //StartAddr = Loader->GetSymbolAddr(StartSymbol);
  }
  if( StartAddr == 0x00ull ){
    // load "main" symbol
    StartAddr = Loader->GetSymbolAddr("main");
    if( StartAddr == 0x00ull ){
      output.fatal(CALL_INFO, -1,
                   "Error: failed to auto discover address for <main> for main thread\n");
    }
  }
```

jleidel, 2023-10-25:
please confirm this is working and we'll close

marcinkoss, 2023-10-26:
It's working now. Closing.

closed 2023-10-26 by marcinkoss