tactcomplabs/rev #169: [P0] FATAL: RevCPU[cpu:CalcPhysAddr:12000]: Segmentation Fault: Virtual address 0xfffffffe00000000 (PhysAddr = 0xffffffffffffffff) was not found in any mem segment

marcinkoss, 2023-10-19:
For now I do not have any workaround for that and similar behaviour can be seen in many tests.
Please note 'args' in config file is set to 'x'.

Build with gnu gcc or llvm clang code:

```
/* Copyright (c) Intel Corporation */
#define assert(x) if (!(x)) { asm(".dword 0x00000000"); }
#include <string.h>
int main(int argc, char *argv[]) {
        assert(argv[1][0] == 'x');
        return 0;
}

```
e.g.
clang arg.c -o args.exe
or
riscv64-linux-gnu-gcc args.c -o argc.exe

Run as:
REV_EXE=args.exe sst rev_config.py.
Config described below.

**Expected behavior**
Code should run with any trouble

**Trace**
Application fails with:
```
FATAL: RevCPU[cpu:CalcPhysAddr:12000]: Segmentation Fault: Virtual address 0xfffffffe00000000 (PhysAddr = 0xffffffffffffffff) was not found in any mem segments
SST Fatal Backtrace Information:
    0 : sst(_ZNK3SST6Output5fatalEjPKcS2_iS2_z+0x4ae) [0x56262eb5bc6e]
    1 : /home/dev/02_update_rev_devel/tigre/frameworks.simulators.agile.rev/src/librevcpu.so(_ZN3SST6RevCPU6RevMem12CalcPhysAddrEmm+0x39d) [0x7f98dcaff61d]
    2 : /home/dev/02_update_rev_devel/tigre/frameworks.simulators.agile.rev/src/librevcpu.so(_ZN3SST6RevCPU6RevMem7ReadMemEjmmPvRKNS0_6MemReqEj+0xc1) [0x7f98dcb00e11]
    3 : /home/dev/02_update_rev_devel/tigre/frameworks.simulators.agile.rev/src/librevcpu.so(_ZN3SST6RevCPU4loadIhEEbPNS0_10RevFeatureEPNS0_10RevRegFileEPNS0_6RevMemENS0_7RevInstE+0x336) [0x7f98dcb55906]
    4 : /home/dev/02_update_rev_devel/tigre/frameworks.simulators.agile.rev/src/librevcpu.so(_ZN3SST6RevCPU6RevExt7ExecuteEjRKNS0_7RevInstEtPNS0_10RevRegFileE+0x9d) [0x7f98dcaf101d]
    5 : /home/dev/02_update_rev_devel/tigre/frameworks.simulators.agile.rev/src/librevcpu.so(_ZN3SST6RevCPU7RevProc9ClockTickEm+0x4cd) [0x7f98dcb37b2d]
    6 : /home/dev/02_update_rev_devel/tigre/frameworks.simulators.agile.rev/src/librevcpu.so(_ZN3SST6RevCPU6RevCPU9clockTickEm+0x183) [0x7f98dcade733]
    7 : sst(_ZN3SST5Clock7executeEv+0x10c) [0x56262eaec9cc]
    8 : sst(_ZN3SST15Simulation_impl3runEv+0x3d3) [0x56262eb854d3]
    9 : sst(+0xf808e) [0x56262eacf08e]
   10 : sst(main+0x186b) [0x56262eab4e0b]
   11 : /lib/x86_64-linux-gnu/libc.so.6(+0x29d90) [0x7f98e659fd90]
   12 : /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0x80) [0x7f98e659fe40]
   13 : sst(_start+0x25) [0x56262eacc4b5]
--------------------------------------------------------------------------
MPI_ABORT was invoked on rank 0 in communicator MPI_COMM_WORLD
with errorcode 11.

NOTE: invoking MPI_ABORT causes Open MPI to kill all MPI processes.
You may or may not see output from other processes, depending on
exactly when Open MPI kills them.
--------------------------------------------------------------------------
```

Now,
I modified REV code to print more info and re-compiled it with _REV_DEVEL_ and it seems that something is wrong with address
0xfffffffe00000000

Main arguments are written at 0xfffffffe, but later execution tries to load it from 0xfffffffe00000000, not sure why
```
Access for page:16383 addrShift:18 vAddr: 0xfffffaa8 PhsyAddr: 0x3faa8 Next Page: 2
Writing 2 Bytes Starting at 0xfffffffe
        Data > x^@
Access for page:16383 addrShift:18 vAddr: 0xfffffffe PhsyAddr: 0x3fffe Next Page: 2
Writing 9 Bytes Starting at 0xfffffff5
        Data > args.exe^@
```

config file - removed some common things:

```
program = os.getenv("REV_EXE", "xxx")

max_addr_gb = 1

# Define the simulation components
comp_cpu = sst.Component("cpu", "revcpu.RevCPU")
comp_cpu.addParams({
    "verbose": verbose_level,                    # Verbosity
    "numCores": 1,                               # Number of cores
    "clock": "1.0GHz",                           # Clock
    "memSize": 4096*1024*1024,                   # Memory size in bytes
    "machine": "[0:RV64IMAFD]",                  # Core:Config
    "memCost": "[0:1:10]",                       # Memory loads required 1-10 cycles
    "program": program,                          # Target executable
    "splash": 0,                                 # Display the splash message
    **"args": "x"**
})

```

marcinkoss, 2023-10-19:
For some reason address '0xfffffffe00000000' has is 'longer' than other addresses in log file.

```
Access for page:0 addrShift:18 vAddr: 0x112bc PhsyAddr: 0x512bc Next Page: 2
Writing 8 Bytes Starting at 0xfffffa70
Access for page:16383 addrShift:18 vAddr: 0xfffffa70 PhsyAddr: 0x3fa70 Next Page: 2
Writing 8 Bytes Starting at 0xfffffa68
Access for page:16383 addrShift:18 vAddr: 0xfffffa68 PhsyAddr: 0x3fa68 Next Page: 2
Writing 4 Bytes Starting at 0xfffffa64
Access for page:16383 addrShift:18 vAddr: 0xfffffa64 PhsyAddr: 0x3fa64 Next Page: 2
Writing 4 Bytes Starting at 0xfffffa60
Access for page:16383 addrShift:18 vAddr: 0xfffffa60 PhsyAddr: 0x3fa60 Next Page: 2
Writing 8 Bytes Starting at 0xfffffa58
Access for page:16383 addrShift:18 vAddr: 0xfffffa58 PhsyAddr: 0x3fa58 Next Page: 2
NEW READMEM: Reading 8 Bytes Starting at 0xfffffa58
NEW READMEM: Reading 8 Bytes Starting at 0xfffffabc
Access for page:16383 addrShift:18 vAddr: 0xfffffabc PhsyAddr: 0x3fabc Next Page: 2
NEW READMEM: Reading 1 Bytes Starting at 0xfffffffe00000000
```

leekillough, 2023-10-19:
> For some reason address '0xfffffffe00000000' has is 'longer' than other addresses in log file. Address is similar what we are expecting but shifted by ~64bits~ 32 bits somehow.

It is probably a 32-bit/64-bit mismatch somewhere, either in the message `printf` format string, in the `RevLoader`, or somewhere else in the simulator.

RISC-V and an x86-64 host are little endian, but if this is run on a big endian host, and 64 bits are printed or read somewhere when only 32 should be, that would explain the shift.

jleidel, 2023-10-19:
this is likely an offset issue in the loader, not an addressing issue.

marcinkoss, 2023-10-19:
I agree with  John - it's not printf issue or addressing issue, but something in loader corrupts memory and then later this 'explodes'

vcave, 2023-10-19:
This is related to local changes we have made and does not impact devel. I think you can close this.

marcinkoss, 2023-10-20:
@vcave this bug is quite tricky and it can be reproduced without our internal changes as well.
Problem is that if you use different SHA then it might not reproduce on one test and start to reproduce on the other one.

With above test - in fact issue does not reproduce on commit:
```
commit e4ed55a37047bd488972398103e92a9ea26f6509 (HEAD -> rev-devel, origin/rev-devel)
Merge: 89baec0 9bc9ccf
Author: Ryan Kabrick <31963805+rkabrick@users.noreply.github.com>
Date:   Thu Oct 19 10:18:07 2023 -0400

    Merge pull request #167 from tactcomplabs/syscall-fixes

    Syscall fixes

```

but on this commit it can be reproduced with (and it was crashing for me earlier as well):
```
#include <iostream>

int main() {
        std::cout << std::endl;
        return 0;
}

```

Output:
```
WARNING: Building component "cpu" with no links assigned.
RevCPU[cpu:RevCPU:0]: Building Rev with 1 cores and 1 hart(s) on each core
RevCPU[cpu:RevCPU:0]: Initialization of RevCPUs complete.
 | BaseAddr:  0xfffffc00 | TopAddr: 0x100000000 | Size: 1024 Bytes
 | BaseAddr:  0x10040 | TopAddr: 0x11040 | Size: 4096 Bytes
 | BaseAddr:  0x10000 | TopAddr: 0x3f000 | Size: 192512 Bytes
 | BaseAddr:  0x3f2b0 | TopAddr: 0xc32b0 | Size: 540672 Bytes
 | BaseAddr:  0xc3a08 | TopAddr: 0xc4a08 | Size: 4096 Bytes
 | BaseAddr:  0xc4a10 | TopAddr: 0x4c8a10 | Size: 4210688 Bytes
 | BaseAddr:  0x0 | TopAddr: 0x1000 | Size: 4096 Bytes
 | BaseAddr:  0xffeffc00 | TopAddr: 0xfffffc00 | Size: 1048576 Bytes
FATAL: RevCPU[cpu:CalcPhysAddr:32000]: Segmentation Fault: Virtual address 0xffffffffffffffe8 (PhysAddr = 0xffffffffffffffff) was not found in any mem segments
SST Fatal Backtrace Information:
    0 : sst(_ZNK3SST6Output5fatalEjPKcS2_iS2_z+0x4ae) [0x55d723b30c6e]
    1 : rev/src/librevcpu.so(_ZN3SST6RevCPU6RevMem12CalcPhysAddrEmm+0x465) [0x7f4db7f3cd85]
    2 : rev/src/librevcpu.so(_ZN3SST6RevCPU6RevMem7ReadMemEjmmPvRKNS0_6MemReqEj+0x3e) [0x7f4db7f3db2e]
    3 : rev/src/librevcpu.so(_ZN3SST6RevCPU4loadIlEEbPNS0_10RevFeatureEPNS0_10RevRegFileEPNS0_6RevMemENS0_7RevInstE+0x2b7) [0x7f4db7f94827]
    4 : rev/src/librevcpu.so(_ZN3SST6RevCPU6RevExt7ExecuteEjRKNS0_7RevInstEtPNS0_10RevRegFileE+0x9d) [0x7f4db7f2ea4d]
    5 : rev/src/librevcpu.so(_ZN3SST6RevCPU7RevProc9ClockTickEm+0x4ad) [0x7f4db7f72cfd]
    6 : rev/src/librevcpu.so(_ZN3SST6RevCPU6RevCPU9clockTickEm+0x183) [0x7f4db7f1c213]
    7 : sst(_ZN3SST5Clock7executeEv+0x10c) [0x55d723ac19cc]
    8 : sst(_ZN3SST15Simulation_impl3runEv+0x3d3) [0x55d723b5a4d3]
    9 : sst(+0xf808e) [0x55d723aa408e]
   10 : sst(main+0x186b) [0x55d723a89e0b]
   11 : /lib/x86_64-linux-gnu/libc.so.6(+0x29d90) [0x7f4dc5b6ed90]
   12 : /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0x80) [0x7f4dc5b6ee40]
   13 : sst(_start+0x25) [0x55d723aa14b5]

```

I checked with llvm and gnu.

```

$> rev-tests/test/cpp# riscv64-linux-gnu-g++ cout.cpp -static

$> rev-tests/test/cpp# REV_EXE=a.out sst $SST_CONFIG
WARNING: Building component "cpu" with no links assigned.
RevCPU[cpu:RevCPU:0]: Building Rev with 1 cores and 1 hart(s) on each core
RevCPU[cpu:RevCPU:0]: Initialization of RevCPUs complete.
 | BaseAddr:  0xfffffc00 | TopAddr: 0x100000000 | Size: 1024 Bytes
 | BaseAddr:  0x10000 | TopAddr: 0x129000 | Size: 1150976 Bytes
 | BaseAddr:  0x129628 | TopAddr: 0x13f628 | Size: 90112 Bytes
 | BaseAddr:  0xffeffb3c | TopAddr: 0xfffffc00 | Size: 1048772 Bytes
FATAL: RevCPU[cpu:CalcPhysAddr:17000]: Segmentation Fault: Virtual address 0xffffffffffffffe8 (PhysAddr = 0xffffffffffffffff) was not found in any mem segments
SST Fatal Backtrace Information:
    0 : sst(_ZNK3SST6Output5fatalEjPKcS2_iS2_z+0x4ae) [0x55dac3d04c6e]
    1 : rev/src/librevcpu.so(_ZN3SST6RevCPU6RevMem12CalcPhysAddrEmm+0x465) [0x7f2aad2e0d85]
    2 : rev/src/librevcpu.so(_ZN3SST6RevCPU6RevMem7ReadMemEjmmPvRKNS0_6MemReqEj+0x3e) [0x7f2aad2e1b2e]
    3 : rev/src/librevcpu.so(_ZN3SST6RevCPU4loadIlEEbPNS0_10RevFeatureEPNS0_10RevRegFileEPNS0_6RevMemENS0_7RevInstE+0x2b7) [0x7f2aad338827]
    4 : rev/src/librevcpu.so(_ZN3SST6RevCPU6RevExt7ExecuteEjRKNS0_7RevInstEtPNS0_10RevRegFileE+0x9d) [0x7f2aad2d2a4d]
    5 : rev/src/librevcpu.so(_ZN3SST6RevCPU7RevProc9ClockTickEm+0x4ad) [0x7f2aad316cfd]
    6 : rev/src/librevcpu.so(_ZN3SST6RevCPU6RevCPU9clockTickEm+0x183) [0x7f2aad2c0213]
    7 : sst(_ZN3SST5Clock7executeEv+0x10c) [0x55dac3c959cc]
    8 : sst(_ZN3SST15Simulation_impl3runEv+0x3d3) [0x55dac3d2e4d3]
    9 : sst(+0xf808e) [0x55dac3c7808e]
   10 : sst(main+0x186b) [0x55dac3c5de0b]
   11 : /lib/x86_64-linux-gnu/libc.so.6(+0x29d90) [0x7f2ab6e6fd90]
   12 : /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0x80) [0x7f2ab6e6fe40]
   13 : sst(_start+0x25) [0x55dac3c754b5]
```

marcinkoss, 2023-10-20:
g++ version I used:

```
# riscv64-linux-gnu-g++ --version
riscv64-linux-gnu-g++ (Ubuntu 11.4.0-1ubuntu1~22.04) 11.4.0
Copyright (C) 2021 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

leekillough, 2023-11-02:
Closing per above comments

closed 2023-11-02 by leekillough