tactcomplabs/rev #196: Off by one in mmap size region calculation

ukasz, 2023-11-03:
**Describe the bug**
When performing mmap of a 4KB page we expect that addresses from base + 0 through base + 4095 will be accessible, but addresses above that will no (assuming there are no valid mappings there).
In practice we can touch base + 4096, and an error is detected only when touching base + 4097 address.
This implies that either memory access check is not performed correctly or the region size during mapping was not calculated correctly. REV logs indicates the latter is likely the problem here - see full log at the bottom.
```
 | BaseAddr:  0x7051 | TopAddr: 0x8052 | Size: 4097 Bytes
```

**To Reproduce**
Compile the following program:
```
// riscv64-linux-gnu-gcc -march=rv64imafd munmap.c -o munmap.exe
#include "../../../common/syscalls/syscalls.h"
#include "unistd.h"

int main() {
  char *addr;
  addr = (char*)rev_mmap(0, 4096, PROT_READ | PROT_WRITE | PROT_EXEC, MAP_PRIVATE | MAP_ANONYMOUS, -1, 0);
  addr[4096] = 0; // this should segfault, but it does not
  addr[4097] = 0; // this segfaults, which implies off by one at the end of the memory region
  return 0;
}
```
REV config:
```
#
# Copyright (C) 2017-2023 Tactical Computing Laboratories, LLC
# All Rights Reserved
# contact@tactcomplabs.com
#
# See LICENSE in the top level directory for licensing details
#
# rev-test-ex1.py
#

import os
import sst

# Define SST core options
sst.setProgramOption("timebase", "1ps")

# Tell SST what statistics handling we want
sst.setStatisticLoadLevel(8)

max_addr_gb = 1

# Define the simulation components
comp_cpu = sst.Component("cpu", "revcpu.RevCPU")
comp_cpu.addParams({
        "verbose" : 10,                                # Verbosity
        "numCores" : 1,                               # Number of cores
        "clock" : "1.0GHz",                           # Clock
        "memSize" : 1024*1024*1024,                   # Memory size in bytes
        "machine" : "[0:RV64IMAFDC]",                      # Core:Config; RV64I for core 0
        "startAddr" : "[0:0x00000000]",               # Starting address for core 0
        "memCost" : "[0:1:10]",                       # Memory loads required 1-10 cycles
        "program" : "munmap.exe",  # Target executable
        "splash" : 1                                  # Display the splash message
})

# sst.setStatisticOutput("sst.statOutputCSV")
sst.enableAllStatisticsForAllComponents()

# EOF
```

REV VERSION: c4f21a7cb9066b

**Expected behavior**
Mapped region should have size of requested size, not requested size + 1.
[... 661 lines cut ...]
FATAL: RevCPU[cpu:CalcPhysAddr:45000]: Segmentation Fault: Virtual address 0x8052 (PhysAddr = 0xffffffffffffffff) was not found in any mem segments
SST Fatal Backtrace Information:
    0 : sst(_ZNK3SST6Output5fatalEjPKcS2_iS2_z+0x4ae) [0x55d987750c6e]
    1 : /usr/local/lib/sst-elements-library/librevcpu.so(_ZN3SST6RevCPU6RevMem12CalcPhysAddrEmm+0x465) [0x7f60c9e1f9c5]
    2 : /usr/local/lib/sst-elements-library/librevcpu.so(_ZN3SST6RevCPU6RevMem8WriteMemEjmmPKv+0x6d) [0x7f60c9e2027d]
    3 : /usr/local/lib/sst-elements-library/librevcpu.so(_ZN3SST6RevCPU5storeIhEEbPNS0_10RevFeatureEPNS0_10RevRegFileEPNS0_6RevMemENS0_7RevInstE+0xb0) [0x7f60c9e600b0]
    4 : /usr/local/lib/sst-elements-library/librevcpu.so(_ZN3SST6RevCPU6RevExt7ExecuteEjRKNS0_7RevInstEtPNS0_10RevRegFileE+0x9d) [0x7f60c9e1041d]
    5 : /usr/local/lib/sst-elements-library/librevcpu.so(_ZN3SST6RevCPU7RevProc9ClockTickEm+0x374) [0x7f60c9e58634]
    6 : /usr/local/lib/sst-elements-library/librevcpu.so(_ZN3SST6RevCPU6RevCPU9clockTickEm+0x1c3) [0x7f60c9de9a73]
    7 : sst(_ZN3SST5Clock7executeEv+0x10c) [0x55d9876e19cc]
    8 : sst(_ZN3SST15Simulation_impl3runEv+0x3d3) [0x55d98777a4d3]
    9 : sst(+0xf808e) [0x55d9876c408e]
   10 : sst(main+0x186b) [0x55d9876a9e0b]
   11 : /lib/x86_64-linux-gnu/libc.so.6(+0x29d90) [0x7f60d3794d90]
   12 : /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0x80) [0x7f60d3794e40]
   13 : sst(_start+0x25) [0x55d9876c14b5]
--------------------------------------------------------------------------
MPI_ABORT was invoked on rank 0 in communicator MPI_COMM_WORLD
with errorcode 11.

NOTE: invoking MPI_ABORT causes Open MPI to kill all MPI processes.
You may or may not see output from other processes, depending on
exactly when Open MPI kills them.

```

rkabrick, 2023-11-10:
Apologies for the delay. I'm now back from vacation. 

https://github.com/tactcomplabs/rev/pull/204

rkabrick, 2023-11-10:
Just merged. Can you confirm fix and close? @ukasz

ukasz, 2023-11-13:
It works fine on 647cf2f22f thank you - closing it.

closed 2023-11-13 by ukasz