tactcomplabs/rev #197: Odd segfault at the end of .text section

ukasz, 2023-11-06:
**Describe the bug**

We observe this behavior on multiple occassions, but it is not quite clear what is the root cause.
The problem doesn't happen on a particular instruction, but rather at the particular offset group.
When .text size is slightly below 0x1000 (0xffc, 0xff8, 0xff4) REV reports a segmentation fault at the address which is technically at the end of .text section.
Similar behavior is observed when .text size is around muliples of a 0x1000.
The odd thing is that I am not able to reproduce this issue with GCC compiler, it only happens on LLVM.
There are two main differences I can see between code generated by those compilers a) .text offset address and b) slightly different prologue/epilogue.
Given the fact the the problem doesn't occur on any particular instruction I assume that b) has probably no effect.
The easiest way to reproduce the problem is to create a binary with main function filled with NOP instructions to the point
where .text section size is 0xff4-0xffc.

**To Reproduce**
- The source code branch being used rev-devel
- The last commit hash: c4f21a7cb906
- The compiler utilized to build Rev: gcc version 11.4.0 (Ubuntu 11.4.0-1ubuntu1~22.04)
- The compiler utilized to build the test: Intel LLVM at beed157291c05

Build the following source code:
```
//clang -march=rv64gxinteltigrea -O0 -o sum.exe sum.c -nostdlib
#define NOP asm("nop;");
#define NOP2 NOP NOP
#define NOP4 NOP2 NOP2
#define NOP8 NOP4 NOP4
#define NOP16 NOP8 NOP8
#define NOP32 NOP16 NOP16
#define NOP64 NOP32 NOP32
#define NOP128 NOP64 NOP64
#define NOP256 NOP128 NOP128
#define NOP512 NOP256 NOP256

int main() {
        NOP512
        NOP256
        NOP128
        NOP64
        NOP32
        NOP16
        NOP4
        NOP
		return 0;
}
```
Lets confirm .text size of 0xffc
```
llvm-readelf --sections sum.exe | grep text
  [ 3] .text             PROGBITS        00000000000111e4 0001e4 000ffc 00  AX  0   0  4
```
Entry point is main:
```
llvm-readelf -s sum.exe | grep main
     3: 00000000000111e4  4092 FUNC    GLOBAL DEFAULT     3 main
```
Now run it with python config file >

** Python Config File**
```
import os
import sst
import sys

program = os.getenv("REV_EXE", "sum.exe")
ep = "[0:0x111e4]" # main address

# Define SST core options
sst.setProgramOption("timebase", "1ps")
sst.setStatisticLoadLevel(4)

max_addr_gb = 1
[... 306 lines cut ...]
 | BaseAddr:  0x8ffc00 | TopAddr: 0x9ffc00 | Size: 1048576 Bytes
FATAL: RevCPU[cpu:CalcPhysAddr:1022000]: Segmentation Fault: Virtual address 0x121e4 (PhysAddr = 0xffffffffffffffff) was not found in any mem segments
SST Fatal Backtrace Information:
    0 : sst(_ZNK3SST6Output5fatalEjPKcS2_iS2_z+0x4ae) [0x55e9cad2ec6e]
    1 : /usr/local/lib/sst-elements-library/librevcpu.so(_ZN3SST6RevCPU6RevMem12CalcPhysAddrEmm+0x465) [0x7f470484b9c5]
    2 : /usr/local/lib/sst-elements-library/librevcpu.so(_ZN3SST6RevCPU6RevMem7ReadMemEjmmPvRKNS0_6MemReqEj+0x3e) [0x7f470484c78e]
    3 : /usr/local/lib/sst-elements-library/librevcpu.so(_ZN3SST6RevCPU13RevPrefetcher4FillEm+0x4db) [0x7f47048b33fb]
    4 : /usr/local/lib/sst-elements-library/librevcpu.so(_ZN3SST6RevCPU13RevPrefetcher4FillEm+0x674) [0x7f47048b3594]
    5 : /usr/local/lib/sst-elements-library/librevcpu.so(_ZN3SST6RevCPU13RevPrefetcher9InstFetchEmRbRj+0x13a) [0x7f47048b397a]
    6 : /usr/local/lib/sst-elements-library/librevcpu.so(_ZN3SST6RevCPU7RevProc10DecodeInstEv+0x5c) [0x7f470488349c]
    7 : /usr/local/lib/sst-elements-library/librevcpu.so(_ZN3SST6RevCPU7RevProc9ClockTickEm+0x8d0) [0x7f4704884b90]
    8 : /usr/local/lib/sst-elements-library/librevcpu.so(_ZN3SST6RevCPU6RevCPU9clockTickEm+0x1c3) [0x7f4704815a73]
    9 : sst(_ZN3SST5Clock7executeEv+0x10c) [0x55e9cacbf9cc]
   10 : sst(_ZN3SST15Simulation_impl3runEv+0x3d3) [0x55e9cad584d3]
   11 : sst(+0xf808e) [0x55e9caca208e]
   12 : sst(main+0x186b) [0x55e9cac87e0b]
   13 : /lib/x86_64-linux-gnu/libc.so.6(+0x29d90) [0x7f4708683d90]
   14 : /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0x80) [0x7f4708683e40]
   15 : sst(_start+0x25) [0x55e9cac9f4b5]
--------------------------------------------------------------------------
MPI_ABORT was invoked on rank 0 in communicator MPI_COMM_WORLD
with errorcode 11.

NOTE: invoking MPI_ABORT causes Open MPI to kill all MPI processes.
You may or may not see output from other processes, depending on

jleidel, 2023-11-06:
This may be due to the prefetcher attempting to fetch beyond the end of the `.text` space.

rkabrick, 2023-11-10:
> This may be due to the prefetcher attempting to fetch beyond the end of the `.text` space.

If this is indeed the prefetcher how would we prevent it from happening?

jleidel, 2023-11-10:
> > This may be due to the prefetcher attempting to fetch beyond the end of the `.text` space.
> 
> If this is indeed the prefetcher how would we prevent it from happening?

We would need to pass the address of the end of the text space down to the prefetcher such that we don't overreach.

leekillough, 2024-07-17:
This is not reproducible anymore; can we close this?

ukasz, 2024-07-22:
I'll try to reproduce it and report back.