tactcomplabs/rev #198: Using snprintf to populate buffer for rev_write segfaults

ajmandadi, 2023-11-06:
**Describe the bug**
To print a variable from inside rev, we use snprintf to populate a buffer. But this causes a segfault inside rev's calcphysaddr.

**To Reproduce** 
- sst --version - SST-Core Version (13.0.0),
- sst-config --CXX - g++
- sst-config --ELEMENT_CXXFLAGS -  -std=c++17 -fPIC -DHAVE_CONFIG_H -I/opt/SST/13.0.0/include
- The source code branch being used - devel
- The last commit hash - 870c8cf5f3e1b3b7c230c3da9f7e30e5351ffc9e
- The compiler utilized to build Rev - gcc (GCC) 8.5.0 20210514 (Red Hat 8.5.0-21)
- The compiler utilized to build the test - riscv64-unknown-elf-gcc (g2ee5e430018) 12.2.0
- The python or JSON simulation input - added as attachment in additional context
- C/CXX test source (or mention which test in the current source tree failed) - added as attachment in additional context

**Expected behavior**
To print the value 98

**BackTrace**
```
RevCPU[cpu:DecodeInst:147000]: Core 0; Hart 0; Thread 1; PC:InstPayload = 0x13afc:0x97ba7722
RevCPU[cpu:clockTick:148000]: Cycle: 148
RevCPU[cpu:DecodeInst:148000]: Core 0; Hart 0; Thread 1; PC:InstPayload = 0x13afe:0x439c97ba
RevCPU[cpu:clockTick:149000]: Cycle: 149
RevCPU[cpu:DecodeInst:149000]: Core 0; Hart 0; Thread 1; PC:InstPayload = 0x13b00:0x8782439c
 | BaseAddr:  0x3ffffc00 | TopAddr: 0x40000000 | Size: 1024 Bytes
 | BaseAddr:  0x10000 | TopAddr: 0x1f000 | Size: 61440 Bytes
 | BaseAddr:  0x1f000 | TopAddr: 0x21000 | Size: 8192 Bytes
 | BaseAddr:  0x3feffc00 | TopAddr: 0x3ffffc00 | Size: 1048576 Bytes
FATAL: RevCPU[cpu:CalcPhysAddr:149000]: Segmentation Fault: Virtual address 0x130 (PhysAddr = 0xffffffffffffffff) was not found in any mem segments
SST Fatal Backtrace Information:
    0 : sst(_ZNK3SST6Output5fatalEjPKcS2_iS2_z+0x3d9) [0x53abf9]
    1 : /nethome/amandadi3/rev_latest/rev/src/librevcpu.so(_ZN3SST6RevCPU6RevMem12CalcPhysAddrEmm+0x3f0) [0x7f5b581d08f8]
    2 : /nethome/amandadi3/rev_latest/rev/src/librevcpu.so(_ZN3SST6RevCPU6RevMem7ReadMemEjmmPvRKNS0_6MemReqEj+0x78) [0x7f5b581d2802]
    3 : /nethome/amandadi3/rev_latest/rev/src/librevcpu.so(_ZN3SST6RevCPU6RevMem7ReadValIjEEbjmPT_RKNS0_6MemReqEj+0x4d) [0x7f5b58246067]
    4 : /nethome/amandadi3/rev_latest/rev/src/librevcpu.so(_ZN3SST6RevCPU4loadIiEEbPNS0_10RevFeatureEPNS0_10RevRegFileEPNS0_6RevMemENS0_7RevInstE+0x426) [0x7f5b58239bf3]
    5 : /nethome/amandadi3/rev_latest/rev/src/librevcpu.so(_ZN3SST6RevCPU5RV32I3clwEPNS0_10RevFeatureEPNS0_10RevRegFileEPNS0_6RevMemENS0_7RevInstE+0x75) [0x7f5b58223667]
    6 : /nethome/amandadi3/rev_latest/rev/src/librevcpu.so(_ZN3SST6RevCPU6RevExt7ExecuteEjRKNS0_7RevInstEtPNS0_10RevRegFileE+0x155) [0x7f5b581c5fd3]
    7 : /nethome/amandadi3/rev_latest/rev/src/librevcpu.so(_ZN3SST6RevCPU7RevProc9ClockTickEm+0xa29) [0x7f5b58218655]
    8 : /nethome/amandadi3/rev_latest/rev/src/librevcpu.so(_ZN3SST6RevCPU6RevCPU9clockTickEm+0xdf) [0x7f5b58159be1]
    9 : /nethome/amandadi3/rev_latest/rev/src/librevcpu.so(_ZN3SST10SSTHandlerIbmNS_6RevCPU6RevCPUEvE13operator_implEm+0x71) [0x7f5b581c4d21]
   10 : sst(_ZN3SST5Clock7executeEv+0x14a) [0x4f484a]
   11 : sst(_ZN3SST15Simulation_impl3runEv+0x1cb) [0x557ceb]
   12 : sst() [0x4dfd01]
   13 : sst(main+0x6eb) [0x4c609b]
   14 : /lib64/libc.so.6(__libc_start_main+0xe5) [0x7f5b6edb2d85]
   15 : sst(_start+0x2e) [0x4def5e]
```

**Additional context**
Attached the binary's objdump as well and looks like the PC during failure (0x13b00) is a LOAD instruction inside the snprint function call. 
Code compiled with below command with compressed instruction enabled - 
`"/opt/riscv/bin/riscv64-unknown-elf-gcc" -march=rv64imafdc -O0 -o revprint.exe rev_print.c -static.`
[code.tar.gz](https://github.com/tactcomplabs/rev/files/13272929/code.tar.gz)

ajmandadi, 2023-11-20:
in this test case, i'm using snprintf which is a compressed instruction so removing the them throws an error that "Compressed instructions not enabled!". Hence unsure if this is also a side effect of compressed instructions problem.

donofrio, 2023-11-21:
When running this test against this branch, which fixes compressed instructions,: https://github.com/tactcomplabs/rev/tree/TestDbg we no longer see the failure. This branch currently has a vexing memory corruption issue preventing it from being merged to devel, but the changes here should fix this issue when merged

ajmandadi, 2023-11-27:
Probably still being worked on but just wanted to update that I tested it out the above branch, and though i dont see a crash anymore, there is no print to console i.e the ECALL to rev_write doesnt seem to have been made.

leekillough, 2024-07-17:
Not reproducible anymore. Probably fixed by https://github.com/tactcomplabs/rev/pull/298, https://github.com/tactcomplabs/rev/pull/296 and https://github.com/tactcomplabs/rev/pull/288.