tactcomplabs/rev #203: munmap test fails with memH enabled

ajmandadi, 2023-11-10:
**Describe the bug**
The test case rev/test/syscalls/munmap fails when memH is enabled. 
Notice two separate errors with cache and without cache plus membackend.

**To Reproduce**
sst --version - SST-Core Version (13.0.0),
sst-config --CXX - g++
sst-config --ELEMENT_CXXFLAGS - -std=c++17 -fPIC -DHAVE_CONFIG_H -I/opt/SST/13.0.0/include
The source code branch being used - devel
The last commit hash - 5a4406e456c2ba3839fccdcf7c521b9871cb2bb0
The compiler utilized to build Rev - gcc (GCC) 8.5.0 20210514 (Red Hat 8.5.0-21)
The compiler utilized to build the test - riscv64-unknown-elf-gcc (g2ee5e430018) 12.2.0
The python or JSON simulation input - added as attachment in additional context
C/CXX test source (or mention which test in the current source tree failed) - same as test present in rev/test/syscalls/munmap 

**Expected behavior**
Test passes successfully by running simulation to completion.

**Trace**
With Cache + membackend
```
RevCPU[cpu:RevCPU:0]: Initialization of RevCPUs complete.
[RevMemCtrl 0]: Detected cache layers; default line size=64
 | BaseAddr:  0x3ffffc00 | TopAddr: 0x40000000 | Size: 1024 Bytes
 | BaseAddr:  0x10000 | TopAddr: 0x14000 | Size: 16384 Bytes
 | BaseAddr:  0x14000 | TopAddr: 0x15000 | Size: 4096 Bytes
 | BaseAddr:  0x3feffc00 | TopAddr: 0x3ffffc00 | Size: 1048576 Bytes
FATAL: RevCPU[cpu:CalcPhysAddr:1969000]: Segmentation Fault: Virtual address 0x18 (PhysAddr = 0xffffffffffffffff) was not found in any mem segments
SST Fatal Backtrace Information:
    0 : sst(_ZNK3SST6Output5fatalEjPKcS2_iS2_z+0x3d9) [0x53abf9]
    1 : /nethome/amandadi3/rev_latest1/rev/src/librevcpu.so(_ZN3SST6RevCPU6RevMem12CalcPhysAddrEmm+0x435) [0x7f0840cea205]
    2 : /nethome/amandadi3/rev_latest1/rev/src/librevcpu.so(_ZN3SST6RevCPU6RevMem8WriteMemEjmmPKv+0x68) [0x7f0840ceaa78]
    3 : /nethome/amandadi3/rev_latest1/rev/src/librevcpu.so(_ZN3SST6RevCPU5storeImEEbPNS0_10RevFeatureEPNS0_10RevRegFileEPNS0_6RevMemENS0_7RevInstE+0xa1) [0x7f0840d1fe11]
    4 : /nethome/amandadi3/rev_latest1/rev/src/librevcpu.so(_ZN3SST6RevCPU6RevExt7ExecuteEjRKNS0_7RevInstEtPNS0_10RevRegFileE+0x99) [0x7f0840ce1d29]
    5 : /nethome/amandadi3/rev_latest1/rev/src/librevcpu.so(_ZN3SST6RevCPU7RevProc9ClockTickEm+0x31a) [0x7f0840d1812a]
    6 : /nethome/amandadi3/rev_latest1/rev/src/librevcpu.so(_ZN3SST6RevCPU6RevCPU9clockTickEm+0x1bb) [0x7f0840cb954b]
    7 : sst(_ZN3SST5Clock7executeEv+0x14a) [0x4f484a]
    8 : sst(_ZN3SST15Simulation_impl3runEv+0x1cb) [0x557ceb]
    9 : sst() [0x4dfd01]
   10 : sst(main+0x6eb) [0x4c609b]
   11 : /lib64/libc.so.6(__libc_start_main+0xe5) [0x7f0857f8dd85]
   12 : sst(_start+0x2e) [0x4def5e]  
```

With Membackend:
```
[RevMemCtrl 0]: No cache detected; disabling caching
RevCPU[cpu:ECALL_mmap:463000]: ECALL: mmap called
FATAL: RevCPU[cpu:FetchAndDecodeInst:98811000]: Error: Core 0 failed to decode instruction at PC=0x1371c; Inst=0
SST Fatal Backtrace Information:
    0 : sst(_ZNK3SST6Output5fatalEjPKcS2_iS2_z+0x3d9) [0x53abf9]
    1 : /nethome/amandadi3/rev_latest1/rev/src/librevcpu.so(_ZN3SST6RevCPU7RevProc18FetchAndDecodeInstEv+0x13f) [0x7f7b65b6708f]
    2 : /nethome/amandadi3/rev_latest1/rev/src/librevcpu.so(_ZN3SST6RevCPU7RevProc9ClockTickEm+0x894) [0x7f7b65b686a4]
    3 : /nethome/amandadi3/rev_latest1/rev/src/librevcpu.so(_ZN3SST6RevCPU6RevCPU9clockTickEm+0x1bb) [0x7f7b65b0954b]
    4 : sst(_ZN3SST5Clock7executeEv+0x14a) [0x4f484a]
    5 : sst(_ZN3SST15Simulation_impl3runEv+0x1cb) [0x557ceb]
    6 : sst() [0x4dfd01]
    7 : sst(main+0x6eb) [0x4c609b]
    8 : /lib64/libc.so.6(__libc_start_main+0xe5) [0x7f7b7ced8d85]
    9 : sst(_start+0x2e) [0x4def5e]
MemHierarchy::MemoryController memory
  Outstanding events: 0
  Link Status:   MemHierarchy::MemLink: No status given
End MemHierarchy::MemoryController
```

**Additional context**
Both configuration files attached. 
[config.tar.gz](https://github.com/tactcomplabs/rev/files/13321406/config.tar.gz)

rkabrick, 2023-11-10:
Not a fix but this test fails with memh only when compressed instructions are enabled. If you change both the python and the Makefile to use `RV64G` its passing on my machine. 

We'll have to investigate what's going on but it seems like it's a separate issue not directly related to the system call itself.

rkabrick, 2023-12-05:
Can you confirm this is fixed?

rkabrick, 2024-02-27:
Closing because fixed

closed 2024-02-27 by rkabrick