tactcomplabs/rev #218: Statistics not begin accumulated

jleidel, 2023-12-01:
**Describe the bug**
Statistics don't appear to be accumulated.  This is like due to some issue in `RevProc` not actually adding statistics values to the executed instructions.  

**To Reproduce**
Run the `big_loop` test from the test harness and look at the CSV output.  It should *not* have zero values in it: 
```
ComponentName, StatisticName, StatisticSubId, StatisticType, SimTime, Rank, Sum.u64, SumSQ.u64, Count.u64, Min.u64, Max.u64
cpu, TotalCycles, core_0, Accumulator, 10229786000, 0, 0, 0, 0, 0, 0
cpu, CyclesWithIssue, core_0, Accumulator, 10229786000, 0, 0, 0, 0, 0, 0
cpu, FloatsRead, core_0, Accumulator, 10229786000, 0, 0, 0, 0, 0, 0
cpu, FloatsWritten, core_0, Accumulator, 10229786000, 0, 0, 0, 0, 0, 0
cpu, DoublesRead, core_0, Accumulator, 10229786000, 0, 0, 0, 0, 0, 0
cpu, DoublesWritten, core_0, Accumulator, 10229786000, 0, 0, 0, 0, 0, 0
cpu, BytesRead, core_0, Accumulator, 10229786000, 0, 0, 0, 0, 0, 0
cpu, BytesWritten, core_0, Accumulator, 10229786000, 0, 0, 0, 0, 0, 0
cpu, FloatsExec, core_0, Accumulator, 10229786000, 0, 0, 0, 0, 0, 0
cpu, TLBHitsPerCore, core_0, Accumulator, 10229786000, 0, 0, 0, 0, 0, 0
cpu, TLBMissesPerCore, core_0, Accumulator, 10229786000, 0, 0, 0, 0, 0, 0
```

**Expected behavior**
Statistics should be generated

**Trace**
If applicable, post a trace of the execution.  

**Additional context**
Add any other context about the problem here.

leekillough, 2023-12-06:
Statistics are accumulated inside of `RevProc` in a `RevProcStats` inner class, but these stats are not registered with SST Core, which outputs the `.csv` files.

Statistics are registered with SST Core inside of `RevCPU`, and the `RevProc`'s `RevProcStats` are added to the `RevCPU` stats at each clock tick, but only if it is `Enabled`, and right now only `cyclesBusy`, `totalCycles`, `floatsExec` and `memStats` are used -- not all of the other `RevProcStats`:
```c++

  struct RevProcStats {
    uint64_t totalCycles;
    uint64_t cyclesBusy;
    uint64_t cyclesIdle_Total;
    uint64_t cyclesStalled;
    uint64_t floatsExec;
    float    percentEff;
    RevMem::RevMemStats memStats;
    uint64_t cyclesIdle_Pipeline;
    uint64_t cyclesIdle_MemoryFetch;
  };
  RevProcStats GetStats() { Stats.memStats = mem->memStats; return Stats; }
```

```c++
bool RevCPU::clockTick( SST::Cycle_t currentCycle ){
  bool rtn = true;

  output.verbose(CALL_INFO, 8, 0, "Cycle: %" PRIu64 "\n", currentCycle);

  // Execute each enabled core
  for( size_t i=0; i<Procs.size(); i++ ){
    // Check if we have more work to assign and places to put it
    UpdateThreadAssignments(i);
    if( Enabled[i] ){
      if( !Procs[i]->ClockTick(currentCycle) ){
        if(EnableCoProc && !CoProcs.empty()){
          CoProcs[i]->Teardown();
        }
        UpdateCoreStatistics(i);                                         <--------------- HERE
        Enabled[i] = false;
        output.verbose(CALL_INFO, 5, 0, "Closing Processor %zu at Cycle: %" PRIu64 "\n",
                       i, currentCycle);
      }
      if(EnableCoProc &&
         !CoProcs[i]->ClockTick(currentCycle) &&
         !DisableCoprocClock){
        output.verbose(CALL_INFO, 5, 0, "Closing Co-Processor %zu at Cycle: %" PRIu64 "\n",
                       i, currentCycle);

      }
    }

    // See if any of the threads on this proc changes state
    HandleThreadStateChangesForProc(i);

    if( Procs[i]->HasNoBusyHarts() ){
      Enabled[i] = false;
    }
  }
```

```c++
void RevCPU::UpdateCoreStatistics(unsigned coreNum){
  RevProc::RevProcStats stats = Procs[coreNum]->GetStats();
  TotalCycles[coreNum]->addData(stats.totalCycles);
  CyclesWithIssue[coreNum]->addData(stats.cyclesBusy);
  FloatsRead[coreNum]->addData(stats.memStats.floatsRead);
  FloatsWritten[coreNum]->addData(stats.memStats.floatsWritten);
  DoublesRead[coreNum]->addData(stats.memStats.doublesRead);
  DoublesWritten[coreNum]->addData(stats.memStats.doublesWritten);
  BytesRead[coreNum]->addData(stats.memStats.bytesRead);
  BytesWritten[coreNum]->addData(stats.memStats.bytesWritten);
  FloatsExec[coreNum]->addData(stats.floatsExec);
  TLBHitsPerCore[coreNum]->addData(stats.memStats.TLBHits);
  TLBMissesPerCore[coreNum]->addData(stats.memStats.TLBMisses);
}
```

The stats do not apparently get cleared after `RevCPU::UpdateCoreStatistics` is called, so if stats are added every clock tick, this may be adding them multiple times.

The zero outputs might mean that the stats output are not ones that we are accumulating. I did not see any difference in the output levels of the stats registered by `RevCPU`, so if we register 30+ stats as we do, we should get more output than the small `.csv` with only a few columns. There's also interdependence between the C++ code and the Python code -- maybe something is wrong with the Python code in choosing how statistics are outputted.

Only the `totalCycles`, `cyclesBusy`, `floatsExec` and `memStats` statistics are transferred from `RevProc` to `RevCPU`. The `cyclesIdle_Total`, `cyclesStalled`, `percentEff`, `cyclesIdle_Pipeline` and `cyclesIdle_MemoryFetch` are collected by `RevProc`, but are not transferred to `RevCPU` at each clock.

I need advice from @rkabrick on how to proceed with statistics. It is not just a simple case of not collecting them.

We also need to know how to clear stats after they are transferred from `RevProc` to `RevCPU` -- the `RevMemStats` member of `RevProcStats` gets copied from  `RevMem` to `RevProcStats` every time `RevCPU` requests stats from `RevProc`. How/when should `RevMemStats` (on the `RevMem` side) and `RevProcStats` be cleared after `RevCPU` transfers the stats from `RevProc`?

leekillough, 2023-12-08:
Fixed by #220

closed 2023-12-08 by leekillough