tactcomplabs/rev #223: fcvt.w.s does not support rounding modes

ukasz, 2023-12-13:
**Describe the bug**
fcvt.w.s instruction supports rounding modes, so we can decide whether round up or down.
On REV this field seems to be ignored resulting in rounding down always.

**To Reproduce**
Compile and run the following code:
```
//clang bug.c -o bug -march=rv64imafd
#define assert(x) if (!(x)) { asm(".dword 0x00000000"); }
int main() {
  float x = 3.14f;
  int result;

  asm volatile("fcvt.w.s %0, %1" : "=r"(result): "f"(x));
  assert(result == 3);

  asm volatile("fcvt.w.s %0, %1, rup" : "=r"(result): "f"(x));
  assert(result == 4);
  return 0;
}
```

** Setup **
Tested based on the following REV commit:
```
Author: Kenneth Griesser <kgriesser@tclcomplabs.com>
Date:   Tue Nov 28 09:46:21 2023 -0700

    bug test for simple struct with constructor
```
**Expected behavior**
Attached program should not fail on invalid instruction assert.

**Additional info**
Please not that there are also other instructions with rm field encoded that may suffer from similar behavior.

leekillough, 2023-12-13:
Rounding modes and FP exceptions is a work currently in progress.

leekillough, 2023-12-22:
Just to update: Rounding modes for all FP instructions, not just `fcvt`, is being developed.

On an IEEE754-compliant host, it is assumed that setting rounding modes for FP32 and FP64 will have the same effect on the host machine as on the RISC-V hardware, with a few _exceptions_ (pun intended) which can be handled in software (such as RISC-V's requirement that `0*Inf + z` and `Inf*0 + z` must raise invalid exceptions in `fmadd`-type instructions regardless of `z`). The rounding mode will be set before every RISC-V instruction is executed.

If a RISC-V instruction can raise FP exceptions, then the FP exceptions are cleared and made non-trapping before the simulation of the RISC-V instruction on the host machine, and any exceptions raised will be detected after the instruction simulation has executed, and will be reflected in the FCSR architectural register.

RISC-V instructions which cannot raise FP exceptions, such as load and store, `FMV`, or `copysign`-like instructions, are not affected by FP rounding flags.

Related: #67

leekillough, 2024-01-08:
https://github.com/tactcomplabs/rev/pull/238

ukasz, 2024-04-08:
I hope this feature is not abandoned, it would be lovely to see at least fcvt rounding mode support merged in the near future, it that is possible. This particular bug affects ceil() and floor() functions in LLVM.

leekillough, 2024-04-08:
It is not abandoned. It is a requirement of the ISA.

`fcvt` is the most complicated instruction of "R"ISC-V.

Feel free to try out the `fenv` branch.

leekillough, 2024-06-25:
Please verify it has been fixed.

leekillough, 2024-06-27:
Fixed in #238

closed 2024-06-27 by leekillough

ukasz, 2024-06-27:
Sorry I could not verify it in a timely manner. Thank you for fixing it.