tactcomplabs/rev #228: A problem with FPU register dependency tracking

ukasz, 2023-12-19:
**Describe the bug**
We found an issue which is reproducible with config using memory hierarchy (tests/memh_2), but not on basic one (tests/ex2).
Failing example is shown below.
```
// clang acc.c -o acc -march=rv64imafd
#include <assert.h>
#define N 3
 
int main() {
        float a[2];
        float res = 0;
        for(int i = 0; i< 2; i++)
                a[i] = i;
        for(int i = 0 ;i < 2 ;i++)
                res+=a[i];
        assert(res != 3.0f); //just to keep optimizer away
        return 0;
}
```
Alternatively you may be able to reproduce it with C++ std::accumulate:
```
template <typename T> T test_accumulate() {
        T tab[N];
        for (int i = 0; i < N; i++)
                tab[i] = i;
        T sum = std::accumulate(tab, tab + N, 0);
        return sum;
}
template <typename T> T expected(T x) {
        return (x*(x-1))/2;
}

int main()
{
        float ret_float = test_accumulate<float>();
        assert(expected<float>(N) == ret_float);
}
```

What the program does is a sequence of stores followed by a sequence of loads and floating point addition.
On basic config calculated sum is equal to the expected value. On memh config it is not.
In dissassembly we have:
```
00000000000120ac <main>:
   120ac: 13 01 01 fd   addi    sp, sp, -48
   120b0: 23 34 11 02   sd      ra, 40(sp)
   120b4: 23 30 81 02   sd      s0, 32(sp)
   120b8: 13 04 01 03   addi    s0, sp, 48
   120bc: 13 05 00 00   li      a0, 0
   120c0: 23 26 a4 fe   sw      a0, -20(s0)
   120c4: 23 20 a4 fe   sw      a0, -32(s0)
   120c8: 23 2e a4 fc   sw      a0, -36(s0)
   120cc: 6f 00 40 00   j       0x120d0 <main+0x24>
   120d0: 83 25 c4 fd   lw      a1, -36(s0)
   120d4: 13 05 10 00   li      a0, 1
   120d8: 63 54 b5 00   bge     a0, a1, 0x120e0 <main+0x34>
   120dc: 6f 00 40 03   j       0x12110 <main+0x64>
   120e0: 6f 00 40 00   j       0x120e4 <main+0x38>
   120e4: 03 25 c4 fd   lw      a0, -36(s0)
   120e8: d3 77 05 d0   fcvt.s.w        fa5, a0
   120ec: 93 15 25 00   slli    a1, a0, 2
   120f0: 13 05 44 fe   addi    a0, s0, -28
   120f4: 33 05 b5 00   add     a0, a0, a1
   120f8: 27 20 f5 00   fsw     fa5, 0(a0)
   120fc: 6f 00 40 00   j       0x12100 <main+0x54>
   12100: 03 25 c4 fd   lw      a0, -36(s0)
   12104: 1b 05 15 00   addiw   a0, a0, 1
   12108: 23 2e a4 fc   sw      a0, -36(s0)
   1210c: 6f f0 5f fc   j       0x120d0 <main+0x24>
```
When analyzing logs the first difference between basic and memh config happens at:
120d0 and then 120e8 where values in destination registers are different.
What I think is happening here is that register tracking does not work correctly when dependence is between RV and FPU registers.

**To Reproduce**
- The source code branch being used rev-devel
- The last commit hash: e77b85e4c00629
- The compiler utilized to build Rev: gcc version 11.4.0 (Ubuntu 11.4.0-1ubuntu1~22.04)
- The compiler utilized to build the test: clang
- The python or JSON simulation input tests: memh_2/rev-test-memh2.py
- C/CXX test source (or mention which test in the current source tree failed)

**Expected behavior**
Calculated sum of floating point variables should be identical on both test configs.

**Trace**
If applicable, post a trace of the execution.  

**Additional context**
In commit bb30d90c DependencyCheck function does not return any value and it's output is calculated on register leftover value:
```
bool RevProc::DependencyCheck(unsigned HartID, const RevInst* I) const {
  const RevRegFile* regFile = GetRegFile(HartID);
  const RevInstEntry* E = &InstTable[I->entry];

  return;
    // check LS queue for outstanding load
    LSQCheck(HartID, regFile, I->rs1, E->rs1Class) ||
    LSQCheck(HartID, regFile, I->rs2, E->rs2Class) ||
    LSQCheck(HartID, regFile, I->rs3, E->rs3Class) ||
    LSQCheck(HartID, regFile, I->rd , E->rdClass) ||

    // Iterate through the source registers rs1, rs2, rs3 and find any dependency
    // based on the class of the source register and the associated scoreboard
    ScoreboardCheck(regFile, I->rs1, E->rs1Class) ||
    ScoreboardCheck(regFile, I->rs2, E->rs2Class) ||
    ScoreboardCheck(regFile, I->rs3, E->rs3Class);
}
```

ukasz, 2023-12-19:
Additional traces:
memh:
```
WARNING: Element revcpu.RevBasicMemCtrl has slot with an invalid name: memIface%(memIface_ports)d
[RevMemCtrl 0]: No cache detected; disabling caching
RDT: Executed PC = 120ac Inst:    addi %rd, %rs1, $imm r2= fffffbc8 float[nan] double[0] r2= fffffbc8 r0= 0 imm = fd0
RDT: Address of RD = 0x0x55ffd289cbe8
RDT: Executed PC = 120b0 Inst:     sd %rs2, $imm(%rs1) r0= 0 float[nan] double[0] r2= fffffbc8 r1= 0 imm = 28
RDT: Address of RD = 0x0x55ffd289cbd8
RDT: Executed PC = 120b4 Inst:     sd %rs2, $imm(%rs1) r0= 0 float[nan] double[0] r2= fffffbc8 r8= 1a238 imm = 20
RDT: Address of RD = 0x0x55ffd289cbd8
RDT: Executed PC = 120b8 Inst:    addi %rd, %rs1, $imm r8= fffffbf8 float[nan] double[0] r2= fffffbc8 r0= 0 imm = 30
RDT: Address of RD = 0x0x55ffd289cc18
RDT: Executed PC = 120bc Inst:    addi %rd, %rs1, $imm r10= 0 float[nan] double[0] r0= 0 r0= 0 imm = 0
RDT: Address of RD = 0x0x55ffd289cc28
RDT: Executed PC = 120c0 Inst:     sw %rs2, $imm(%rs1) r0= 0 float[nan] double[0] r8= fffffbf8 r10= 0 imm = fec
RDT: Address of RD = 0x0x55ffd289cbd8
RDT: Executed PC = 120c4 Inst:     sw %rs2, $imm(%rs1) r0= 0 float[nan] double[0] r8= fffffbf8 r10= 0 imm = fe0
RDT: Address of RD = 0x0x55ffd289cbd8
RDT: Executed PC = 120c8 Inst:     sw %rs2, $imm(%rs1) r0= 0 float[nan] double[0] r8= fffffbf8 r10= 0 imm = fdc
RDT: Address of RD = 0x0x55ffd289cbd8
RDT: Executed PC = 120cc Inst:           jal %rd, $imm r0= 0 float[nan] double[0] r0= 0 r0= 0 imm = 4
RDT: Address of RD = 0x0x55ffd289cbd8
RDT: Executed PC = 120d0 Inst:      lw %rd, $imm(%rs1) r11= fffffffffffffab0 float[nan] double[0] r8= fffffbf8 r0= 0 imm = fdc
RDT: Address of RD = 0x0x55ffd289cc30
RDT: Executed PC = 120d4 Inst:    addi %rd, %rs1, $imm r10= 1 float[nan] double[0] r0= 0 r0= 0 imm = 1
RDT: Address of RD = 0x0x55ffd289cc28
RDT: Executed PC = 120d8 Inst:    bge %rs1, %rs2, $imm r0= 0 float[nan] double[0] r10= 1 r11= 0 imm = 8
RDT: Address of RD = 0x0x55ffd289cbd8
RDT: Executed PC = 120e0 Inst:           jal %rd, $imm r0= 0 float[nan] double[0] r0= 0 r0= 0 imm = 4
RDT: Address of RD = 0x0x55ffd289cbd8
RDT: Executed PC = 120e4 Inst:      lw %rd, $imm(%rs1) r10= 1 float[nan] double[0] r8= fffffbf8 r0= 0 imm = fdc
RDT: Address of RD = 0x0x55ffd289cc28
RDT: Executed PC = 120e8 Inst:      fcvt.s.w %rd, %rs1 r15= 0 float[1] double[-nan] r10= 1 r0= 0 imm = 0
RDT: Address of RD = 0x0x55ffd289cc50
RDT: Executed PC = 120ec Inst:    slli %rd, %rs1, $imm r11= 0 float[nan] double[0] r10= 0 r0= 0 imm = 2
RDT: Address of RD = 0x0x55ffd289cc30
RDT: Executed PC = 120f0 Inst:    addi %rd, %rs1, $imm r10= fffffbdc float[nan] double[0] r8= fffffbf8 r0= 0 imm = fe4
RDT: Address of RD = 0x0x55ffd289cc28
RDT: Executed PC = 120f4 Inst:     add %rd, %rs1, %rs2 r10= fffffbdc float[nan] double[0] r10= fffffbdc r11= 0 imm = 0
RDT: Address of RD = 0x0x55ffd289cc28
RDT: Executed PC = 120f8 Inst:    fsw %rs2, $imm(%rs1) r0= 0 float[nan] double[0] r10= fffffbdc r15= 0 imm = 0
RDT: Address of RD = 0x0x55ffd289cbd8
RDT: Executed PC = 120fc Inst:           jal %rd, $imm r0= 0 float[nan] double[0] r0= 0 r0= 0 imm = 4
RDT: Address of RD = 0x0x55ffd289cbd8
RDT: Executed PC = 12100 Inst:      lw %rd, $imm(%rs1) r10= fffffffffffffbdc float[nan] double[0] r8= fffffbf8 r0= 0 imm = fdc
RDT: Address of RD = 0x0x55ffd289cc28
RDT: Executed PC = 12104 Inst:   addiw %rd, %rs1, $imm r10= 1 float[nan] double[0] r10= 1 r0= 0 imm = 1
RDT: Address of RD = 0x0x55ffd289cc28
RDT: Executed PC = 12108 Inst:     sw %rs2, $imm(%rs1) r0= 0 float[nan] double[0] r8= fffffbf8 r10= 1 imm = fdc
RDT: Address of RD = 0x0x55ffd289cbd8
RDT: Executed PC = 1210c Inst:           jal %rd, $imm r0= 0 float[nan] double[0] r0= 0 r0= 0 imm = 1fffc4
RDT: Address of RD = 0x0x55ffd289cbd8
```
basic
```
WARNING: Building component "cpu" with no links assigned.
WARNING: Element revcpu.RevBasicMemCtrl has slot with an invalid name: memIface%(memIface_ports)d
RDT: Executed PC = 120ac Inst:    addi %rd, %rs1, $imm r2= fffffbc8 float[nan] double[0] r2= fffffbc8 r0= 0 imm = fd0
RDT: Address of RD = 0x0x562101934d48
RDT: Executed PC = 120b0 Inst:     sd %rs2, $imm(%rs1) r0= 0 float[nan] double[0] r2= fffffbc8 r1= 0 imm = 28
RDT: Address of RD = 0x0x562101934d38
RDT: Executed PC = 120b4 Inst:     sd %rs2, $imm(%rs1) r0= 0 float[nan] double[0] r2= fffffbc8 r8= 1a238 imm = 20
RDT: Address of RD = 0x0x562101934d38
RDT: Executed PC = 120b8 Inst:    addi %rd, %rs1, $imm r8= fffffbf8 float[nan] double[0] r2= fffffbc8 r0= 0 imm = 30
RDT: Address of RD = 0x0x562101934d78
RDT: Executed PC = 120bc Inst:    addi %rd, %rs1, $imm r10= 0 float[nan] double[0] r0= 0 r0= 0 imm = 0
RDT: Address of RD = 0x0x562101934d88
RDT: Executed PC = 120c0 Inst:     sw %rs2, $imm(%rs1) r0= 0 float[nan] double[0] r8= fffffbf8 r10= 0 imm = fec
RDT: Address of RD = 0x0x562101934d38
[... 12 lines cut ...]
RDT: Executed PC = 120e0 Inst:           jal %rd, $imm r0= 0 float[nan] double[0] r0= 0 r0= 0 imm = 4
RDT: Address of RD = 0x0x562101934d38
RDT: Executed PC = 120e4 Inst:      lw %rd, $imm(%rs1) r10= 0 float[nan] double[0] r8= fffffbf8 r0= 0 imm = fdc
RDT: Address of RD = 0x0x562101934d88
RDT: Executed PC = 120e8 Inst:      fcvt.s.w %rd, %rs1 r15= 0 float[0] double[-nan] r10= 0 r0= 0 imm = 0
RDT: Address of RD = 0x0x562101934db0
RDT: Executed PC = 120ec Inst:    slli %rd, %rs1, $imm r11= 0 float[nan] double[0] r10= 0 r0= 0 imm = 2
RDT: Address of RD = 0x0x562101934d90
[... cut ...]
RDT: Executed PC = 12100 Inst:      lw %rd, $imm(%rs1) r10= 0 float[nan] double[0] r8= fffffbf8 r0= 0 imm = fdc
RDT: Address of RD = 0x0x562101934d88
RDT: Executed PC = 12104 Inst:   addiw %rd, %rs1, $imm r10= 1 float[nan] double[0] r10= 1 r0= 0 imm = 1
RDT: Address of RD = 0x0x562101934d88
RDT: Executed PC = 12108 Inst:     sw %rs2, $imm(%rs1) r0= 0 float[nan] double[0] r8= fffffbf8 r10= 1 imm = fdc
RDT: Address of RD = 0x0x562101934d38
RDT: Executed PC = 1210c Inst:           jal %rd, $imm r0= 0 float[nan] double[0] r0= 0 r0= 0 imm = 1fffc4
RDT: Address of RD = 0x0x562101934d38
```

jleidel, 2023-12-19:
I'm attempting to reproduce this on the current `devel` branch.  Regarding the `RevProc::DependencyCheck` method, the return statement is actually return a complex set of OR's between `LSQCheck` and `ScoreboardCheck`

jleidel, 2023-12-19:
I can't reproduce using RISC-V GCC 12.2.0 or Clang from 8.8.2023.  I may need to update our Clang toolchain.

ukasz, 2023-12-19:
My bad about the return. Also I should have use different assert in c repro. != 3 will always pass because we're adding 0 + 1 only.
```
assert(res == 1.0f);
```

jleidel, 2023-12-19:
confirming the issue with RISCV GCC 12.2.0

ukasz, 2023-12-19:
Great, a bit too late, but maybe it will help. It should demonstrate similar scenario when there is no clear reg-> reg dependency but rather reg->mem->reg.
```
#define assert(x) if (!(x)) { asm(".byte 0x00"); asm(".byte 0x00"); asm(".byte 0x00"); asm(".byte 0x00"); }

__attribute__((naked)) float test() {
        asm("addi sp, sp, -64;");
        asm("sd ra, 56(sp);");
        asm("sd s0, 48(sp);");
        asm("addi s0, sp, 64;");
        asm("li a0, 0;");
        asm("sw a0, -36(s0);");
        asm("lw a1, -36(s0);");
        asm("fcvt.s.w fa0, a1;");
        asm("ld ra, 56(sp);");
        asm("ld s0, 48(sp);");
        asm("addi sp, sp,64;");
        asm("ret;");
}

int main() {
        assert(test() == 0.0f);
	return 0;
}
```

jleidel, 2023-12-19:
Excellent, thanks for the concise test.  Rebuilt the latest clang compiler and it clearly happens with the C test above.

leekillough, 2023-12-19:
I am trying to reproduce it, to no avail. I am trying to remove any dependence on the compiler or its version, so I use raw assembly:
```
        .text
        .align  2
        .globl  main
        .type   main, @function
main:
        add     sp, sp, -16
        sw      zero, 0(sp)
        lw      a0, 0(sp)
        fcvt.s.w fa0, a0
        fmv.w.x fa1, zero
        feq.s   a0, fa0, fa1
        bnez    a0, 1f
        .word   0
1:
        add     sp, sp, 16
        ret
```

@ukasz's  concise test case above with inline assembly, should not depend on the compiler, unless the prolog/epilog code is different and triggering it. Or maybe it's in the runtime startup.

leekillough, 2023-12-19:
I had `MemH` turned off. I'm still getting used to the new test framework.

leekillough, 2024-01-13:
Should be fixed. Please test and confirm.

ukasz, 2024-01-15:
Fix confirmed, thank you!

commit f1ad71c0 (2024-01-13): Merge pull request #230 from tactcomplabs/memh_tests

Fix incorrect register classes causing dependency errors; add new MemH test for dependency checks; remove obsolete MemH tests

Fixes #228
closed 2024-01-15 by ukasz