tactcomplabs/rev #253: Out of Memory error when using rev_mmap (with the memH memory backend)

Artlands, 2024-02-23:
**Describe the bug**

Running the following program will give the out of memory error.

`int main () {
    size_t total_size = 1024;
    void *ptr = (void *)rev_mmap(0,
                                                      total_size,
                                                      PROT_READ | PROT_WRITE,
                                                      MAP_PRIVATE | MAP_ANONYMOUS,
                                                       -1,
                                                      0);
}`

**To Reproduce**
- SST-Core Version (13.1.0), gcc-10.1.0
- The devel branch
- The last commit hash: 9b7ef56ebd52f8960b8aa2c6098ac6f7f1c5acc3
- The compiler utilized to build Rev: gcc-10.1.0
- The compiler utilized to build the test: riscv64-unknown-elf-gcc 12.2.0
- The Python simulation input:
`
import os
import sst

VERBOSE = 1
MEM_SIZE = 1024*1024*1024-1

# Define the simulation components
comp_cpu = sst.Component("cpu", "revcpu.RevCPU")
comp_cpu.addParams({
        "verbose" : VERBOSE,                                # Verbosity
        "numCores" : 1,                               # Number of cores
        "clock" : "2.0GHz",                           # Clock
        "memSize" : MEM_SIZE,                         # Memory size in bytes
        "machine" : "[0:RV64GC]",                      # Core:Config; RV64G for core 0
        "startAddr" : "[0:0x00000000]",               # Starting address for core 0
        "memCost" : "[0:1:10]",                       # Memory loads required 1-10 cycles
        "program" : os.getenv("REV_EXE", "malloc.exe"),  # Target executable
        "enable_memH" : 1,                            # Enable memHierarchy support
        "splash" : 1                                  # Display the splash message
})

# Create the RevMemCtrl subcomponent
comp_lsq = comp_cpu.setSubComponent("memory", "revcpu.RevBasicMemCtrl")
comp_lsq.addParams({
      "clock"           : "2.0Ghz",
      "max_loads"       : 64,
      "max_stores"      : 64,
      "max_flush"       : 64,
      "max_llsc"        : 64,
      "max_readlock"    : 64,
      "max_writeunlock" : 64,
      "max_custom"      : 64,
      "ops_per_cycle"   : 64
})

iface = comp_lsq.setSubComponent("memIface", "memHierarchy.standardInterface")

l1cache = sst.Component("l1cache", "memHierarchy.Cache")
l1cache.addParams({
    "access_latency_cycles" : "4",
    "cache_frequency" : "2 Ghz",
    "replacement_policy" : "lru",
    "coherence_protocol" : "MESI",
    "associativity" : "4",
    "cache_line_size" : "64",
    "L1" : "1",
    "cache_size" : "16KiB"
})

memctrl = sst.Component("memory", "memHierarchy.MemController")
memctrl.addParams({
    "clock" : "2GHz",
    "addr_range_start" : 0,
    "addr_range_end" : MEM_SIZE,
    "backing" : "mmap",
})

memory = memctrl.setSubComponent("backend", "memHierarchy.simpleMem")
memory.addParams({
    "access_time" : "100ns",
    "mem_size" : "8GB"
})

link1 = sst.Link("link1")
link1.connect( (iface, "port", "1ns"), (l1cache, "high_network_0", "1ns") )
link2 = sst.Link("link2")
link2.connect( (l1cache, "low_network_0", "1ns"), (memctrl, "direct_link", "1ns") )

`
- The test source is similar to `/test/syscalls/munmap` but using the aforementioned code.

**Expected behavior**
The simulation should complete without any issues.

**Trace**
`RevCPU[cpu:RevCPU:0]: Initialization of RevCPUs complete.
Expanding heap by 1073740635 Bytes
Expanding heap from 0x159b1 to 0x4001550c
MaxHeapSize = 0xfffffff
FATAL: RevCPU[cpu:ExpandHeap:648500]: Out Of Memory --- Attempted to expand heap to 0x4001550c which goes beyond the maxHeapSize = 0xfffffff set in the python configuration. If unset, this value will be equal to 1/4 of memSize.`

From the trace, we can see that the `total_size` is not passed to rev_mmap correctly. 

**Additional context**
The value of `total_size` is not loaded to `a1` before executing rev_mmap. This is because RevProc::ECALL_mmap does not check the scoreboard. The scoreboard should be checked for all registers used in the system calls to avoid this issue. I add the following code before reading the registers in RevProc::ECALL_mmap to fix the bug:

`if (ScoreboardCheck(RegFile, static_cast<uint16_t>(RevReg::a0), RevRegClass::RegGPR) || 
      ScoreboardCheck(RegFile, static_cast<uint16_t>(RevReg::a1), RevRegClass::RegGPR)) {
    return EcallStatus::CONTINUE;
  }`

leekillough, 2024-02-26:
Should be fixed now in `devel`. If confirmed, please close this issue.

Artlands, 2024-02-26:
Thank you! The update fixes the reported bug.

closed 2024-02-26 by Artlands