tactcomplabs/rev #261: Tracer is printing compressed register move instructions as c.jr instructions

leekillough, 2024-03-19:
**Describe the bug**

When a compressed register <- register copy ( `c.mv` ) is encountered, Rev Tracer prints it as a `c.jr` instruction.

**To Reproduce**

Latest `devel` 

**Expected behavior**

It should print `c.mv` when `rs2==0`

**Trace**

```
RevCPU[cpu:GetNextHartToDecodeID:41465000]: Core 0; Hart switch from 0 to 0
RevCPU[cpu:FetchAndDecodeInst:41465000]: Core 0; Hart 0; Thread 1; PC:InstPayload = 0x19fce:0xe90187aa
RevCPU[cpu:Render:41465000]: Core 0; Hart 0; Thread 1; *I 0x19fce:e90187aa    c.jr %rs1  0x0<-x0 0x0<-x10 x15<-0x0
```
```
RevCPU[cpu:GetNextHartToDecodeID:41467000]: Core 0; Hart switch from 0 to 0
RevCPU[cpu:FetchAndDecodeInst:41467000]: Core 0; Hart 0; Thread 1; PC:InstPayload = 0x19fd2:0xf0ef8522
RevCPU[cpu:Render:41467000]: Core 0; Hart 0; Thread 1; *I 0x19fd2:f0ef8522    c.jr %rs1  0x0<-x0 0x90<-x8 x10<-0x90
```

**Additional context**

`C.MV` copies the value in register `rs2` into register `rd`. `C.MV` expands into `add rd, x0, rs2`. `C.MV`
is only valid when `rs2 != x0`; the code points with `rs2 = x0` correspond to the `C.JR` instruction. The
code points with `rs2 != x0` and `rd = x0` are HINTs.

`C.MV` expands to a different instruction than the canonical `MV` pseudoinstruction, which instead
uses `ADDI`. Implementations that handle `MV` specially, e.g. using register-renaming hardware,
may find it more convenient to expand `C.MV` to `MV` instead of `ADD`, at slight additional hardware
cost.

donofrio, 2024-03-20:
Is this using tracer in conjunction with Spike?

leekillough, 2024-03-20:
No

PR #262 (2024-03-27): Add instruction table predicate option, to disambiguate compressed instructions
This adds an optional predicate to the instruction table entries, which can be set by calling `.SetPredicate()`.

This fixes #261 and #260 by having separate table entries for the different mnemonics, and using the `rd/rs1` or `rs2` register fields to disambiguate them.

A predicate can be added to an instruction table entry as simply as `.SetPredicate( []( uint32_t Inst ){ return DECODE_LOWER_CRS2(Inst) != 0; } )`, which in this example makes sure that the compressed-encoded `rs2` register is not `x0` for the `c.m [... cut]
closed 2024-03-27 by leekillough