tactcomplabs/rev #282: rev_read operation slowdown after large number of lines read

jleidel, 2024-05-22:
**Describe the bug**
`rev_read` performance significantly degrades after reading thousands of lines of text.  Performance on the first 30K lines is relatively consistent.  Afterwards, performance drops to < 10% of the same rate.  

SST 13.1.0; rev `devel` branch

**To Reproduce**
use the code below to create a sample class and underlying reader.  create a large file with ~50K lines

**Expected behavior**
consistent reader performance

**Trace**
If applicable, post a trace of the execution.  

**Additional context**
Buffered read code is as follows:
```
#include <rev/common/syscalls/syscalls.h>
#include <fcntl.h>
#include <cstddef>
#include <optional>
#include <string_view>

/// Buffered file reader.
template <std::size_t _buffer_size = 512>
struct Reader
{
    int const _fd;
    char _read_buffer[_buffer_size]{};
    char const* _current = _read_buffer;
    char* _end = _read_buffer;

    Reader(int const fd) : _fd(fd) {
    }

    auto next() ->std::optional<std::string_view>
    {
        if (auto e = _search_newline()) {
            std::string_view const str(_current, e);
            _current = e + 1;
            return str;
        }

        // no newline was found, we might be at the end of a file without a
        // newline
        if (_size()) {
            std::string_view const str(_current, _end);
            _current = _end;
            return str;
        }

        // No more data.
        return std::nullopt;
    }

  private:
    /// Return the number of valid bytes in the buffer.
    auto _size() const -> std::size_t {
        return _end - _current;
    }

    /// Return the amount of free space at the end of the buffer.
    auto _free() const -> std::size_t {
        return _buffer_size - (_end - _read_buffer);
    }

    /// Move any existing data down to the base of the buffer.
    auto _compact() -> std::size_t
    {
        auto const n = _size();

        // copy the bytes down
        for (auto i = 0zu; i < n; ++i) {
            _read_buffer[i] = _current[i];
        }

        // reset the pointers for the valid region
        _current = _read_buffer;
        _end = _read_buffer + n;

        // return the size of the buffer
        return n;
    }

    /// Refill the tail of the buffer.
    auto _refill() -> std::size_t
    {
        auto const size = _compact();
        auto const free = _buffer_size - size;
        auto const read = rev_read(_fd, _end, free);
        _end += read;
        return _size();
    }

    auto _search_newline() -> char const* {
        return _find_newline() ?: (_refill(), _find_newline());
    }

    auto _find_newline() const -> char const*
    {
        for (auto i = _current; i < _end; ++i) {
            if (*i == '\n') {
                return i;
            }
        }
        return nullptr;
    }
};
```

leekillough, 2024-05-23:
I cannot reproduce. What was your `.py` config? Are you using a Release build of Rev?

With 20,000 lines around 80 characters wide each:
```
real    0m3.999s
user    0m3.240s
sys     0m0.601s
```
With 50,000 lines around 80 characters wide each:
```
real    0m8.242s
user    0m7.434s
sys     0m0.652s
```
`8.242 / 3.999` = `2.061`

`50000.0 / 20000.0` = `2.5`

So the 50,000 line one is actually much faster per-line than the 20,000 line one.

That was with RevMem. Here it is for MemH 20,000 lines:
```
real    1m45.388s
user    1m45.139s
sys     0m0.084s
```
MemH 50,000 lines:
```
real    4m20.000s
user    4m19.723s
sys     0m0.100s

```
`260.000 / 105.388` = `2.467`, so like RevMem, MemH is faster per-line with 50,000 lines, but just marginally.

I did not test with SST 13.1.0 since I'll need to build it. But if the problem is in `rev_read`, it shouldn't depend on the SST version.

leekillough, 2024-05-31:
This test is not reproducible, and after communicating with Luke, he cannot reproduce it either, with a loop which simply reads and prints lines.

We suspect it may be something to do with his tuple parsing code, which this reduced test case has removed, rather than rev_read().

I asked for a profiler run to be done to identify the hotspots, but Luke is too busy with other tasks now.

closed 2024-05-31 by leekillough