tactcomplabs/rev #293: [FEATURE REQUEST] Different testing levels

leekillough, 2024-06-23:
A testing level should be added to control the level and timing of testing. Such as Low, Medium, High or simply Short and Long.

This is preferred to explicitly naming tests, such as `BUILD_ASM_TESTING` or `ENABLE_FP_TESTING`.

rkabrick, 2024-06-23:
What tests will be a part of what levels?

leekillough, 2024-06-23:
That's for the solver of this issue to decide

leekillough, 2024-06-24:
https://github.com/tactcomplabs/rev/pull/238#issuecomment-2184314058

leekillough, 2024-06-24:
You can look at the time each test takes by running it today. From there, you can do a Huffman-like compression, by sorting the times, and taking all of the fastest tests which take 1/3 of the total time. That would be the "Low" category. Then among the remaining tests, take the fastest which add up to 1/3 of the total time. They and the Low tests would be in the Med category. Finally, the remaining tests and Med would be in the High category. That's one way to do it in thirds.

But another way would be to put all of the fastest tests in the Low category up to where there is a huge climb in the timing of the tests. Sort the tests by runtime, and plot the runtimes. At the first large jump, put everything below that jump in the Low category. Then do the same with the remaining tests, finding a large jump and putting all tests under that jump in the Med category. Put any remaining tests in the High category.

Numbers like 1, 2, 3 can be used instead of category names, so that in CMake you can add a tag like `testevel=1` to each test with a number on it, and then only run a test if its number is less than or equal to the testing level.

PR #294 (2024-06-24): Add TEST_LEVEL variable to control which tests are run
`-DTEST_LEVEL=3` runs all tests, including the FP tests on the `fenv` branch. It is the default level.

`-DTEST_LEVEL=2` runs all tests except the `fenv` tests. It runs the `ISA` tests, which are no longer controlled by `BUILD_ASM_TESTING`.

`-DTEST_LEVEL=1` runs a small number of tests, and takes less than a minute to run.

This fixes #293 .

The `all` label has been removed from tests.
closed 2024-06-24 by leekillough