tactcomplabs/rev #319: Deprecated functions in SST 14 cause build failure

mjeje, 2024-09-06:
**Describe the bug**
Building Rev fails with SST 14.0.0. Some source code lines reference deprecated functions which no longer exist in version 14. These function calls will need to be replaced.
Additionally, can a dependency version list be added to the README?

[sendInitData](https://github.com/sstsimulator/sst-core/blob/94dd6f5eab5ad7931702b4fc5e2ba2faf622382e/src/sst/core/interfaces/simpleNetwork.h#L233): `[[deprecated(
        "sendInitData() has been deprecated and will be removed in SST 14.  Please use sendUntimedData() instead.")]]`
[recvInitData](https://github.com/sstsimulator/sst-core/blob/94dd6f5eab5ad7931702b4fc5e2ba2faf622382e/src/sst/core/interfaces/simpleNetwork.h#L244): `[[deprecated(
        "recvInitData() has been deprecated and will be removed in SST 14.  Please use recvUntimedData() instead.")]]`

**To Reproduce**
```
git clone git@github.com:tactcomplabs/rev.git
cd rev/build
cmake ..
make -j
```

All issues must be accompanied by: 
- The version of SST utilized (`sst --version`; `sst-config --CXX`; `sst-config --ELEMENT_CXXFLAGS`): 
```
SST-Core Version (14.0.0)
g++ -std=c++17
-std=c++17 -ffp-contract=off -fPIC -DHAVE_CONFIG_H -I/foo/bar/sst/install/include
```
- The source code branch being used
`main`
- The last commit hash
`b2d0b966538c85a3951ac3c35302696e504b676f`
- The compiler utilized to build Rev
```
Apple clang version 15.0.0 (clang-1500.3.9.4)
Target: arm64-apple-darwin23.6.0
Thread model: posix
InstalledDir: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin
```
- The compiler utilized to build the test
```
Apple clang version 15.0.0 (clang-1500.3.9.4)
Target: arm64-apple-darwin23.6.0
Thread model: posix
InstalledDir: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin
```
- The python or JSON simulation input
`n/a`
- C/CXX test source (or mention which test in the current source tree failed)
`n/a`

**Expected behavior**
Build successfully with SST 14.0.0

**Trace**
```
[ 10%] Building CXX object src/CMakeFiles/revcpu.dir/RevNIC.cc.o
clang: warning: -Wl,-undefined: 'linker' input unused [-Wunused-command-line-argument]
clang: warning: -Wl,dynamic_lookup: 'linker' input unused [-Wunused-command-line-argument]
clang: warning: argument unused during compilation: '-shared' [-Wunused-command-line-argument]
/foo/bar/rev/src/RevNIC.cc:71:14: error: no member named 'sendInitData' in 'SST::Interfaces::SimpleNetwork'
      iFace->sendInitData(req);
      ~~~~~  ^
/foo/bar/rev/src/RevNIC.cc:75:65: error: no member named 'recvInitData' in 'SST::Interfaces::SimpleNetwork'
  while( SST::Interfaces::SimpleNetwork::Request * req = iFace->recvInitData() ) {
                                                         ~~~~~  ^
2 errors generated.
make[2]: *** [src/CMakeFiles/revcpu.dir/RevNIC.cc.o] Error 1
make[1]: *** [src/CMakeFiles/revcpu.dir/all] Error 2
make: *** [all] Error 2
```

**Additional context**
`n/a`

rkabrick, 2024-09-06:
All development occurs on the devel branch. 

Not really a bug but @jleidel if you want to update main go for it... otherwise this can be closed

jleidel, 2024-09-06:
> All development occurs on the devel branch.
> 
> Not really a bug but @jleidel if you want to update main go for it... otherwise this can be closed

All current development is definitely on the `devel` branch.  `main` is considered "stable", but not useful for development/debugging

mjeje, 2024-09-06:
I'll close this issue. Thanks.

mjeje, 2024-09-06:
I ran into this issue while building dependencies for an unrelated project, for which I have the latest version of SST installed. It looks like the deprecation fixes are added to `devel`, and will presumably be pushed to `main`. So, I'll close this one. Thanks

closed 2024-09-06 by mjeje