tactcomplabs/rev #331: SST Segmentation Fault When Accessing Memory Across Pages with memHierarchy Backend

Artlands, 2024-10-04:
**Describe the bug**
A segmentation fault occurs in SST when reading memory across pages using the memHierarchy backend due to incorrect logic in _RevMem::ReadMem_.

**To Reproduce**
The issue is observed in the devel branch when the page size is changed to a smaller value, such as 4KB, during the execution of the _test/syscalls/munmap_ test.

**Additional context**
In the _RevMem::ReadMem_ function, the code responsible for handling flag responses and clearing hazards (lines 770-783 in _RevMem.cc_) should be placed inside the else block when using the internal RevMem model.

**Other comments/questions**
When using the memHierarchy backend, the cross-paging access is not relevant not only for memory read (as commented in line 771) but also for memory write. There is redundancy in calling `ctrl->sendWRITERequest` in _RevMem::WriteMem_, and the logic in these functions can be simplified. Additionally, the BaseMem object passed to `ctrl->sendWRITERequest` and `ctrl->sendREADRequest` is never used; only the virtual address is needed when sending memory requests to memHierarchy. Therefore, the memory controller interface could also be simplified.

leekillough, 2024-10-04:
Can you please attach a test code?

Artlands, 2024-10-04:
> Can you please attach a test code?
The test code is `/test/syscalls/munmap/munmap.c`

After changing the pageSize to 4096 in RevMem.cc and reinstalling REV, run `sst rev-model-options-config.py --model-options="--program=./syscalls/munmap/munmap.exe --enableMemH=1"` in the test folder.

leekillough, 2024-10-04:
If I run as you say, I get another error, an assertion failure:
```
RevCPU[cpu:ECALL_mmap:386000]: ECALL: mmap called
FATAL: RevCPU[cpu:FetchAndDecodeInst:169784000]: Error: Core 0 failed to decode instruction at PC=0x11278; Inst=0
```

leekillough, 2024-10-04:
The assertion failure might be based on a page size assumption of the test. If I use the default page size, the test passes. There is no segfault anymore.

leekillough, 2024-10-07:
Please close this issue if you can verify that it's fixed.

closed 2024-10-07 by Artlands

Artlands, 2024-10-10:
Thank you for fixing the previous bug!

While reviewing the RevMem.cc code in the devel branch, I came across another issue that I wanted to bring to your attention. I believe it’s better to mention it here rather than opening a new issue.

When using MemH as the memory backend, RevMem::LR sends a ReadLock request (line 118 in RevMem.cc in the devel branch). However, the SC interface calls WriteMem, which does not send a WriteUnlock request. Since RevMem already has an LRSC data structure to manage the LR/SC operations, it might be more appropriate to replace the ReadLock request with a standard Read request.

leekillough, 2024-10-10:
Thanks for your comments.

I recently reimplemented the `Zalrsc` part of the `A` extension, making the hart hold a reservation and invalidating any hart's reservation when another hart writes to its address range.

But I'm not certain about the timing and whether it is being simulated correctly.

I have a [PR](https://github.com/tactcomplabs/rev/pull/333) open addressing your other comments about memory.